fwupd/firmware-lenovo #236: ThinkPad Yoga X1 Gen 6 / Carbon Gen 9 Thunderbolt Retimer Firmware fails to start 1.2.28.0 with code 10

karianna, 2022-06-21:
Can no longer use my eGPU. Rolling back firmware to 1.2.26.0 did not resolve the issue.  

_This device cannot start. (Code 10)

Firmware update was unsuccessful._

Tried a couple of the resolutions in https://www.reddit.com/r/thinkpad/comments/tt0lk6/fmp_capsule_update_fail_and_thinkpad_x1_yoga_gen/ but no luck.

Windows 11 fully updated, Lenovo Advantage Installed an all drivers up to date and hardware scan reports no issues.

mrhpearson, 2022-06-21:
Hi - did you install the update from Linux/LVFS or from Windows? 

The ticket you are linking to is all Windows related and I saw recently that there are lots of updates/fixes  to the TBT FW updater utility for Windows (that doesn't apply to Linux) . This forum is for Linux updates - so you might be in the wrong place :)

If the issue is happening from Linux let me know and I'll flag to the FW team

Mark

karianna, 2022-06-21:
It is indeed for Windows (so sorry!). Where would you like me to move this to?