fwupd/firmware-lenovo #238: ThinkPad Universal USB-C Dock , missing FW

joakim-tjernlund, 2022-06-24:
Just got an ThinkPad Universal USB-C Dock and tried to update FW but there is no FW:
```
├─ThinkPad Universal USB-C Dock:
│     Device ID:          7a2034b0a5bfbc758ec5bfdce77b0484bdc0b46f
│     Summary:            Dock Management Controller Device
│     Current version:    0.0.0.0
│     Vendor:             Lenovo (USB:0x17EF)
│     Install Duration:   1 minute
│     Serial Number:      1SSD21B41469ZVR08KVN
│     GUIDs:              959b1e72-280e-5453-a5eb-d9d5c7674211 ← USB\VID_17EF&PID_30A9
│                         fa0b1b28-bca0-5577-a670-a7e507f03bda ← USB\VID_17EF&PID_30A9&REV_3073
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
│                         • Device can recover flash failures
│                         • Signed Payload
```

mrhpearson, 2022-06-24:
The FW is under embargo but should be moving to stable soon.
I have tested it myself - but they wanted the Lenovo QA team to check it too. I've asked for an ETA.

As a side Q - which dock do you have? Hoping it's the 40AS or 40AY model....

joakim-tjernlund, 2022-06-26:
Don't know the exact model(I am remote ATM), I was hoping you could see that from list above?
Also, is it not odd that Current version is 0.0.0.0 ? 

hmm, lsusb says:
Bus 003 Device 017: ID 17ef:30a9 Lenovo 40AY

joakim-tjernlund, 2022-06-26:
PXE boot does not seem to work on this dock, just tried remotely to initiate a PXE boot but the computer never came up.

joakim-tjernlund, 2022-06-28:
Got PXE working now, ipconfig in kernel is slow to complete bringing up USB IP I/F though.
Really need that Dock FW now ...

madushan1000, 2022-07-09:
@mrhpearson 40AS(Thinkpad usc c dock gen 2) is getting support too? Hopefully DP hub updates are supported as well, I can't even get the DP hub to update on windows. It just times out.

joakim-tjernlund, 2023-02-20:
We still experience some stability problems with the Lenovo 40AY Dock.  Fwupdmgr say we have the latest FW: 0.0.4.6
However looking at https://support.lenovo.com/us/en/downloads/ds548947 there is FW 3.0.80 for Windows. The FW versioning numbering does not match Linux and it is impossible to tell what FW Linux really uses. How can I match Linux FW with Windows ?

mrhpearson, 2023-02-23:
Hi - I've asked the dock team about this. Afraid I don't know the answer.
Will let you know when I hear back
Mark

mrhpearson, 2023-02-24:
Hi - they're asking:

1.      Details on the stability issue - ideally a pic or video if possible so they can understand the details of that
2.      Can you help to confirm the current Dock FW version? Can you get me the output from fwupdmgr get-devices
3.      Any details on how to reproduce; what kernel version & distro are you running

Thanks
mark

joakim-tjernlund, 2023-02-27:
1. No pics/video, sorry. I can describe what we have seen:
 - Loss of Dock eth port, it is often gone after an Linux reboot, replugging Dock USB cable usually restores it
 - When   Dock eth work we see alot of this kernel msg:
     r8152 2-3.1:1.0: load rtl8153b-2 v1 10/23/19 successfully
 -  persistent UCSI errors:
     ucsi_acpi USBC000:00: PPM init failed (-110)
    That seems to steam from this problem: https://lore.kernel.org/linux-usb/20210823180626.tb6m7h5tp6adhvt2@fastboi.localdomain/
    applying that patch makes ucsi a little happier but still see this:
      ucsi_acpi USBC000:00: ucsi_handle_connector_change: GET_CONNECTOR_STATUS failed (-110)

- Rare, but loss off Dock HDMI  port, replugging USB cable did not help but reboot of Linux did.
 
2.
```
─ThinkPad Universal USB-C Dock:
│ │   Device ID:          7a2034b0a5bfbc758ec5bfdce77b0484bdc0b46f
│ │   Summary:            Dock Management Controller Device
│ │   Current version:    0.0.4.6
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Install Duration:   1 minute
│ │   Serial Number:      1SSD21B41469ZVR08KVN
│ │   GUIDs:              959b1e72-280e-5453-a5eb-d9d5c7674211 ← USB\VID_17EF&PID_30A9
│ │                       fa0b1b28-bca0-5577-a670-a7e507f03bda ← USB\VID_17EF&PID_30A9&REV_3073
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Signed Payload
│ │ 
│ ├─CCG4:
│ │     Device ID:        1f5112276f265b9726e224103cd608c0acb6b026
│ │     Current version:  3.2.1.1652
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             671dec65-a2d3-5899-bb65-678605cdf74f ← USB\VID_17EF&PID_30A9&CID_02
│ │   
│ └─DMC:
│       Device ID:        194de6dea19ac74c18f9013822a06281226c3cfe
│       Current version:  3.3.0.1746
│       Vendor:           Lenovo (USB:0x17EF)
│       GUID:             2f9f0d4e-3713-5086-af85-b8f815549986 ← USB\VID_17EF&PID_30A9&CID_00
```
3.
kernel 6.1.x, x=10-14
dist: Gentoo
I don't think this is distribution specific. More kernel, BIOS FW and Dock FW.
I Got a ThinkPad T14 Gen 3 but we see this on all our  laptops

joakim-tjernlund, 2023-03-07:
Now we are seeing this OOPS also(kernel >= 6.1.14) when unplugging Dock USB:
```
[ 2286.499615] BUG: kernel NULL pointer dereference, address: 0000000000000398
[ 2286.499618] #PF: supervisor write access in kernel mode
[ 2286.499619] #PF: error_code(0x0002) - not-present page
[ 2286.499620] PGD 0 P4D 0 
[ 2286.499621] Oops: 0002 [#1] PREEMPT SMP NOPTI
[ 2286.499622] CPU: 0 PID: 3299 Comm: kworker/0:0 Tainted: G           O       6.1.15-x86_64 #1
[ 2286.499623] Hardware name: LENOVO 21AH00CSMX/21AH00CSMX, BIOS N3MET11W (1.10 ) 12/07/2022
[ 2286.499624] Workqueue: kacpi_notify acpi_os_execute_deferred
[ 2286.499627] RIP: 0010:queue_work_on+0x4/0x30
[ 2286.499630] Code: 82 48 c7 c7 88 97 0d 83 e8 99 5d 5c 00 3b 05 83 86 98 01 89 c1 41 89 c6 0f 82 65 ff ff ff e9 cb fe ff ff 0f 1f 0
0 53 9c 5b fa <f0> 48 0f ba 2a 00 73 0a 31 c0 80 e7 02 74 01 fb 5b c3 e8 75 fb ff
[ 2286.499631] RSP: 0018:ffffc90003cd7e50 EFLAGS: 00010002
[ 2286.499632] RAX: 0000000000000368 RBX: 0000000000000202 RCX: 000000000037c080
[ 2286.499633] RDX: 0000000000000398 RSI: ffff88810004c600 RDI: 0000000000000040
[ 2286.499633] RBP: 0000000000000004 R08: ffff88810d1a4d38 R09: 0000000000000000
[ 2286.499634] R10: 0000000000000000 R11: 000000000000005f R12: ffff88844f42aa40
[ 2286.499634] R13: ffff88844f431d00 R14: 0000000000000000 R15: ffff88844f431d05
[ 2286.499635] FS:  0000000000000000(0000) GS:ffff88844f400000(0000) knlGS:0000000000000000
[ 2286.499635] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2286.499636] CR2: 0000000000000398 CR3: 000000000480a004 CR4: 0000000000770ef0
[ 2286.499637] PKRU: 55555554
[ 2286.499637] Call Trace:
[ 2286.499638]  <TASK>
[ 2286.499640]  ucsi_acpi_notify+0x5f/0x69 [ucsi_acpi]
[ 2286.499645]  acpi_ev_notify_dispatch+0x37/0x4c
[ 2286.499647]  acpi_os_execute_deferred+0xb/0x20
[ 2286.499648]  process_one_work+0x1db/0x390
[ 2286.499649]  worker_thread+0x45/0x3c0
[ 2286.499650]  ? rescuer_thread+0x3a0/0x3a0
[ 2286.499650]  kthread+0xd2/0x100
[ 2286.499652]  ? kthread_complete_and_exit+0x20/0x20
[ 2286.499653]  ret_from_fork+0x1f/0x30
[ 2286.499655]  </TASK>
[ 2286.499655] Modules linked in: rfcomm snd_hrtimer snd_seq nf_conntrack_netlink nfnetlink xfrm_user xt_addrtype dm_thin_pool dm_per
sistent_data dm_bio_prison dm_bufio cmac snd_usb_audio snd_usbmidi_lib snd_rawmidi usbhid snd_seq_device bnep btusb btrtl btbcm btmtk
 btintel uvcvideo videobuf2_vmalloc bluetooth videobuf2_memops videobuf2_v4l2 ecdh_generic ecc videodev videobuf2_common xt_CHECKSUM 
xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp ip6table_mangle ip6table_nat ip6table_filter ip6_tables iptable_mangle
 iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_filter ip_tables x_tables snd_ctl_led snd_soc_skl_hda_dsp snd_
soc_intel_hda_dsp_common snd_soc_hdac_hdmi snd_sof_probes snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic vfat fat joy
dev hid_multitouch snd_soc_dmic snd_sof_pci_intel_tgl snd_sof_intel_hda_common soundwire_intel soundwire_generic_allocation soundwire
_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof
[ 2286.499678]  i2c_designware_platform i2c_designware_core snd_sof_utils snd_soc_hdac_hda snd_hda_ext_core snd_soc_acpi_intel_match 
snd_soc_acpi soundwire_bus x86_pkg_temp_thermal snd_soc_core snd_compress intel_rapl_msr iwlmvm ac97_bus think_lmi mac80211 firmware_
attributes_class snd_hda_intel wmi_bmof snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec kvm_intel i915 iwlwifi snd_hda_core thinkpa
d_acpi processor_thermal_device_pci intel_lpss_pci snd_hwdep kvm intel_gtt nvram intel_lpss processor_thermal_device snd_pcm ledtrig_
audio irqbypass ghash_clmulni_intel pcspkr efi_pstore ucsi_acpi platform_profile cfg80211 mfd_core i2c_i801 typec_ucsi processor_ther
mal_rfim snd_timer rfkill drm_buddy i2c_smbus roles mei_me processor_thermal_mbox xhci_pci snd processor_thermal_rapl mei intel_rapl_
common xhci_hcd thunderbolt drm_display_helper video typec int3403_thermal int340x_thermal_zone i2c_hid_acpi i2c_hid intel_hid pinctr
l_tigerlake wmi sparse_keymap int3400_thermal pinctrl_intel
[ 2286.499698]  acpi_thermal_rel vboxnetadp(O) vboxnetflt(O) vboxdrv(O) msr fuse dm_mod dax nfsd autofs4
[ 2286.499703] CR2: 0000000000000398
[ 2286.499703] ---[ end trace 0000000000000000 ]---
```
And the laptop needs a power off to recover.

Any update on Dock FW?

mrhpearson, 2023-03-08:
I've been looking into that - and it's weird. I see it with the TBT dock FWIW.
I don't see the problem with a vanilla build from Linus's tree of the 6.2 kernel - but with the Fedora 6.2 kernel I do. As soon as I add any instrumentation to debug it the issues goes away (though then the IGC kernel driver barfs...).

joakim-tjernlund, 2023-03-08:
My guess would that you somehow mange to emit some USCI notification from the dock even though
UCSI init fails in the kernel and then kernel driver isn't fully setup to handle UCSI notifications.

Double error of sorts, kernel does not properly check all UCSI msgs and does a NULL ptr access when
it gets some unexpected UCSI msg.

OOPS happens is the end of ucsi_acpi_notify, maybe  ae->complete  isn't setup ?

mrhpearson, 2023-03-10:
I'm not sure - but as a random thing to try if you don't mind...can you `rmmod igc` before disconnecting the dock?

When I instrument the kernel I get errors in the IGC module where it's unable to read HW (presumably because it's gone). I'm wondering if something in there is trampling or causing issues. At least in my build without the IGC module it's working - but I'm hesitant to point fingers yet...

joakim-tjernlund, 2023-03-11:
I have blacklisted the ucsi module for now. My kernel does not build igc so there is nothing to remove.
Do I need it for some Lenovo machines ?

mrhpearson, 2023-03-12:
Turns out smarter people than me have figured this out...
https://lore.kernel.org/linux-usb/20230308154244.722337-1-hdegoede@redhat.com/T/#t

Not tested yet - but it should fix the problem. I'll give this a whirl on my system shortly to confirm.

I think the IGC driver is used on the TBT4 dock - but you can ignore it for your case.

Mark

joakim-tjernlund, 2023-03-12:
That 3 patches looks nice, I will wait until these hit stable though.

joakim-tjernlund, 2023-03-12:
Would be nice if you could look into these as well;
* Loss of Dock eth port, it is often gone after an Linux reboot, replugging Dock USB cable usually restores it

* When Dock eth work we see alot of this kernel msg:
r8152 2-3.1:1.0: load rtl8153b-2 v1 10/23/19 successfully

The kernel log is full of this msg.

mrhpearson, 2023-03-17:
For the original source of this ticket - USB-c dock FW on LVFS.

Team have uploaded the update to 3.0.8.5, but it's currently under embargo. 
I've just tested it and successfully updated. The Lenovo QA team need to sign off before it will be released to stable so I'm pushing to make that happen

Haven't tested the eth port issues - but I've not seen problems there myself and I am using the Ethernet port successfully (and was previously too)

mrhpearson, 2023-03-30:
Ack - it's still under testing and is scheduled to be complete in the next couple of weeks.
Mark

Leobaillard, 2023-04-06:
Hi!

Sorry for barging in but I don't know where else to find info. I just got a brand new Thinkpad E15 Gen4 with a 40AY and can't get the displays working. I have a bunch of thunderbolt errors in dmesg and no thunderbolt devices showing up in /sys. I don't know much about fwupd but the dock isn't showing up in there but is showing up with lsusb.

Can you help with things I should check?

OS : Debian 11
Kernel : 5.10.0-21-amd64

Thanks!

joakim-tjernlund, 2023-04-06:
> 
> Can you help with things I should check?
> 
> OS : Debian 11 Kernel : 5.10.0-21-amd64

That is way too old kernel to talk to these docks, use latest 6.1.23 (.22 and .23 has usci fixes need)

Leobaillard, 2023-04-06:
> > Can you help with things I should check?
> > OS : Debian 11 Kernel : 5.10.0-21-amd64
> 
> That is way too old kernel to talk to these docks, use latest 6.1.23 (.22 and .23 has usci fixes need)

Thanks a lot for your answer, that was it! For now, there is only up to 6.1.20 available in the debian unstable repository, but at least I can have both screens working through the dock. I guess I'll wait for a newer release to get the stability improvements.

joakim-tjernlund, 2023-04-07:
> Thanks a lot for your answer, that was it! For now, there is only up to 6.1.20 available in the debian unstable repository, but at least I can have both screens working through the dock. I guess [...]

Meanwhile, don't unplug dock but if you need to, make sure screen is not locked.
There are OOPSes in there waiting ...

joakim-tjernlund, 2023-04-24:
> > For the original source of this ticket - USB-c dock FW on LVFS.
> > Team have uploaded the update to 3.0.8.5, but it's currently under embargo. I've just tested it and successfully updated. The Lenovo QA team need to sign off before it will be released to stable so I'm pushing to make that happen
> 
> Great ! Should be any day now I hope.

@mrhpearson , still no new USB dock FW. Some problem ?

mrhpearson, 2023-04-25:
I pinged them a week ago and the update was still with the QA team and not ready for release yet. There weren't any issues reported so hopefully it's just chewing it's way through the process

micw, 2023-06-09:
Hello,
I fund this when I was looking for a fix to the eth (RTL8153) issue with my Lenovo 40AY
The precise issue I have is:
* ethernet is connected
* CPU is on high load (mostly when having teams meetings ^^ but recently also when I tried to brute-force a forgotten PDF password)
* the usb ethernet device is shown as "disconnected" and re-discovered a few seconds later

I found some recommendations to disable auto-suspend on the USB device but it did not help.

The exact same issue occurs if a separate RTL8153 usb adapter is connected to the dock. The issue does not appear if the adapter is directly connected to my thinkpad.

Do you know if 3.0.8.5 will fix that issue?

joakim-tjernlund, 2023-06-09:
> Do you know if 3.0.8.5 will fix that issue?

I think not but that also depend on what Dock FW you have now. Just upgrade with fwupd and find out ?
We always found Dock USB eth unstable, even missing(Linux does not find it at all so no eth interface)

micw, 2023-06-09:
the firmware was 0.0.1 or 0.0.0.1 or so. I updated and had a teams call with no issues (which does not mean much, issue is not 100% reproducible). I will continue to test and report back here.
During lunch, I will run a cpu intensive task while I'm using and monitoring the dock ethernet to see if the issue occurs again.

micw, 2023-06-09:
No issues since I updated. Hope it remains ;-)

galibozek, 2024-07-02:
firmware outdated

joakim-tjernlund, 2024-07-02:
> firmware outdated

So it seems:
```
Package          DP HUB          VIA HUB(Q7,Q8)    PD (CCG4)     DMC          Audio codec   Date
---------        ---------       ---------         ---------     ---------    ---------     ---------
V3.0.92          5.07.003    	 5134              5.4.109       0.2.086      49-0E-43      2023/12/08
V3.0.91          5.07.103    	 5134              5.4.107       0.2.086      49-0E-43      2023/10/16
V3.0.85          5.07.001    	 5134              5.4.95        0.2.086      49-0E-43      2023/05/26	
V3.0.80          5.06.004    	 5114              5.4.95        0.2.083      49-0E-42      2022/04/29
V3.0.73          5.06.001    	 5114              5.4.93        0.2.079      49-0E-42      2022/01/14
V3.0.69          5.06.001    	 5114              5.4.87        0.2.077      49-0E-41      2021/12/21
V3.0.59          5.05.009    	 5104              5.4.49        0.2.069      49-0E-40      2021/09/14
V3.0.57          5.05.008    	 5104              5.4.41        0.2.067      49-0E-40      2021/07/20								
V3.0.44          5.05.007  	 50F3              5.4.15        0.2.048      49-0E-37      2021/04/13
```

The 2 latest FW seems missing. @mrhpearson ?

mrhpearson, 2024-07-03:
Checking with the docking team

joakim-tjernlund, 2024-07-10:
> Checking with the docking team

Any progress? Really want the new FW as we see some problems here when replugging USB-C
while in SDDM login screen. The external monitor never comes back after that.
One needs to reboot to get the monitor back.

mrhpearson, 2024-07-10:
Yes.
V 0.91 was not uploaded to LVFS as there were issues with it - hence why it was skipped.
V 0.92 is in progress. Will be a couple of weeks before it is available (approx - I don't have a committed ETA). 
Nag me the week of the 29th July if nothing and I'll update on the status (with a caveat that I'm on vacation the week before so my inbox that week is going to be a special form of hell....)

Mark

mrhpearson, 2024-08-21:
I'll have to check with the docking team on the specifics

Just as a note, because I updated my TBT4 dock this morning and hit similar issues. In my case I:
 - power cycled the dock. The LED on it flashed for a few mins before going solid.
 - rebooted my laptop
After that everything seems recovered and working. I've forwarded logs to the docking team.

Can you also check the output from 'fwupdmgr get-devices' and see if the dock and it's sub-components are listed.

soutzis, 2024-08-23:
> I'll have to check with the docking team on the specifics
>
> Just as a note, because I updated my TBT4 dock this morning and hit similar issues. In my case I:
>
> * power cycled the dock. The LED on it [...]

Thank you for your input. I just don't understand what is going on, this is an issue that affects our whole organisation.

Please let me know what logs are required, and I will ask the System Administrators to extract these from a few computers.

### What actions I took
We mostly use new Lenovo Thinkpad laptops and the docking station is [this one](https://pcsupport.lenovo.com/cy/en/accessories/pd500519) (based on the S/N). I installed the latest drivers on my laptop, as well as the latest FW, according to the instructions. I re-installed the firmware after the first installation (kind of) bricked the docking station and after 2 attempts, the docking station started charging again and the USB hubs started working.

### What are the issues

- The docking station will reliably provide power to the monitors, but not to the USB hubs and to the ethernet port. It usually takes anywhere from 5 to 40 minutes until the rest of the ports are powered and functioning.
- Sometimes, if you fiddle with the docking stations by turning it off/on or connecting/disconnecting the laptop, it might stop charging (happened to me once, on the 21st of August 2024 - was upgraded to latest FW version).

mrhpearson, 2024-08-23:
Can you share the details of 
 - which platform you were on.
 - which distro you were using (and version)
 - which fwupdmgr version.

If you have any logs from the update that would be great - but I appreciate you might not have those any more.

We'll need to see if we can reproduce.
Thanks
Mark

mrhpearson, 2024-08-26:
Updated request from the dock team. Can we get these details:

1.     Dock FW version before updated.  
2.     Dock SN information on label
3.     External monitor model and connect method, HDMI or DP port?
4.     User NB model info 

Probably best to send me the details directly (not good to put SN on a public forum). Please forward to mpearson-lenovo at squebb dot ca.

Thanks
Mark

soutzis, 2024-08-26:
> Updated request from the dock team. Can we get these details:
>
> 1.     Dock FW version before updated.
> 2.     Dock SN information on label
> 3.     External monitor model and connect method, HDMI or [...]

Thanks for your responses Mark. My issue is with a Windows-based workstation and as you've kindly noted, this project is for Linux updates to the dock.
Will keep working on it.

ariasuni, 2025-02-19:
I couldn’t update my USB-C dock (40AY) from 0.0.3.6 to 3.0.9.6; when unplugging the dock after being told to do it, the led didn’t blink and after replugging the dock I got `device version not updated on success, 3.0.9.6 != 0.0.3.6`, and it soft-bricked all the USB type A ports except the one between the HDMI and the (rectangle) power port.

However, I could update from 0.0.3.6 to 0.0.4.6, which fixed the USB ports, and then to 3.0.9.6. I suppose fwupd should force to first update to 0.0.4.6, right?