fwupd/firmware-lenovo #239: Issue with Intel ME Update: Available for wrong device

arawaaa, 2022-06-27:
The "Lenovo Intel ME" update version 240.35.1951 is available for a device that is not a Thinkpad L13. The device in question is actually a Lenovo Yoga 9i 14ITL5 82BG.

ChiWei-Chen, 2022-06-28:
Thanks, we'll need FW team to look this.
(LO-1813 for @mrhpearson and my tracking)

ChiWei-Chen, 2022-07-05:
@arawaaa , the FW team changed the setting of FW on LVFS, could you let us know if the issue is still there on your machine?
If you could still see the issue, could you provide detailed steps on how you duplicated it? Thanks!

arawaaa, 2022-07-24:
The incorrect firmware is still present, and reproducible through "fwupdmgr get-updates"
[image]

ChiWei-Chen, 2022-07-26:
@arawaaa , thanks for the feedback. I'll give you back if any progress from FW team.

mrhpearson, 2022-07-29:
Hi @arawaaa - The FW team told me that the FW has been fixed. Could you check on your side and confirm please?
Thanks!

arawaaa, 2022-07-30:
Hello, there has been a change, but the behavior is still incorrect. The ME update, instead of being for the L13, is now the firmware for the X1 Titanium Gen 1.

mrhpearson, 2022-08-02:
I believe that's correct - it's not the right FW for the L13. You shouldn't see it being offered to you on the L13 any more - let me know if you're still seeing it (you'll need to refresh the fwupdmgr cache)

It looks like the ME FW for the L13 isn't on LVFS - at least not that I can find. I'll check with the FW team when they will be doing that.

ChiWei-Chen, 2022-08-03:
Hello @arawaaa , could you let us know what generation is the L13 system, is it Gen 1 or Gen 2?

Thanks, Mark.

mrhpearson, 2022-08-25:
L13 G1 ME FW is now on LVFS but under embargo. Will monitor for when it moves to stable.

arawaaa, 2022-09-24:
Thank you, the firmware update no longer appears for the Yoga 9i.

closed 2022-09-24 by arawaaa

kmauleon, 2022-11-18:
hi... fyr please see L13 links as below... thank you very much
Lenovo-ThinkPadL13-ConsumerMEFirmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR15RG.firmware
Lenovo-ThinkPadL13Gen1-CorporateMEFirmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR15MB.firmware