fwupd/firmware-lenovo #240: ThinkPad T480 - Unable to update Intel Management Engine on Pop!_OS 22.04

vancanhuit, 2022-06-30:
I updated the UEFI Device Firmware (via command line and firmware update manager UI on POP!_OS) after received a firmware notification update from Pop!_OS and reboot. After that, I still receive the firmware update notification again.

**Steps to reproduce**

- Run `sudo fwupdmgr refresh --force && sudo fwupdmgr update` or click update button in the firmware update manager UI on Pop!_OS.
- Reboot
- The firmware update status is ok but it is not actually not installed.

**Other system information**

```bash
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.0
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.17.15-76051715-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.0
runtime   org.freedesktop.gusb          0.3.10
```

```bash
$ fwupdmgr get-devices --show-all-devices
20L5CTO1WW
│
├─Core™ i7-8650U CPU @ 1.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               b87a926d-189c-49a1-b1ff-921099b1de89
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─GP108M [GeForce MX150] (ThinkPad T480):
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              7991065c-5075-5809-8414-df6c2a42b6a8 ← PCI\VEN_10DE&DEV_1D10
│                         90c099eb-40ec-5752-9656-8543a094b1a3 ← PCI\VEN_10DE&DEV_1D10&REV_A1
│                         93dac76a-8bac-5485-ad78-444b2849264d ← PCI\VEN_10DE&DEV_1D10&SUBSYS_17AA225E
│                         a6618e83-2ced-5c49-b384-6b309580f44a ← PCI\VEN_10DE&DEV_1D10&SUBSYS_17AA225E&REV_A1
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.90.3987
│     Minimum Version:    184.90.3987
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               dd31d983-cf3b-4c84-acc9-70caa94e827b
[... 137 lines cut ...]
Upload report now? (Requires internet connection) [Y|n]:
```

```bash
$ tree /boot
/boot
├── config-5.17.15-76051715-generic
├── config-5.17.5-76051705-generic
├── efi  [error opening dir]
├── initrd.img -> initrd.img-5.17.15-76051715-generic
├── initrd.img-5.17.15-76051715-generic
├── initrd.img-5.17.5-76051705-generic
├── initrd.img-5.3.0-7648-generic
├── initrd.img.old -> initrd.img-5.17.5-76051705-generic
├── System.map-5.17.15-76051715-generic
├── System.map-5.17.5-76051705-generic
├── vmlinuz -> vmlinuz-5.17.15-76051715-generic
├── vmlinuz-5.17.15-76051715-generic
├── vmlinuz-5.17.5-76051705-generic
└── vmlinuz.old -> vmlinuz-5.17.5-76051705-generic

1 directory, 13 files
```

It is quite similar to this issue #57.

mrhpearson, 2022-06-30:
@ChiWei-Chen - looks like this user has the BIOS fix and is still seeing the problem.

@vancanhuit - this was supposed to be fixed in BIOS 1.44. Did you do both updates at once? or do the ME update before the BIOS update by any chance? Wondering if this is a sequencing issue (need the BIOS update before the ME).

Mark

vancanhuit, 2022-06-30:
@mrhpearson Running ` sudo dmidecode -s bios-version` returns `N24ET69W (1.44 )` on my device.

LinuxOnTheDesktop, 2022-07-01:
Cf. also #122, though the reporter their somehow managed to fix their problem. And I too, on an X1C6 on Linux Mint, cannot update the Intel Management Engine.

ChiWei-Chen, 2022-07-01:
@vancanhuit , there is a situation is the ME updated successfully, but the BIOS didn't delete the old variable correctly. Could you let us know what's the ME Firmware version BIOS main page? (Press F1 in preboot stage) 
Thanks!

rotanid, 2022-07-01:
> @vancanhuit , there is a situation is the ME updated successfully, but the BIOS didn't delete the old variable correctly. Could you let us know what's the ME Firmware version BIOS main page? (Press F1 in preboot stage) Thanks!

i also have this issue, see https://github.com/fwupd/firmware-lenovo/issues/201#issuecomment-1172755381

vancanhuit, 2022-07-02:
@ChiWei-Chen Here is my BIOS main page:
[image]

ChiWei-Chen, 2022-07-05:
@vancanhuit thanks for sharing the information. This is very useful.
FW team provided following workaround to try.
please try the below steps to make sure the machine updatde the system firmware before MEFW(4222).

1. Flash to older MEFW (such as 184.90.3987)with local update
2. Update to the latest BIOS 69W(1.44): sudo fwupdmgr --allow-reinstall install <Name>.cab
3. Update to the latest MEFW 184.92.4222.

If you still get the same failure(fwupdmgr shows old version). Please drop me (cchen50 at lenovo dot com) a mail, I will share you the tool to dump the BIOS file. 
Thank you.

hughsie, 2022-07-05:
@ChiWei-Chen do we need to add a BIOS version requirement on the MEFW firmware?

ChiWei-Chen, 2022-07-05:
Richard, thanks, sounds good for me, I'll get FW team's comment on this advice.

vancanhuit, 2022-07-09:
@ChiWei-Chen After trying your workaround, I still get the same issue. I sent you an email for further support.

ChiWei-Chen, 2022-07-12:
@vancanhuit , thanks for the feedback.
I know the BIOS team is rolling out a fix with new BIOS version and got positive results on other platforms(T480s), I’ll keep you update if I got any progress for T480.

ChiWei-Chen, 2022-07-13:
@vancanhuit , the FW team needs some weeks to complete the internal release process, thanks.

librarianmage, 2022-08-03:
Is this still an issue?

iAmG-r00t, 2022-08-07:
Yep still an issue.

iAmG-r00t, 2022-08-07:
@ChiWei-Chen so I am having a similar issue, where from my Bios Menu it shows I have the following;

```
BIOS Ver:                     N24ET69W (1.44)
Embedded Controller Ver:      N24HT37W (1.22)
ME Firmware Ver:              11.8.92.4222
```

So I was trying a different approach where I wanted to downgrade the BIOS, then the ME. I was able to downgrade the BIOS but when downgrading the ME to version `184.90.3987` during reboot it gives me the below error;

```
Error 8758: The image provided is not supported by the platform
```

Downgrade command; `sudo fwupdmgr install --allow-older` (for both BIOS & ME)

The other route I have tried is downloading both the ME old and new version `.cab` files as you have advised [here](https://github.com/fwupd/firmware-lenovo/issues/208#issuecomment-1205974164) but now I can not install it gives me this error: `No supported devices found`

I hope this information might help.

ChiWei-Chen, 2022-08-09:
@iAmG-r00t , thanks for sharing the information, since T460s case is a little bit different from your T480 machine, please wait for a few weeks for the next BIOS relase :)

vancanhuit, 2022-08-21:
[image]

With the new BIOS update, I can confirm that the issue is gone on my T480 machine.

iAmG-r00t, 2022-08-21:
What did you install first, firmware then IME?

vancanhuit, 2022-08-21:
I just did install firmware first then the ME version will be automatically updated correctly.

rotanid, 2022-08-21:
can confirm.

iAmG-r00t, 2022-08-21:
Yep, worked on my end too.

iAmG-r00t, 2022-08-21:
You can now close the issue since it has been fixed. @vancanhuit

closed 2022-08-21 by vancanhuit

ullgren, 2022-08-21:
@vancanhuit Sorry for commenting on a closed issue but I see the same issue on a Lenovo ThinkPad P1. Will there be a fix for that model as well or should I open a new issue for this ?

LinuxOnTheDesktop, 2022-08-21:
> Sorry for commenting on a closed issue but I see the same issue on a Lenovo ThinkPad P1. Will there be a fix for that model as well  [. .]?

And what please about the X1CG6? For, that is another model that has suffered (for a long time now) with being unable to update its Intel firmware.

rotanid, 2022-08-21:
@ullgren @LinuxOnTheDesktop this issue was about T480, so if you have issues with other models, feel free to open an issue for your device. please check first, if there is already one, as there are some similar issues already for some other models like e.g. T480s and T470s