fwupd/firmware-lenovo #242: Unable to update Intel Menagement Engine on Thinkpad X1 (Gen 9)

ptrv, 2022-07-04:
I've tried to update a couple of firmawares with Fedora's Software applicatoin.

The update process stuck and I had to use the reset button to get out of that state. Luckily the machine booted again but now there are two devices than can't be updated.

```
 sudo fwupdmgr get-devices               
20XWCTO1WW
│
├─ELAN0672:00 04F3:3187:
│     Device ID:          0867c07f040e810fa5ea153981e63e324c29eb1f
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             HIDRAW:0x04F3
│     GUIDs:              ab16bce2-e219-52f5-89a6-fd6b1c4244c0 ← HIDRAW\VEN_04F3&DEV_3187
│                         940805ff-ca7a-56b2-af86-32e995004af9 ← HIDRAW\VEN_04F3&DEV_3187&REV_00
│                         e4c6161e-2b1c-59cb-ad07-c8fb7feb155e ← HIDRAW\VEN_04F3&DEV_3187&MOD_0002
│                         4b808d27-c833-50cc-8177-9a401f5c08fe ← ELANTP\ICTYPE_14
│                         5b73c06e-3dfd-5e6c-b06a-24e3489206c2 ← ELANTP\ICTYPE_14&MOD_0002
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Embedded Controller:
│     Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
│     Summary:            UEFI ESRT device
│     Current version:    0.1.32
│     Minimum Version:    0.1.32
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627
│     Current version:    62.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Serial Number:      0001
│     GUIDs:              30bb6882-12ea-5e20-84a6-29bb797a7b76 ← USB\VID_04F2&PID_B6EA
│                         2592097f-953e-50b8-904b-e07cdbac3bb4 ← USB\VID_04F2&PID_B6EA&REV_6218
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          a083ebc5138e5e071ef7270cc9a8280722cc7adf
│     Summary:            UEFI ESRT device
│     Current version:    240.23.1706
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               03c27e00-8229-4c14-ba40-2afe2276555c
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─MZVL21T0HCLR-00BL7:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    AL2QGXA7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S64PNX0R817034
│     GUIDs:              bec63ed7-a95f-54fe-b8cc-8e9fee64ba5a ← NVME\VEN_144D&DEV_A80A
│                         60c89aac-f321-515b-b419-3cf02aa9d375 ← NVME\VEN_144D&DEV_A80A&REV_00
│                         310f81b5-6fce-501e-acfb-487d10501e78 ← NVME\VEN_144D&DEV_A80A&SUBSYS_144DA801
[... 195 lines cut ...]
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               0d803ee9-f231-4ad7-9cb8-563bcbe75c13
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
________________________________________________

Devices that were not updated correctly:

 • UEFI Device Firmware (1.2.22.0 → 1.2.26.0)
 • Intel Management Engine (240.23.1706 → 240.35.1951)

Devices that have been updated successfully:

 • Prometheus (10.01.3273255 → 10.01.3478575)
 • Embedded Controller (0.1.31 → 0.1.32)
 • System Firmware (0.1.51 → 0.1.52)
```

I found the update on the Lenovo support website as file to download. Should I run the Intel ME update with downloaded firmware file?
Could it happen that I brick the machine with my attempt?

kmauleon, 2022-07-05:
hi @ptrv yes could you please try updating MEFW alone this time?  you may download the actual file in Lenovo Support Site
https://pcsupport.lenovo.com/ph/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-9th-gen-type-20xw-20xx/downloads/ds554201-intel-management-engine-150-firmware-comsumer-update-utility-for-linux-thinkpad-x1-yoga-6th-gen-x1-carbon-9th-gen?category=Motherboard%20Devices%20%28core%20chipset,%20onboard%20video,%20PCIe%20switches%29
please follow this procedure...  thank you very much
  1. Make sure the AC adapter is firmly connected to the target computer
  2. Extract package zip file
  3. Launch Terminal
  4. Login as root user
  5. Navigate to the directory where the cabinet file was placed
  6. Execute as below:

        fwupdmgr --allow-older --allow-reinstall install <cab_filename.cab>

  7. Restart the system manually if system did not restart automatically
  8. After restart, "Please wait while we install a system update" will be displayed
  9. Wait for the system update to complete then system will automatically restart

kmauleon, 2022-07-07:
please don't install together with other firmware especially with BIOS or EC and please make sure to disable secure boot and with at least more than 25% battery and AC connected... thank you very much

ptrv, 2022-07-08:
Tried to upgrade. I am getting this when running `fwupdmgr --allow-older --allow-reinstall install <cab_filename.cab>`

```
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer
```

Where can I get the signature?

kmauleon, 2022-07-12:
hi @ptrv please set OnlyTrusted=false in /etc/fwupd/daemon.conf ... and revert it back to previous setting after the update... thank you very much...

ptrv, 2022-07-12:
@kmauleon Thanks for your help! I have successfully updated MEFW. And after that, `fwupdmgr update` installed two more updates successfully.

closed 2022-07-12 by ptrv

kmauleon, 2022-07-13:
@ptrv great to know that!  thank you very much ^_^

mrhpearson, 2022-07-13:
@kmauleon - does this mean we should be updating the ME metadata so it doesn't update unless we're at a certain value of BIOS/EC?

kmauleon, 2022-07-14:
@mrhpearson for inhouse products I usually add restrictions for BIOS/EC if needed otherwise should be OK as it is