fwupd/firmware-lenovo #243: fwupd offering a camera update for the Thinkpad T14s while I'm on an X1 Gen 8

sukahiroaki, 2022-07-07:
This is a bit of a weird one. Today I got offered a new firmware update on my Thinkpad X1 Gen 8 (Fedora Silverblue 36) which calls itself "ThinkPad 14s Narrow Hybrid Camera Firmware" (version 0.0.24580) - which from the surface seems wrong. Or isn't it? If this is a genuine update, indeed, I'd say this naming is highly irritating for users - and also problematic from a security view point (cause you make people not care about such details)

mrhpearson, 2022-07-26:
Sorry for the slow reply - with vacation and conference travel I'm behind.
I believe the Camera FW team are fixing this - it was their first time delivering FW on LVFS and there were some mistakes.
Mark

sukahiroaki, 2022-07-28:
Yeah, that seems to be fixed now. I've received a proper / fitting update in the meantime. So: Closing, I guess