fwupd/firmware-lenovo #244: Lenovo T14 Gen2a (Ryzen) has flat version FW numbering

joakim-tjernlund, 2022-07-07:
product: 20XK007DMX (LENOVO_MT_20XK_BU_Think_FM_ThinkPad T14 Gen 2a)
fwupd: 1.8.1

There are no FW updates on LVFS and system FW  has flat version number:
```
├─System Firmware:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Summary:            UEFI ESRT device
│ │   Current version:    65553
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              8fb7e1f9-447b-41fb-8832-ae5ef955665f
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  267
│       Minimum Version:  267
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            1e9c3d61-4bd9-514e-ad87-0a9b61b9e77d ← UEFI\CRT_156645FC005F8617669DF76E1DE6FAE02B3856D24B624E47D97BD2B8ADFFB6E0
│                         304a9ede-aff2-58f6-b5e5-0e651b065f1a ← UEFI\CRT_156645FC005F8617669DF76E1DE6FAE02B3856D24B624E47D97BD2B8ADFFB6E0&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
```

joakim-tjernlund, 2022-07-07:
Also, I think this FW, https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1MBL_00060108.firmware is
for this computer but fwupdmgr install *.cab says No supported devices found

mrhpearson, 2022-07-26:
Sorry for the slow reply - between holiday and conference travel I'm way behind on my inbox.

I found https://fwupd.org/downloads/31d0f02dd6e624ef774495714615f399ef20a34a00318944a236cf365fae7f35-r1mul47w.cab for your system.
The FW is marked as triplet format in the metainfo:
    <value key="LVFS::VersionFormat">triplet</value>
So not sure why you're not seeing that in your display. Are you using cached FW or something like that?

That FW should be for your system, but it's for the battery FW. Not sure why it's not showing up, I'll see if I can find out - it's possible you might need the latest fwupd.

joakim-tjernlund, 2022-09-01:
I have System Firmware: 0.1.19 for a while now but my battery still malfunctions.
If I let the battery drain completely and put in the dock and start it, then battery status in Linux is discharging and will turn off the laptop in about a minute. Then it will stay off, not charging.

Then I have to toggle power to the dock, let laptop stay off for c.a 10 min. Now I can turn laptop on and battery is finally charging.