fwupd/firmware-lenovo #246: Lenovo TB3 dock and TB4 dock major issues

ExecuteRestart66, 2022-07-25:
so here is what i have- and it happens across basically all of them. 

T14s Gen 2
T15 gen 2
TB3 dock
TB4 dock
Ultra dock

setup:
all laptops are hooked into either a TB3 or TB4 dock with 2 external monitors. (mainly TB4). our monitors do not have DisplayPort. So we end up using a hdmi>hdmi cable and a DP>HDMI cable. but i have even tried using 2 DP>HDMI cables. 

Issues:
1. Screen flicker- i describe this as 'shaking'. it almost looks like you bumped your monitor but the monitor is standing still while the screen itself is shaking. I noticed that this can happen on either monitor regardless of the cable you are using. 
- When i end up doing a dock firmware update this goes away. which i still dont understand. 
-This usually happens before issue #2 

2. monitor off/on- one monitor will go off and on. at random times, with no obvious cause. I have noticed this almost ALWAYS happens to the monitor that is using HDMI>HDMI. If you use 2 DP>HDMI then its just the luck of the draw. 
-this usually starts happening after the dock firmware update. I don't often see both #1 and #2 combined. maybe i wasn't looking. 

3. Monitor goes off but does not come back on until you either 1. unplug/replug dock power or 2. unplug/replug monitor into dock. 
- i mainly see this on the ultras. But the ultras also suffer #1 and #2 which was a big let down because I was really hoping this was some firmware issue with the TB docks. 

what i have tried ? .... EVERYTHING. i have spent hours reading over every post, every forum, ever service announcement. nothing works. when i first started down this path, I was convinced this was due to the GPU drivers Lenovo was pushing out. but after trying old GPU drivers (almost all of them), and Intel OEM DCH drivers- I still have all these issues. 

what happens when you plug in 2 monitors direct to the laptop ? WORKS PERFECT. so this tells me it cant be the GPU. 

What happens when you plug in a Dell TB4 dock ? WORKS PERFECT. this tells me this has to be either an issue with the dock itself, or the TB controller on these laptops. 

Lenovo has been totally silent on this even though its all over their forums, reddit, etc. I have never experienced this behavior in my entire career, and its super frustrating that it seems this is 100% a lenovo issue- but there is no information on it. What kind of business line laptops/docks cant run 2 monitors ? regardless of those monitors age/ connection type. 

any help is super appreciated.

roliverio, 2022-08-04:
I also have had this issue from the day that i bought the dock (early 2020), and it's just as you described. I can certify that it is the dock), however, i think that this is not the correct place to report it, unless Lenovo is really listening to these issues.

Recently a new version for the dock's firmware was released, would be nice if you can update it (as i don't have a windows machine to test this out, and the dock update hasn't hit fwupd's repo) and let me know.

j5k, 2022-08-04:
I have the very same issue and will test the new firmware as soon as possible.

vrumjantsev, 2022-11-26:
I've ran into same issue with ThinkPad Thunderbolt 4 Workstation Dock, both of external monitors are not working at the same time, via dp+dp or dp+hdmi. 

Each of them separately works fine

FlyingCurryMonster, 2023-09-23:
Using a brand new thinkpad x1 extreme gen5 with 2 thunderbolt 4 thinkpad workstation docks.

I have mysterious dock issues as well but nothing to do with the displays.  I have two seperate thunderbolt 4 docks.  It seems when I plug in either of my docks to my computer, I lose bluetooth connectivity (despite it saying I'm connected to my bluetooth mouse,) and sometimes firefox installed from snap won't load anything new, and be impossible to exit out of.  Terminal commands won't even go through.  

These issues go away if I forcefully restart my computer with the dock plugged in before it restarts.  But the issues also reappear when I switch over to my second dock, and become forced to restart my computer again with the second dock plugged in.  I also occasionally find that my computer freezes after suspending (with the dock sucessfully disconnected prior to the suspend command)

I don't think I have anything I can provide from journalctl.  It hasn't given me any indication of what's going wrong with the machine.   Apologies for the  vagueness, but I'm really clueless as to what's even going on and have no idea where to start diagnsosing.  I figured an update to dock fimware 10.16 on ubuntu through fwupdmgr would be nice.  I think only 10.14 is supported right now on fwupd.

roliverio, 2023-09-26:
In my case, using the TB3 Gen 2 Workstation dock on an X1E Gen2 I've discovered that a combination of NVIDIA's issues and the dock itself were causing the issue. I was constantly suspicious of the dock itself but it has been rock solid avoiding a newer version of NVIDIA's driver (I'm currently on 525.85), the driver cannot be updated until NVIDIA solves this: https://forums.developer.nvidia.com/t/525-89-brings-back-thunderbolt-3-connected-displays-flicker-and-suspend-issues/245530.

OTOH, upgrading the kernel also helps with device enumeration on the dock, and using a recent version of fwupd also helps applying updates to the machine's firmware.

From what i read about the changelog in TB4 docks, this one has a lot of display related issues, that only seem partially fixed.

In your case the first thing i would try is upgrading the kernel. If you're using the propietary nvidia driver, take into account that the best right now is 525.85, anything else will make your displays on the dock flicker, also, as a tip, getting rid of firefox as a snap is a sane recommendation, and, i wouldn't dare to run any DE under Wayland using NVIDIA drivers, it's just unusable.

FlyingCurryMonster, 2023-09-26:
Many thanks for your reply @roliverio  I tried reverting from 535 to 525 on a similar suspicion from past experience with nvidia drivers.  I still find system instability with the 525, but I'm not sure if it's the same problems that persist or new ones.  Maybe I should try running with 525 for a couple days.  

I believe I'm on Linux 5.17.0-1035-oem, and I seem up to date with fwupdmgr.  There is a weird issue of not being able to update to Intel ME 257.27.2176 on my thinkpad, but I'm on version "1.27.2176" which is up to date according to Lenovo, there's just an error in the naming.  

I installed firefox from the a ppa, afer uninstalling from snap.  The snap version of firefox would cause my system to freeze, terminal commands wouldn't go through,  all while docked.  It was a nightmare.  Pretty sure I'm on x11 and not wayland.

roliverio, 2023-09-26:
@FlyingCurryMonster I'm currently on kernel 6.5. 

Intel ME's firmware will not update if the lid is closed, or if the Laptop is connected to a dock.

FlyingCurryMonster, 2023-09-28:
Seems like I'm on Linux 5.17.0-1035-oem.  I tried using mainline to update to the latest kernel (6.5.5.), but I couldn't connect an external display with HDMI (nor the displays on the dock.)    Nvidia-smi would indicated I couldn't communicate with the nvidia drivers on kernel 6.5.5.  

I tried updating to ubuntu 23.04 for an up to date kernel, but accidentally hit ctrl+c during the update and my system became unstable.  I wasn't able to connect to wifi and ended up reinstalling ubuntu 22.04 LTS.  After the reinstall the system hasn't frozen yet.  I found out by default, upon logging in ubuntu prefers wayland, so I manually chose xorg.  I'm using nvidia-driver 535.  Hoping the problem goes away with xorg.

roliverio, 2023-10-09:
@FlyingCurryMonster i don't know if the ME firmware is _device specific_ i.e. a Thinkpad T15 vs another model. However, i did struggle a little with it until after some fwupd in Debian (trixie) allowed me to install it with the lid open.

To go to kernel >=6.5 i've installed xanmod, so that's what i'm using since last year (going back to 6.2, 6.3, ... etc) however, building 525.85 drivers on 6.5 (6.4 too) requires patching and it's cumbersome. However, given the benefits of running a recent kernel, i wouldn't dare to stay on an old version.

So there are two big issues right now that seem to affect you:
1) The dock itself
2) NVIDIA drivers

For the dock, unfortunately, you rely 100% on Lenovo, and even if @mrhpearson does indeed care about our struggles (i can vouch from his involvement both here and the Lenovo Forums) he and his team can only do so much.

For the drivers, we are locked about NVIDIA and they shitty release process and compatibility. If you want to _cleanly_ update the kernel to a recent version, we'll need to wait for a decent recent release that address the display flicker (and builds without patching on recent kernels), as it seems closely tied with power management on the cards. The dock(s) are oblivious to this as it is 100% machine dependent and much more when the machine is an optimus laptop. (as it has locked outputs for the dedicated display card)

mrhpearson, 2023-10-10:
Don't think I'd seen this thread before (which is odd...I try and read all the ones in the firmware-lenovo forum). Will try and tackle some of the points but apologies for what I miss...

Two 4k monitors and those combos. If using Intel for display then it's a known issue and we've been working with Intel on it. It's related to the lack of DSC feature on Linux, but they are adding it. Patches aren't landed yet (and it's not perfect) but there's been a bunch of progress and hopefully we'll see this one fixed 'soon'.

Freezing when connected to the dock - would need to know kernel version, FW version and if possible kernel log. Definitely recommend newer kernels (fixes have gone in), newer Nvidia drivers (ditto) and updating dock FW to the latest (it should be on LVFS).

Bluetooth shouldn't be impacted by the dock... that's something else

And, not mentioned, but an honourable mention for anybody seeing a wired ethernet port issue. It's related to vPro being enabled, and if you update the Realtek FW and use their out of tree driver (I know...urgh) then it goes away. We're still digging into the root of that one.

roliverio, 2023-10-10:
@mrhpearson Thanks.

Regarding the ethernet port issue, i've never had it (and i've always used the in-kernel driver)

DSC should affect non-optimus laptops, if i'm following your discourse well, but the display flicker and freeze with optimus ones is related to NVIDIA 100%. (for TB3 Dock Gen2, releases 3.1.66 up until 3.1.80 there was indeed an issue with the DP hub which was independent to NVIDIA drivers at that time) 

Could you also tell us in this thread when will newer firmwares will roll-out for docks TB3 and TB4 alike? 

Thanks again.

mrhpearson, 2023-11-01:
Sorry - completely missed this in my email swamp...

I think the TB4 dock latest FW is already on LVFS - you do have to have the newer version of fwupd for it to work though. I'm not sure of the status on TB3

Also - Intel has DSC fully working now. Tested by my colleague with two 4k monitors at 60Hz and working well. Patches are not upstream yet but should be 'soon'.

roliverio, 2023-11-01:
@mrhpearson Thanks, can you dig around for updates on the TB3 docks?.

LamJorDANG, 2024-01-04:
Hello All,

Is there any update to the Monitor "flicker" issue with the TB3/TB4 docks?  We have a bunch of them, and they are always having this issue with multiple monitors connected.

mrhpearson, 2024-01-09:
What monitors do you have connected (resolution and frequency) and how? I have some notes on workarounds and fixes but want to check which one applies to your situation. Are you on an Intel or AMD system.
As quick pointers
 - If multiple 4k monitors, you'll want the 6.7 kernel where Intel have added DSC support which helps a lot. In the meantime you can often improve things by lowering the refresh frequency
 - If daisy chaining then need to disable some features as MST doesn't play nice with them. Let me know if that applies and I'll dig up notes.

roliverio, 2024-01-09:
@mrhpearson 

Can you update us on ETA for new FW updates for TB3 Dock gen2?, as i've said earlier, the monitor flicker issue with NVIDIA cards on optimus laptops is dealt with on 525.85.12, but any other driver version after that re-introduces the issue (i'm positive that is 100% NVIDIA's fault, but the dock still lacks improvements with the DP hub which have not been addressed)

On a side note: I'm facing the ethernet port issue on 6.6 (wasn't there on <6.6), however, turning off scatter-gatter on the realtek card on the dock mitigates it:

`ethtool -K $iface sg off`

Thanks for any new information.

LamJorDANG, 2024-01-09:
> What monitors do you have connected (resolution and frequency) and how? I have some notes on workarounds and fixes but want to check which one applies to your situation. Are you on an Intel or AMD [...]

Hello and thank you for the reply!

We are using HP monitors plugged in via DP.  In the recent encounter with this issue, more specifically, it was 2x DP HP Z23 1080P monitors on a TB4 Workstation dock (40B0), and that's it.  The user was using a P15 Gen2, with additional power plugged into the laptop, and Dock on TB port next to it (not the dock powering the laptop).  Both devices fully updated on Lenovo System update as of Dec. 1st.

mrhpearson, 2024-01-09:
@LamJorDANG: OK - the two 4k monitors will be doing it. 
If you can try with the 6.7 kernel that has DSC implemented (Display Stream Compression) and it should help a lot.

@roliverio - the realtek issue is being worked on actively. We were expecting a FW fix but it looks like it will now be a kernel patch. I don't have an ETA but they're being nagged weekly.
For the flicker - that is odd and we would have to investigate.

Which brings me to: This ticket is a bit broad and these kind of bugs don't really belong on the fwupd issues list as they are more OS related than FW. It makes it really hard to track specific issues when an issue is acting as a catch-all.

Would you mind posting on the lenovo forum (forums.lenovo.com/linux) and I can pick them up there and create an internal ticket. I'd like to close this thread.

Thanks
Mark

roliverio, 2024-01-09:
@mrhpearson sure, sorry for cross-posting here. I did share a lot of this in the lenovo forums a while back. in fact, the last post on that thread is mine: [Lenovo Forums - Ubuntu Monitors black](https://forums.lenovo.com/t5/Ubuntu/Thinkpad-P1-Gen2-amp-Thunderbolt-3-Workstation-Dock-amp-Linux-Ubuntu-Monitor-s-black/m-p/4575709?page=9)

I'll gladly open a new thread there if you feel that is better not resurrecting that thread. So you can close this issue.

mrhpearson, 2024-01-09:
That thread is old - I must have missed it :) 

I know this is annoying - but let's create a new one and please put in details of kernel, BIOS, EC, dock FW and the recreate issues and I'll ask the team to reproduce. It's strange as the P1G2 is older now and I have not had many issues reported against it....but we can check it out.

LamJorDANG, 2024-01-09:
> @LamJorDANG: OK - the two 4k monitors will be doing it. If you can try with the 6.7 kernel that has DSC implemented (Display Stream Compression) and it should help a lot.
>
> @roliverio - the realtek [...]

They are just 1080P monitors.  Should I still try it?  The monitors flicker until I turn off one of them, and one works fine.  But, once I turn on the next one the flickering happens ever 10secs.

mrhpearson, 2024-01-10:
My bad - two 1080P monitors should be fine. Can you create a ticket against intel DRM - notes on how to do that are here: https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs

Also please create a new thread on the Lenovo forum (if you haven't already - I haven't had time to parse the forum yet today) and then I can create an internal ticket and we can see if we can reproduce and do some debug too.

For the Realtek issue - it's been narrowed down to a clock in the device being incorrectly enabled. They're working on the patch to fix it.