fwupd/firmware-lenovo #248: Intel Management Engine fails on X1 Extreme Gen 1

tgelter, 2022-07-27:
Related to #140, I'm unable to upgrade the Intel Management Engine on my X1 Extreme Gen 1.
In https://github.com/fwupd/firmware-lenovo/issues/140#issuecomment-1196419255 I was asked to open a new issue with some requested details.

- fwupdmgr update --show-all
```
❯ fwupdmgr update --show-all
Devices with no available firmware updates: 
 • USB2.0 Hub
 • USB2.0 Hub
 • USB3.1 Hub
 • USB3.1 Hub
 • 0000:00:1f.5
 • ThinkPad Thunderbolt 3 Dock
 • ThinkPad Thunderbolt 3 Dock USB Audio
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 192.72.1757 to 192.81.1753?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 0 Intel Platform Update 2021.1 (PV Maintenance Release)                    ║
║                                                                              ║
║ Version 12.0.81.1753 (LVFS: 192.81.1753)                                     ║
║                                                                              ║
║ Problem fixes                                                                ║
║                                                                              ║
║ • Several Intel known issues were fixed in this release. Refer to Intel      ║
║ • Closed Issues at the bottom portion of Intel's release note                ║
║                                                                              ║
║ 20MF000MUS must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: n
Request canceled
```
- system firmware version and Intel Management Engine firmware version in BIOS main page
```
❯ sudo dmidecode -s bios-version  
N2EET58W (1.40 )
```
```
❯ ./intel_csme_version_detection_tool 
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2022, Intel Corporation, All rights reserved.

Application Version: 7.0.2.0
Scan date: 2022-07-27 19:34:43 GMT

*** Host Computer Information ***
Name: x1e
Manufacturer: LENOVO
Model: 20MF000MUS
Processor Name: Intel(R) Core(TM) i7-8850H CPU @ 2.60GHz
OS Version: Arch Linux (5.18.14-arch1-1)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 12.0.81.1753

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is vulnerable.
Explanation:
  The detected version of the Intel(R) Converged Security and Management Engine firmware
  has a vulnerability listed in one or more of the public Security Advisories.
  Contact your system manufacturer for support and remediation of this system.

For more information refer to the Intel(R) CSME Version Detection Tool User Guide
or the related Intel Security Advisory list at:
https://www.intel.com/content/www/us/en/support/articles/000031784/technologies.html
```
- OS and fwupdmgr version
```
❯ cat /etc/os-release 
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux-logo
```
```
❯ fwupdmgr --version        
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.14-arch1-1
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10
```
Please let me know if any details are missing, or if I can provide any other useful information.

pglpm, 2022-07-31:
Something similar is also happening to me: X1E4 with Ubuntu 20.04LTS:

```
$ fwupdmgr update 
Place your finger on the fingerprint reader
Devices with no available firmware updates: 
 • ELAN067A:00 04F3:3197
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 0.41.2142 to 240.41.2142?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad P1 Gen 4 Consumer ME Firmware This stable release fixes the  ║
║ following issues: - Update IOM FW to 24.31.0.0000 - Update TBT FW to         ║
║ 15.0.0.0045 - Update NPHY FW to 15.107.135.5017 - Update PMC FW to           ║
║ 150.2.10.1020                                                                ║
║                                                                              ║
║ 20Y5CTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
```
and
```
pglpm@nino:~/articles$ fwupdmgr get-history     
20Y5CTO1WW
[...]
└─Intel Management Engine:
  │   Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
  │   Previous version:   0.41.2142
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2022-07-30 21:15
  │   GUID:               dd735001-0644-4f83-8a15-a600a7d61a1c
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad P1 Gen 4 Consumer :
        New version:      240.41.2142
        Remote ID:        lvfs
        Release ID:       14643
        Summary:          Lenovo ThinkPad P1 Gen 4 Consumer ME Firmware
        License:          Proprietary
        Size:             3.9 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N40RN14W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad P1 Gen 4 Consumer ME Firmware This stable release fixes the following issues: - Update IOM FW to 24.31.0.0000 - Update TBT FW to 15.0.0.0045 - Update NPHY FW to 15.107.135.5017 - Update PMC FW to 150.2.10.1020
```

Let me know if anyone needs more info.

mark-beeby, 2022-08-01:
Encountering the same...

```
$ fwupdmgr update --show-all
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 0.41.2142 to 240.41.2142?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad P1 Gen 4 Corporate ME Firmware                               ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Update IOM FW to 24.31.0.0000                                              ║
║ • Update TBT FW to 15.0.0.0045                                               ║
║ • Update NPHY FW to 15.107.135.5017                                          ║
║ • Update PMC FW to 150.2.10.1020                                             ║
║                                                                              ║
║ 20Y50024UK must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
```
$ fwupdmgr get-history 
20Y50024UK
~~<REMOVED OTHER DEVICES FOR BREVITY>~~
└─Intel Management Engine:
  │   Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
  │   Previous version:   0.41.2142
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2022-08-01 07:34
  │   GUID:               76ac010a-2b04-497a-807f-72f831d94530
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad P1 Gen 4:
        New version:      240.41.2142
        Remote ID:        lvfs
        Release ID:       14420
        Summary:          Lenovo ThinkPad P1 Gen 4 Corporate ME Firmware
        Licence:          Proprietary
        Size:             10.6 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N40RM14W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad P1 Gen 4 Corporate ME Firmware
        
        This stable release fixes the following issues:
        
        • Update IOM FW to 24.31.0.0000
        • Update TBT FW to 15.0.0.0045
        • Update NPHY FW to 15.107.135.5017
        • Update PMC FW to 150.2.10.1020
```

```
$ cat /etc/os-release 
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os
```

Getting real tired of the lack of firmware professionalism and testing from Lenovo, my Thunderbolt 4 Docking Station doesn't work and locks up the machine due to firmware failures (separate ticket on this board) and now this. I think this might have to be my last Lenovo after following the X1 line for years...

mark-beeby, 2022-08-01:
```
Upgrade Intel Management Engine from 0.41.2142 to 240.41.2142
```

This suggests there might not be a real update at all (and the major version 240 is a typo).

BaronKrause, 2022-08-01:
Same issue on a Lenovo Thinkpad P1 Gen4 (same hardware as X1 Extreme Gen4). Have the newest Lenovo ME 0.41.2142 installed and it keeps trying to tell me to update to a bogus 240.41.2142.

ChiWei-Chen, 2022-08-04:
@tgelter , thanks, I'll give you back if I got any progress from FW team. (LO-1920 for my internal tracking)

@BaronKrause , @mark-beeby , @pglpm , this seems resolved, please check this [comment](https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1205206747). Let me know if this doesn't work for you.

tgelter, 2022-08-26:
> @tgelter , thanks, I'll give you back if I got any progress from FW team. (LO-1920 for my internal tracking)
>
> @BaronKrause , @mark-beeby , @pglpm , this seems resolved, please check this [...]

Hello again @ChiWei-Chen, I see that [updated BIOS releases](https://github.com/fwupd/firmware-lenovo/issues/140#issuecomment-1228087043) are out for other models. Would you mind updating us regarding progress with `LO-1920`?

Thank you!

ChiWei-Chen, 2022-08-30:
@tgelter , FW team already merge the fix in next BIOS release which is in queued. 
I'll keep you updated if there is an ETA available.

tgelter, 2022-08-30:
> @tgelter , FW team already merge the fix in next BIOS release which is in queued. I'll keep you updated if there is an ETA available.

Thank you very much @ChiWei-Chen for the update and for taking communications lead responsibility for this issue across several different models!

kmauleon, 2022-12-13:
hi All... latest BIOS with ESRT fix has been released... please update your systems.  thank you very much.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2EET.firmware