fwupd/firmware-lenovo #250: P15 gen2. Failed ME firmware. No sound on Ubuntu (problems also on Windows 10 )

kleops9, 2022-08-01:
I recently performed a firmware update from my ubuntu 22.04 on Lenovo P15 Gen2.

The update looks like didn't went through and now whenever I reboot the system, I get into a black update screen which always fails. Showing messages:

" Reading ME Firmware...Please do not power off! 48% Complete"
" Failed to recover ME Firmware..."

It seems similar to  [#226](https://github.com/fwupd/firmware-lenovo/issues/226) but my model is different. I tried to update bios, downgrade bios, from windows from bootable disk, from linux with no luck.

I always end up at the ME Firmware screen. I am out of ideas and it seems that sound is not working any more. 

Quite annoying tbh.

Here some extra info
```
$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • Embedded Controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Intel Management Engine
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 0.41.2142 to 240.41.2142?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad P1 Gen 4 Corporate ME Firmware                               ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Update IOM FW to 24.31.0.0000                                              ║
║ • Update TBT FW to 15.0.0.0045                                               ║
║ • Update NPHY FW to 15.107.135.5017                                          ║
║ • Update PMC FW to 150.2.10.1020                                             ║
║                                                                              ║
║ 20YQ000YMH must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: n
Request canceled
```
```
$ sudo dmidecode -s bios-version
N37ET39W (1.20 )
```
```
$ python intel_csme_version_detection_tool.py 
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2022, Intel Corporation, All rights reserved.

Application Version: 7.0.2.0
Scan date: 2022-08-01 14:19:39 GMT

*** Host Computer Information ***
Name: lenovo-p15-g2
Manufacturer: LENOVO
Model: 20YQ000YMH
Processor Name: 11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz
OS Version: Ubuntu 22.04.1 LTS (5.15.0-43-generic)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 15.0.41.2142

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.

For more information refer to the Intel(R) CSME Version Detection Tool User Guide
or the related Intel Security Advisory list at:
https://www.intel.com/content/www/us/en/support/articles/000031784/technologies.html

```
Bios reports same versions

kleops9, 2022-08-01:
After some attempts of installing firmware I managed to fix the ME firmware fail screen. Sound is also back in ubuntu but now the reported version is incorrect. Something is definitely wrong, but still no idea what and how to avoid it next time.

```
klearchos@klearchos-lenovo-p15-g2:~$ sudo fwupdmgr update
Place your right index finger on the fingerprint reader
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 0 to 240.41.2142?                       ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad P1 Gen 4 Corporate ME Firmware                               ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Update IOM FW to 24.31.0.0000                                              ║
║ • Update TBT FW to 15.0.0.0045                                               ║
║ • Update NPHY FW to 15.107.135.5017                                          ║
║ • Update PMC FW to 150.2.10.1020                                             ║
║                                                                              ║
║ 20YQ000YMH must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: n
Request canceled

```

```
 
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2022, Intel Corporation, All rights reserved.

Application Version: 7.0.2.0
Scan date: 2022-08-01 15:14:13 GMT

*** Host Computer Information ***
Name: lenovo-p15-g2
Manufacturer: LENOVO
Model: 20YQ000YMH
Processor Name: 11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz
OS Version: Ubuntu 22.04.1 LTS (5.15.0-43-generic)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 15.0.41.2142

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.

For more information refer to the Intel(R) CSME Version Detection Tool User Guide
or the related Intel Security Advisory list at:
https://www.intel.com/content/www/us/en/support/articles/000031784/technologies.html

```

WwkChina, 2022-08-03:
P1 Gen4 have same problems.

fabiendespinoy, 2022-08-03:
Exactly the same problem with the same config as described by @kleops9 but I did not manage to fix the screen.
For the moment I have a black screen, fan running and no information about what is going on.
Is there a way to fix it ?

kleops9, 2022-08-03:
@fabiendespinoy you can try the reset hole to make laptop boot
[image](https://user-images.githubusercontent.com/20028400/182574501-459daf88-eb7a-4564-9694-16b8ac416e6e.png)
If you can boot the laptop then you can try to force firmware update to the version found in the lenovo support page for your model.

fabiendespinoy, 2022-08-03:
Thanks to @kleops9 I managed to reboot the heating brick after trying an automatic update that did not work (black screen for about 20 mins and no response from pressing power button to +60sec.) by pressing the little button at the back of the laptop.

Just to help people who may encounter this issue later, here is the process I followed to recover the firmware:

Ubuntu 22.04 LTS with kernel:
```
$ uname -a
Linux XXXXXXXX 5.15.0-43-generic #46-Ubuntu SMP Tue Jul 12 10:30:17 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```
Version of the P15 Gen 2 and original firmware version:
```
$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 0.41.2142 to 240.41.2142?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad P1 Gen 4 Corporate ME Firmware                               ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Update IOM FW to 24.31.0.0000                                              ║
║ • Update TBT FW to 15.0.0.0045                                               ║
║ • Update NPHY FW to 15.107.135.5017                                          ║
║ • Update PMC FW to 150.2.10.1020                                             ║
║                                                                              ║
║ 20YRS1PV01 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: n
```

Since I have a corporate version (Intel Core vPRO i9) then I went to this page:
https://pcsupport.lenovo.com/us/fr/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p15-gen-2-type-20yq-20yr/20yr/20yrs1pv01/downloads/driver-list/component?name=Chipset

And took the version:
```
Intel Management Engine 15.0 Firmware (Corporate) Update Utility for Linux - ThinkPad
```

However you have two versions: ```14 Jun 2022``` | ```25 Jul 2022```
Please download the first one (old version from June).

Open the file ```/etc/fwupd/daemon.conf``` as root and modify line 50 to:
```
OnlyTrusted=false
```

Then extract the content of the archive you downloaded and run the following command as root:
```
fwupdmgr --allow-older --allow-reinstall install Lenovo-ThinkPad-P15P17T15gGen2-CorporateMEFirmware-15.0.41.2142.cab
```

Finally, the computer will restart, try again to update the ME firmware until 48% and then fail.
After, you will be blocked for about 60 seconds on the screen ```Failed to recover ME firmware..." and wait for the computer to reboot automatically.

On my side, no firmware update at startup (so no error message) and sound is back to normal!

mrhpearson, 2022-08-03:
@ChiWei-Chen - can you flag this to the FW team please. I know we've been releasing fixes for the ME version number issue but it's a bit worrying if they're going to be hitting issues like this.

Oddly I'm not being offered the update on my P15 G2....not sure why and will have to look into that

WwkChina, 2022-08-04:
All firmware update problems has solved for me (ThinkPad P1 Gen 4) . New version firmware had been released just now.

ChiWei-Chen, 2022-08-04:
@mrhpearson , thanks, this looks P1 Gen 4 MEFW configuration on LVFS causing this problem, I'll ask the FW team to check this. 
Raised LO-1914 for internal tracking.

ChiWei-Chen, 2022-08-04:
Hi,
It seems the P1 Gen4 FW team uploaded the firmware by mistake, it should be removed now.
Please let me know if you still seeing this problem. Thanks!

WwkChina, 2022-08-04:
> Hi, It seems the P1 Gen4 FW team uploaded the firmware by mistake, it should be removed now. Please let me know if you still seeing this problem. Thanks!

240.41.2142 had been removed which has been replaced by 0.1.16 and 0.1.17. I have successfully upgraded.

devopshndz, 2022-08-25:
Hi, P1 Gen4 has same problems. :(

devopshndz, 2022-08-25:
Hi, P15s has same problems :(

ChiWei-Chen, 2022-08-26:
Hi @kleops9 , could you let us know how this work on the P15 Gen2 machine? Thanks!

@devopshndz , thanks for the comment, could you provide more details, the output of `sudo fwupdmgr update`  and `$ sudo dmidecode -s bios-version` on the P15s machine? Maybe we can have a separate ticket? Thanks!

mmege, 2022-08-27:
Hi @ChiWei-Chen , same problem here with my P15.
`sudo fwupmgr update` prints
```Place your finger on the fingerprint reader
Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • MZVLB512HBJQ-000L7
 • Prometheus
 • Prometheus IOTA Config
 • System Firmware
 • UEFI dbx
```
And `dmidecode -s bios-version` prints:
```N34ET52W (1.52 )```

Hope this will be helpfull !

ChiWei-Chen, 2022-08-31:
@mmege , thanks, it seems your system is P15s Gen 2, did you still see any unexpected firmware proposed by `sudo fwupmgr update ` currently?

devopshndz, 2022-08-31:
Hi @ChiWei-Chen Sorry for not answering before :( Apologies. 
`$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • UEFI dbx
No updatable devices
$ sudo dmidecode -s bios-version
N34ET46W (1.46 )

`

mmege, 2022-08-31:
> @mmege , thanks, it seems your system is P15s Gen 2, did you still see any unexpected firmware proposed by `sudo fwupmgr update ` currently?

Hi @ChiWei-Chen ... Yes it is a P15s Gen2. Sorry, I should mentionned that. Here what I have now:
```
20:13 $ sudo fwupdmgr update
Place your finger on the fingerprint reader
Devices with no available firmware updates: 
 • Integrated Camera
Devices with the latest available firmware version:
 • Prometheus
 • Prometheus IOTA Config
 • UEFI dbx
```

ChiWei-Chen, 2022-09-01:
@devopshndz , @mmege , it seems your P15s Gen2 looks good, there is no unexpected FW proposed. Please correct me if I misunderstood. Thanks for the comment.

mmege, 2022-09-01:
@ChiWei-Chen, I still have the problem "Failed ME firmware etc..." at the very start of boot and no sound (speaker nor microhpone).

About 2 minutes after pressing the power button, the screen with the message "Failed ME firmware" disappears htne I am finally able to log in. But after that I don't have sound devices working.

ChiWei-Chen, 2022-09-01:
@mmege , thanks for the clarification, could you provide the output of `fwupmgr get-devices` and kernel version of your Linux?
Thanks!

ChiWei-Chen, 2022-09-02:
@mmege , thanks, I assume this seems the same as this issue.
https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1203769040

Could you refer to the instruction above to download [Lenovo-ThinkPad-T14T15P14sP15sGen2-Consumer-MEFirmware-15.0.41.2142.cab](https://fwupd.org/downloads/52f06e11cd7114e9cb9af950f7189aa50c70b0828ae5070393733d44bb48d077-Lenovo-ThinkPad-T14T15P14sP15sGen2-Consumer-MEFirmware-15.0.41.2142.cab)
And reinstall it with the following command?
`$ fwupdmgr --allow-older --allow-reinstall install Lenovo-ThinkPad-T14T15P14sP15sGen2-Consumer-MEFirmware-15.0.41.2142.cab`

This may reboot automatically multiple times.
Please let me know if this works for you.
Thanks!

adityamukho, 2022-09-06:
Any updates on this issue? I'm stuck with the same issue for days with no resolution.

ChiWei-Chen, 2022-09-06:
@devopshndz , @adityamukho , would you be able to try reinstall the ME firmware with my comment above and let me know how it works? Thanks

mmege, 2022-09-06:
Hi @ChiWei-Chen ! Sorry for the delay.
I did what you mentioned and It has worked perfectly!
Think you :)

ChiWei-Chen, 2022-09-06:
@mmege thanks! It's good to know this helps!

adityamukho, 2022-09-06:
> fwupdmgr --allow-older --allow-reinstall install Lenovo-ThinkPad-T14T15P14sP15sGen2-Consumer-MEFirmware-15.0.41.2142.cab

```
No supported devices found
```

@ChiWei-Chen I have a ThinkPad P15V Gen 2i (Corporate).

Here's the output of `dmidecode |grep SKU`:
```
	Corporate SKU
	SKU Number: LENOVO_MT_21AA_BU_Think_FM_ThinkPad P15v Gen 2i
```

Before posting here, I had downloaded a firmware update from https://pcsupport.lenovo.com/in/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p15v-gen-2-type-21a9-21aa/21aa/21aas0r500/pf3q3dkv/downloads/driver-list/component?name=Motherboard%20Devices%20%28core%20chipset,%20onboard%20video,%20PCIe%20switches%29

(n38rk02w.zip) and run the following command:
```
fwupdmgr --allow-older --allow-reinstall install Lenovo-ThinkPad-T15pP15vGen2-CorporateMEFirmware-15.0.41.2142.cab
```

It succeeded but the issue still remains.

Here's the output of `fwupdmgr get-devices` (The relevant portion of Intel ME):
```
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    240.41.2142
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              8d023cbb-b57d-4c49-bdf4-7c563c36aad6
│                         d044385c-6abc-50f4-9a10-61e71b8e323d ← UEFI\RES_{8D023CBB-B57D-4C49-BDF4-7C563C36AAD6}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
```

Here's my BIOS dump:
```
Product Name:ThinkPad P15v Gen 2i
Manufacturer:LENOVO
Bios Version:N38ET35W (1.16 )
Bios Date:2022-07-14
ME Firmware Version:15.0.41.2142
Embedded Controller Version:N38HT26W (1.08 )
Machine Type:21AAS0R500
System-unit Serial Number:PF3Q3DKV
System Board Serial Number:L1HF23S00XY
Asset Tag:No Asset Information
Cpu Type:11th Gen Intel(R) Core(TM) i9-11950H 
Cpu Speed:2.600GHz
Installed Memory:32768MB
UUID:5540dbcc-356f-11b2-a85c-8a209350a9af
```

ChiWei-Chen, 2022-09-06:
@adityamukho , thanks for sharing the information. 
P15v Gen2 is different platform from P15s Gen2, so the result is expected.
I noticed the Embedded Controller version is pretty old, could you try to install the latest version 1.11 (N38HT29W) and see how it works?
https://pcsupport.lenovo.com/in/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p15v-gen-2-type-21a9-21aa/21aa/21aas0r500/pf3q3dkv/downloads/driver-list/component?name=BIOS%2FUEFI

I cannot find the EC on LVFS, I'll need to check with FW team. Thanks.

adityamukho, 2022-09-07:
@ChiWei-Chen I extracted the two cab files and applied both updates (one was redundant since it was a BIOS update and my BIOS was already at the latest version. The EC update had to be forced, however, because for some strange reason, its current version number shows up as 65544!

```
# fwupdmgr install N38HT29W.cab 
Decompressing…           [***************************************]
Specified firmware is older than installed '0.1.10 < 65544'
```

Here's the relevant output from `dmidecode` for the BIOS and the EC after forcing both updates:
```
Handle 0x002E, DMI type 140, 15 bytes
ThinkPad Embedded Controller Program
        Version ID: N38HT26W
        Release Date: 08/17/2021

...

Handle 0x0013, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: N38ET35W (1.16 )
        Release Date: 07/14/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                PNP is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                3.5"/720 kB floppy services are supported (int 13h)
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.16
        Firmware Revision: 1.8
```

As before, the ME firmware read still fails.

ChiWei-Chen, 2022-09-07:
@adityamukho your feedback is very appreciated, I'll share this with FW team and see their replies.

val06, 2022-09-08:
> Thanks to @kleops9 I managed to reboot the heating brick after trying an automatic update that did not work (black screen for about 20 mins and no response from pressing power button during +60sec.) by pressing the little button at the back of the laptop.
> 
> Just to help people who may encounter this issue later, here is the process I followed to recover the firmware:
> 
> Ubuntu 22.04 LTS with kernel:
> 
> ```
> $ uname -a
> Linux XXXXXXXX 5.15.0-43-generic #46-Ubuntu SMP Tue Jul 12 10:30:17 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
> ```
> 
> Version of the P15 Gen 2 and original firmware version:
> 
> ```
> $ sudo fwupdmgr update
> Devices with no available firmware updates: 
>  • Integrated Camera
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI dbx
> Devices with the latest available firmware version:
>  • Embedded Controller
>  • Intel Management Engine
> ╔══════════════════════════════════════════════════════════════════════════════╗
> ║ Upgrade Intel Management Engine from 0.41.2142 to 240.41.2142?               ║
> ╠══════════════════════════════════════════════════════════════════════════════╣
> ║ Lenovo ThinkPad P1 Gen 4 Corporate ME Firmware                               ║
> ║                                                                              ║
> ║ This stable release fixes the following issues:                              ║
> ║                                                                              ║
> ║ • Update IOM FW to 24.31.0.0000                                              ║
> ║ • Update TBT FW to 15.0.0.0045                                               ║
> ║ • Update NPHY FW to 15.107.135.5017                                          ║
> ║ • Update PMC FW to 150.2.10.1020                                             ║
> ║                                                                              ║
> ║ 20YRS1PV01 must remain plugged into a power source for the duration of the   ║
> ║ update to avoid damage.                                                      ║
> ╚══════════════════════════════════════════════════════════════════════════════╝
> 
> Perform operation? [Y|n]: n
> ```
> 
> Since I have a corporate version (Intel Core vPRO i9) then I went to this page: https://pcsupport.lenovo.com/us/fr/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p15-gen-2-type-20yq-20yr/20yr/20yrs1pv01/downloads/driver-list/component?name=Chipset
> 
> And took the version `Intel Management Engine 15.0 Firmware (Corporate) Update Utility for Linux - ThinkPad`
> 
> However you have two versions: `14 Jun 2022` | `25 Jul 2022` Please download the first one (old version from June).
> 
> Open the file `/etc/fwupd/daemon.conf` as root and modify line 50 to `OnlyTrusted=false`
> 
> Then extract the content of the archive you downloaded and run the following command as root:
> 
> ```
> $ fwupdmgr --allow-older --allow-reinstall install Lenovo-ThinkPad-P15P17T15gGen2-CorporateMEFirmware-15.0.41.2142.cab
> ```
> 
> Finally, the computer will restart, try again to update the ME firmware until 48% and then fail. After, you will be blocked for about 60 seconds on the screen ```Failed to recover ME firmware..." and wait for the computer to reboot automatically.
> 
> On my side, no firmware update at startup (so no error message) and sound is back to normal!

had the same issue, the fix worked for me, sound is back to normal!

isagalaev-dd, 2022-09-23:
Another "me too" to @adityamukho . Having the same problem (and ran all the same updates, with all the same messages) for P15v Gen 2.

adityamukho, 2022-11-03:
@ChiWei-Chen any updates on this issue? I really don't get why it should take this long to fix crucial broken firmware.

isagalaev-dd, 2022-11-03:
@adityamukho for what it's worth, my issue has been fixed by installing the SOF firmware package with `sudo apt install firmware-sof-signed`. It's available in Ubuntu 22.04. So it looks like an Ubuntu packaging issue where they don't install it by default for some reason, not Lenovo's.

ChiWei-Chen, 2022-11-04:
@adityamukho sorry for the late, I know the FW assigned someone to look at this, but I didn't get much info, I'll need to follow up again.
BTW, could you give it a try with the latest SOF firmware? This seems more like the Ubuntu issue than the BIOS/EC FW issue for my view. Thanks.

@adityamukho thanks for the comment.

adityamukho, 2022-11-07:
Thanks @isagalaev-dd. I checked that the SOF package is already installed in my system. The issue remains unsolved.

@ChiWei-Chen

ChiWei-Chen, 2022-11-08:
@adityamukho , thanks, I shared this with FW team, and I'll give you back once any feedback from the team

mrhpearson, 2022-12-20:
Feedback from the FW team is a BIOS update is need to fix this and is in the works. Still under testing and not ready for release but expect to have it in the new year.

kmauleon, 2023-01-06:
can you please try as suggested here >> https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135

sepi4, 2023-01-12:
> can you please try as suggested here >> [#295 (comment)](https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135)

Thanks. This worked for my p15s.
```
fwupdmgr reinstall
```
and selected (Intel Management Engine) for reinstall. After restart: 
- showed recover message again 
- black screened for couple a minutes
- message with Reset System

After that system rebooted and no more recover message and audio workings fine.

kmauleon, 2023-01-13:
another suggestion for those still has issues >> https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135  please try... thank you very much

kmauleon, 2023-01-29:
hi another user updated firmware using Windows 10 Go-To USB >>> https://github.com/fwupd/firmware-lenovo/issues/280#issuecomment-1406307948

kmauleon, 2023-02-23:
@kleops9 any update from your side? OK to close this one?  please advise... thank you very much.

kleops9, 2023-02-23:
It can be closed. I did not have any issue again.

closed 2023-02-23 by kleops9

adityamukho, 2023-03-27:
Not fixed for me yet. The issue persists exactly as before. Please see comment [here](https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1237877231) for my specific hardware and issues faced.

kmauleon, 2023-03-27:
hi @adityamukho can you please try to update BIOS and EC firmware first if you hadn't.  [BIOS 1.25](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN37ET.firmware) and [EC 1.10](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN37HT.firmware) are available already

adityamukho, 2023-03-27:
@kmauleon these firmware upgrades are not compatible with my hardware (Lenovo (ThinkPad) ThinkPad T15p Gen 2i/ P15v Gen 2i).

I did try `fwupdmgr upgrade` and it ran a few upgrades but the failure still persists.

adityamukho, 2023-03-27:
Actually, it tries to run the same from-version to to-version upgrades every single time. It never really get upgraded.
```
# fwupdmgr upgrade
Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.8 to 0.1.11?                            ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad T15p Gen 2i/P15v Gen 2i Embedded controller Firmware         ║
║ N38HT29W version 1.11                                                        ║
║                                                                              ║
║ 21AAS0R500 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 240.41.2142 to 240.42.2235?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Intel Platform Update 2022.3 Product Version Maintenance Release             ║
║                                                                              ║
║ Version 15.0.42.2235 (LVFS: 240.42.2235)                                     ║
║                                                                              ║
║ Problem Fixes                                                                ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities under issues.Please       ║
║ see fixed issues for details.                                                ║
║                                                                              ║
║ 21AAS0R500 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • Prometheus
[... 17 lines cut ...]

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
 • UEFI dbx
 • WDC PC SN730 SDBQNTY-1T00-1001

An update requires a reboot to complete. Restart now? [y|N]:
```

After this I restart the system, and the ME upgrade fails as always. All 3 firmware upgrades get rolled back at some point since the vesion numbers remain unchanged no matter how many times I run this.

kmauleon, 2023-03-27:
@adityamukho since you have a different system can you please open a separate issue for proper tracking... thank you very much...