fwupd/firmware-lenovo #252: fwupdmgr fails to update UEFI on my Lenovo P52 - UEFI capsule updates not available or enabled in firmware setup

rugubara, 2022-08-04:
PF16W6Y2 ~ # fwupdmgr update
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Devices with no available firmware updates:
 • Corsair MP400
 • Corsair MP400
No updatable devices

superm1, 2022-08-04:
One more suggestion is to try to reset BIOS default settings to see if it fixes this issue, otherwise it should be for Lenovo to dig into further for your system.

rugubara, 2022-08-05:
Resetting the BIOS to default settings didn't change anything. Still no way to update UEFI. I'll create an issue in the LEnovo firmware project.

mrhpearson, 2022-08-17:
I just tested this out on my P52 and it updated the BIOS with no issues.

I was running Fedora 36 - would you mind trying the update from a F36 live session just to see if it makes any difference? (and being nosy - which distro are you running?)

Also - which FW version are you currently on? You can check from the BIOS setup screen.

rugubara, 2022-08-18:
I'm running Gentoo. 
The version of BIOS reported in the Thinkpad Setup - N2CET63W (1.46)
The version of EC N2CHT26W (1.16)
The version of ME 12.0.85.1869

I'll try with F36 live and report.

Karmanuk, 2022-09-09:
turn off Secure Boot (disabled) and everything is ok

Karmanuk, 2022-09-09:
sorry, Windows UEFI Firmware Update turn it on