fwupd/firmware-lenovo #253: All keys have been deleted

solsticedhiver, 2022-08-05:


superm1, 2022-08-05:
fwupd doesn't directly delete any keys like this.  However the capsule update process may have done it.  Transferring over to Lenovo to investigate.

solsticedhiver, 2022-08-05:
I was able to get back those OEM keys by using an option in the BIOS. So no harm done. I use shim for Secure Boot.

joseghasto, 2022-11-22:
Just found this as I had the same happen to me with a Thinkpad T470 when I upgraded to the Lenovo System Firmware 0.1.72 it both wiped my boot entries and put the device in setup mode.  I'm using Arch with custom keys for Secure Boot.

If I remember correctly I had done a previous upgrade on this machine, but I think it was with Fedora and without custom keys for Secure Boot. That one did not impact on the boot entries.

mrhpearson, 2022-11-22:
Thanks @joseghasto - I've added notes to the internal ticket looking at this issue (LO-2077). The FW team are investigating