fwupd/firmware-lenovo #254: Intel Management Engine update fails on T490

frazar, 2022-08-08:
Upgrading the Intel Management Engine fails on my T490. 
Related to #140, where users were asked to open a separate issue for different laptop models.

    fwupdmgr update --show-all

```
❯ fwupdmgr update --show-all
Devices with no available firmware updates:
 • INTEL SSDPEKNW020T8
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 192.72.1757 to 192.90.2072?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Version 12.0.90.2072 (LVFS: 192.90.2072)                                     ║
║                                                                              ║
║ Problem Fixes                                                                ║
║                                                                              ║
║ • Intel CSME IPU 2022.1                                                      ║
║ • Mitigated the following security vulnerabilities under issues:             ║
║                                                                              ║
║ 20N2CTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: n
Request canceled
```

    system firmware version and Intel Management Engine firmware version in BIOS main page

```
❯ sudo dmidecode -s bios-version  
N2IET98W (1.76 )
```
```
❯ ./intel_csme_version_detection_tool 
Intel(R) CSME Version Detection Tool
Copyright(C) 2017-2022, Intel Corporation, All rights reserved.

Application Version: 7.0.2.0
Scan date: 2022-08-08 07:14:56 GMT

*** Host Computer Information ***
Name: zardi-pc
Manufacturer: LENOVO
Model: 20N2CTO1WW
Processor Name: Intel(R) Core(TM) i5-8265U CPU @ 1.60GHz
OS Version: Arch Linux (5.15.59-2-lts)

*** Intel(R) ME Information ***
Engine: Intel(R) Converged Security and Management Engine
Version: 12.0.90.2072

*** Risk Assessment ***
Based on the analysis performed by this tool: This system is not vulnerable. It has already been patched.

For more information refer to the Intel(R) CSME Version Detection Tool User Guide
or the related Intel Security Advisory list at:
https://www.intel.com/content/www/us/en/support/articles/000031784/technologies.html
```

    OS and fwupdmgr version

❯ cat /etc/os-release 
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux-logo

❯ fwupdmgr --version        
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.15.59-2-lts
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10

Let me know if any information is missing.

gustafla, 2022-08-17:
I'm also having this issue on my T490, the ME update just won't "stick".

eaepstein, 2022-08-19:
Same failure to update to 192.90.2072 on P53 running RHEL 8.6
The update process shows no error and stops with "hit any key to continue"
fwupdmgr get-history  shows  for "Intel Management Engine":
   "Update Error:       failed to run update on reboot"
And the BIOS reports "ME Firmware Version"  12.0.90.2072

No problem installing other firmware

AlexandraAlter, 2022-08-28:
Same issue, Intel Management Engine from 192.72.1757 to 192.90.2072. Update does not claim to have failed, only two reboots occur, but fwupdmgr reports a failed update.
Lenovo T590, running Pop!OS 22.04 (an Ubuntu variant).
BIOS reports the current version as 192.90.2072.

ndrs-pst, 2022-10-09:
Dear All,

I'm using T490 with Windows 10 and faced the same problem.
After disabled "Enhanced Windows Biometric Security" in BIOS, I can success fully update Intel Management Engine.
Hope this also applies to this issue too.
[Windows BSOD and other problems after enabling "Enhanced Windows Biometric Security" in BIOS Setup - ThinkPad](https://pcsupport.lenovo.com/ca/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-7th-gen-type-20r1-20r2/solutions/HT511000-PROBLEMS-CAUSED-BY-ENHANCED-WINDOWS-BIOMETRIC-SECURITY-BIOS-SETTING-THINKPAD)

hvhaugwitz, 2022-11-06:
> After disabled "Enhanced Windows Biometric Security" in BIOS, I can success fully update Intel Management Engine. Hope this also applies to this issue too.

Disabling (or enabling) `Enhanced Windows Biometric Security` in BIOS did not change anything on my T490. `fwupdmgr update` still offers me the upgrade from `192.72.1757` to `192.90.2072` even after applying the upgrade via `fwupdmgr`.

mrhpearson, 2022-11-07:
@kmauleon & @ChiWei-Chen - not sure if we've tracked the T490 separately or not? Let me know if we have this one on the list of platforms to get fixes for the ME udpate issue rolled out.
Thanks!
Mark

ChiWei-Chen, 2022-11-08:
@mrhpearson thanks, I'll ask FW team to look at this. (LO-2109 for tracking)

karekaa, 2022-11-24:
The same problem occours for this (new fix for a critical bugfix for nvme-driver, if I undstood right?):
`
# fwupdmgr update
Upgrade Intel Management Engine from 192.90.2072 to 192.92.2145?

Version 12.0.92.2145 (LVFS: 192.92.2145) 

Problem Fixes

 • Intel CSME IPU 2022.3 
 • Mitigated the following security vulnerabilities under issues:

20N3SA9100 must remain plugged into a power source for the duration of the
update to avoid damage.
`
The process above seemingly works fine, but the installation does not happend during the reboot?!

Dagur, 2022-11-24:
@karekaa same here

kmauleon, 2022-11-25:
as per owner, BIOS ESRT fix will be included in the next release.... still asking for the target date though... will update once I received their feedback... thank you very much.

thaarbach, 2022-11-25:
@karekaa i've the same issue on T590. After update from 192.72.1757 -> 292.92.2145 the update is still available. 
OS and fwupdmgr
```
NAME="Fedora Linux"
VERSION="36 (Workstation Edition)"
ID=fedora
VERSION_ID=36
VERSION_CODENAME=""
PLATFORM_ID="platform:f36"
PRETTY_NAME="Fedora Linux 36 (Workstation Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:36"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f36/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=36
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=36
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```
```
runtime   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.hughsie.libjcat           0.1.12
runtime   org.kernel                    6.0.9-200.fc36.x86_64
runtime   com.dell.libsmbios            2.4
compile   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.gusb          0.3.10
```

kmauleon, 2022-11-28:
hi @thaarbach please open a separate tracking for T590 as this system has different owner....  thank you very much

kmauleon, 2022-12-15:
T490 BIOS ESRT fix will be available end of this week as per owner... thank you very much

bowlofeggs, 2022-12-17:
Does this issue cover the T490s, or should we open a separate tracking issue for that device? It has this issue, but I see different ME version numbers than shown in this ticket. I see the same numbers as reported in this comment: https://github.com/fwupd/firmware-lenovo/issues/162#issuecomment-1274452039

kmauleon, 2022-12-18:
T490 BIOS ESRT fix already available... please check >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2IETXXW.firmware

@bowlofeggs different platform sorry.  please open a separate ticket for proper tracking.  thank you very much.

Dagur, 2022-12-19:
1.78 Installed just fine but the 12.0.92.2145 update still fails to install for me :(

marianrh, 2022-12-19:
On my machine everything now works as expected. After installing firmware 1.78, the management engine firmware version is now reported as 192.92.2145 and no update is shown anymore, which is correct.

@Dagur, try just installing firmware 1.78 and rebooting. Afterwards, the management engine update should have disappeared.

Dagur, 2022-12-19:
That's what I did. I installed 1.78 before attempting to install the other one.

kmauleon, 2022-12-21:
@marianrh great! thank you very much for your feedback ^_^
@Dagur can you please share **fwupdmgr get-devices** result? thank you very much

Dagur, 2022-12-21:
I went to check today and the update is gone so I can only assume that it's been installed successfully 🤷

frazar, 2022-12-21:
Works for me too.

closed 2022-12-21 by frazar

kmauleon, 2022-12-21:
great! thank you very much all!