fwupd/firmware-lenovo #255: Intel Management Engine fails on ThinkPad T580/P52s

rochfort1, 2022-08-08:
This follows on from @DianaAlcantara's comments for [https://github.com/fwupd/firmware-lenovo/issues/140].

### (I)   PROCEDURE I USED TO DUPLICATE THE PROBLEM

1. Run:
john@T580:~$ fwupdmgr update

The following shows:
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
╔════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.86.3909 to 184.92.4222? 
╠════════════════════════════════════║
║ MEFW version update to 11.8.92.4222.
║
║ and be aware that, in Intel® CSME FW 11.8.92.4222 version 3, the VCN
║ increased 
║ 
║ to “331”. Therefore, downgrading from Intel® CSME FW 11.8.92.4222
║ version 3 to
║
║ earlier versions is not possible.
║
║ 20L9CTO1WW must remain plugged into a power source for the duration of the
║ update to avoid damage.
╚════════════════════════════════════╝

2. Run y to perform the operation.

The following shows:
Downloading…            ***************************************
Downloading…            ***************************************
Decompressing…        ***************************************
Authenticating…          ***************************************
Waiting…                     ***************************************
Writing…                      ***************************************
Waiting…                     ***************************************
Successfully installed firmware
Waiting…                     ***************************************
Devices with the latest available firmware version:
 • MZVLB1T0HALR-000L7
 • System Firmware

3. Run y to reboot and complete.

Rebooting follows this sequence:
  (a) The screen blanks and then shows: "Reset System".
  (b) The screen blanks and then shows: "Please wait while we install a system update".
  (c) The system then reboots normally.
  (d) After it starts, GNOME's software updates reports: "ThinkPad T580/P52s Corporate ME Update 184.86.3909 → 184.92.4222" and invites you to click an Update button.
  
4. If you click Update, step 3 repeats.

### (II)  INFORMATION YOU REQUESTED

1. fwupdmgr update --show-all

john@T580:~$ fwupdmgr update --show-all
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Intel Management Engine
 • MZVLB1T0HALR-000L7
 • System Firmware
 • Thunderbolt host controller

2. System firmware version and Intel Management Engine firmware version in BIOS main page

Main tab:
UEFI BIOS Version: N27ET46W (1.32)
ME Firmware Version: 11.8.92.4222
UEFI BIOS Date: 2022-05018
UEFI Secure Boot: Off

Security tab:
Secure Boot: Disabled

Startup tab:
UEFI/Legacy Boot: UEFI only
-CSM Support: No
Boot Order Lock: Disabled

3. OS and fwupdmgr version

john@T580:~$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.0-3-amd64
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10

### (III) MY GUESS AT WHAT MAY BE HAPPENING

The firmware is reporting that the ME firmware version is 11.8.92.4222. That seems to be what the upgrade is offering after Step 1. It looks as though fwupdmgr is not recognising that the version of Intel management engine is up-to-date.

afelinczak, 2022-09-07:
Have similar issue on my P52s. In my case seems that upgrade is completed but fwupdmgr doesn't recognize this.

1. BIOS
UEFI BIOS Version: N27ET47W (1.33)
ME Firmware Version: 11.8.92.4222
UEFI BIOS Date: 2022-07-18
UEFI Secure Boot: Off
Secure Boot: Disabled
Boot Order Lock: Disabled

fwupdmgr update --show-all
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • Unifying Receiver
Upgrade Intel Management Engine from 184.86.3909 to 184.92.4222?

fwupdmgr get-updates                                                                                                                                                     
Devices with no available firmware updates:                                                                                                                                
 • UEFI Device Firmware                                                                                                                                                    
 • UEFI Device Firmware                                                                                                                                                    
 • Unifying Receiver                                                                                                                                                       
Devices with the latest available firmware version:                                                                                                                        
 • MZVLB512HAJQ-000L7                                                                                                                                                      
 • System Firmware                                                                                                                                                         
 • Thunderbolt host controller                                                                                                                                             
20LCS27500                                                                                                                                                                 
│                                                                                                                                                                          
└─Intel Management Engine:                                                                                                                                                 
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd                                                                                                         
  │   Summary:            UEFI ESRT device
  │   Current version:    184.86.3909
  │   Minimum Version:    184.86.3909
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              ae82a157-cc81-419a-88e4-a983a76075db
  │                       a2bca6e6-43f2-5782-af9b-a957ea644511
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T580/P52s:
        New version:      184.92.4222
        Remote ID:        lvfs
        Release ID:       14600
        Summary:          Lenovo ThinkPad T580/P52s Corporate ME Firmware
        License:          Proprietary
        Size:             7,5 MB
        Created:          2016-07-08

mrhpearson, 2022-09-07:
Please check what version of ME is installed from the BIOS setup screen (F1 during early boot).
I think you've hit the known issue that the BIOS isn't updating the ESRT table correctly, so fwupdmgr thinks the update didn't take place. This is being fixed and rolled out.

DianaAlcantara, 2022-09-09:
Is it possible that there was a release today? If so, didn't work me :( 

[image]

```javascript
└─ ThinkPad T580/P52s:
│       New version:      0.1.33
│       Remote ID:        lvfs
│       Release ID:       14956
│       Summary:          Lenovo ThinkPad T580/P52s System Firmware
│       License:          Proprietary
│       Size:             9,5 MB
│       Created:          2022-07-18
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       change in this change
│       
│       Version 1.33
│       
│       New functions or enhancements-Updated the Diagnostics module to version 04.25.000.
│       
│       Problem fixes-Nothing
│     
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   184.86.3909
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2022-09-09 13:53
  │   GUID:               72e9217b-717e-4908-b428-fe4df390177b
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T580 P52s:
        New version:      184.92.4222
        Remote ID:        lvfs
        Release ID:       14611
        Summary:          Lenovo ThinkPad T580 P52s  Consumer ME Firmware
        License:          Proprietary
        Size:             2,2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N27RG19W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        MEFW version update to 11.8.92.4222. and be aware that, in Intel® CSME FW 11.8.92.4222 version 3, the VCN increased to “331”. Therefore, downgrading from Intel® CSME FW 11.8.92.4222 version 3 to earlier versions is not possible.
      

```

DianaAlcantara, 2022-09-12:
Hello, good morning/afternoon/evening!!! Is there an ETA for this fix?

mrhpearson, 2022-09-16:
Tagging @ChiWei-Chen - I couldn't find these two specifically but most of the platforms were targetted for beginning and mid september.

ChiWei-Chen, 2022-09-20:
@mrhpearson , thanks I asked FW team to look at this (LO-2036 for internal tracking)

ChiWei-Chen, 2022-09-22:
Hi,
The FW team confirmed the next BIOS will include the fix, it is expected to release a few weeks later. Thanks!

rochfort1, 2022-10-17:
Lenovo have made a new firmware (they call it BIOS) version available. After updating, I now see the following on the Main tab of BIOS main page:
UEFI BIOS Version: N27ET48W (1.34)
ME Firmware Version: 11.8.92.4222
UEFI BIOS Date: 2022-09-16

GNOME's software updates no longer requests an ME Firmware update.

Looks like they have fixed the problem.

ChiWei-Chen, 2022-10-18:
@cponder, could you give it a try with `sudo fwupdmgr update` again?
The FW has moved to stable a few minutes ago.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN27ETxxW.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN27ETxxP.firmware

@rochfort1 , thanks, it's good to know this.

cponder, 2022-10-18:
In the `system -> software -> updates` menu, it looks like the latest updates have cleared the problem out, so it doesn't show there being any updates left to make.
When I run `fwupdmgr get-history` I see this
```
├─Intel Management Engine:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Previous version:   184.86.3909
│ │   Update State:       Failed
│ │   Update Error:       failed to run update on reboot
│ │   Last modified:      2022-10-18 04:39
│ │   GUID:               ae82a157-cc81-419a-88e4-a983a76075db
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T580/P52s:
│       New version:      184.92.4222
│       Remote ID:        lvfs
│       Release ID:       14600
│       Summary:          Lenovo ThinkPad T580/P52s Corporate ME Firmware
│       License:          Proprietary
│       Size:             7.5 MB
│       Created:          2016-07-08
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N27RG19W
│       Vendor:           Lenovo
│       Description:      
│       MEFW version update to 11.8.92.4222.
│       
│       and be aware that, in Intel® CSME FW 11.8.92.4222 version 3, the VCN increased
│       
│       to “331”.Therefore, downgrading from Intel® CSME FW 11.8.92.4222 version 3 to
│       
│       earlier versions is not possible.
│       
│       Do not turn off your computer or remove the AC adapter while the update is in progress.
```
Since these two entries are labeled differently, I can't tell if (A) the first one was an old failure and the second one shows the subsequent success, or (B) they represent different components and the `Intel Management Engine` is still in a backlevel state because it couldn't update.
Is there a command just to only show what's on the system now, as opposed to showing the historical levels?

rochfort1, 2022-10-18:
@cponder If you want to see what versions of the UEFI firmware and management engine firmware are installed, press F1 when you start the computer and look at the BIOS main page Main tab. The following are of interest:
UEFI BIOS Version:
ME Firmware Version:

You appear to have found a problem with what `$ fwupdmgr get-history` reports. In my case, the BIOS main page shows:
UEFI BIOS Version: N27ET48W (1.34)
ME Firmware Version: 11.8.92.4222

But if I run `$ fwupdmgr get-history`, I see the following edited detail:
System Firmware:
│ └─ ThinkPad T580/P52s:
│       New version:      0.1.33
And:
Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   184.86.3909
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2022-09-09 20:42
  │   GUID:               ae82a157-cc81-419a-88e4-a983a76075db
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T580/P52s:
        New version:      184.92.4222
        Remote ID:        lvfs
        Release ID:       14600
        Summary:          Lenovo ThinkPad T580/P52s Corporate ME Firmware
        License:          Proprietary
        Size:             7.5 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N27RG19W
        Vendor:           Lenovo
        Update Message:   Do not turn off your computer or remove the AC adapter while the update is in progress.
        Description:      
        MEFW version update to 11.8.92.4222.

It looks like `$ fwupdmgr get-history` is reporting the status prior to my latest update.

Since it doesn't report that either the UEFI firmware or management engine firmware require updating, GNOME's **Software Updates** utility must obtain the status in a different way.

Maybe, as @mrhpearson suggested, the BIOS still isn't updating the ESRT table correctly.

kmauleon, 2022-11-18:
hi all please update to latest BIOS 1.34 with ESRT issue fix.  Thank you very much.

hughsie, 2022-11-18:
@kmauleon is the ME update supposed to have a `other` requirement on the system BIOS version? That would mean we don't offer the ME update if the system firmware is too old.

kmauleon, 2022-11-18:
@hughsie case by case but basically ME owner should know if a BIOS version is a prerequisite.  For this case, this version has BIOS ESRT issue fix wherein fwupd shows old MEFW version even if system has the latest MEFW already

rochfort1, 2022-11-19:
@kmauleon I don't think the problem has been fixed.

Lenovo [PC Support  >  laptops-and-netbooks   >  thinkpad-t-series-laptops   >  T580 (Type 20L9, 20LA) Laptop (ThinkPad)] tell me that I must use Version 1.34 of the BIOS Update Utility (Linux) and Version 11.8.93.4323 of the Intel Management Engine Firmware. These I have installed.

Currently, my ThinkPad Setup Main page shows:
UEFI BIOS Version: N27ET48W (1.34)
UEFI BIOS Date: 2022-09-16
ME Firmware Version: 11.8.93.4323

However, $ fwupdmgr get-history reports:
System Firmware:
  ThinkPad T580/P52s:
  New Version 0.1.33
  Version 1.33
Intel Management Engine:
  ThinkPad T580/P52s
    New version: 184.92.4222
    
fwupdmgr get-history is still reporting incorrect versions for the system firmware and Intel management engine firmware.

kmauleon, 2022-11-21:
hi @rochfort1 to confirm BIOS ESRT fix, please execute fwupdmgr get-devices and let us know the results... thank you very much

rochfort1, 2022-11-21:
@kmauleon The «fwupdmgr get-devices» report is very big. I edited it to cover just the Intel management engine and system firmware parts, These follow:

`LENOVO 20L9CTO1WW`
`│`
`├─Intel Management Engine:`
`│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd`
`│     Summary:            UEFI ESRT device`
`│     Current version:    184.93.4323`
`│     Minimum Version:    0.0.1`
`│     Vendor:             Lenovo (DMI:LENOVO)`
`│     Update State:       Success`
`│     GUID:               ae82a157-cc81-419a-88e4-a983a76075db`
`│     Device Flags:       • Internal device`
`│                         • Updatable`
`│                         • System requires external power source`
`│                         • Supported on remote server`
`│                         • Needs a reboot after installation`
`│                         • Device is usable for the duration of the update`
`│   `
`├─System Firmware:`
`│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f`
`│ │   Summary:            UEFI ESRT device`
`│ │   Current version:    0.1.34`
`│ │   Minimum Version:    0.0.1`
`│ │   Vendor:             Lenovo (DMI:LENOVO)`
`│ │   Update State:       Success`
`│ │   GUIDs:              52dd29cd-24b9-4d60-8a45-e72670901924`
`│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware`
`│ │   Device Flags:       • Internal device`
`│ │                       • Updatable`
`│ │                       • System requires external power source`
`│ │                       • Supported on remote server`
`│ │                       • Needs a reboot after installation`
`│ │                       • Cryptographic hash verification is available`
`│ │                       • Device is usable for the duration of the update`
`│ │ `
` `
`Devices that were not updated correctly:`
` `
` • Intel Management Engine (184.86.3909 → 184.92.4222)`

Correct values for the versions of the Intel management engine and system firmware are reported. However the bit at the bottom where it says that the Intel management engine didn't update from 184.86.3909 to 184.92.4222 correctly is wrong. The last update was from 184.92.4222 to 184.93.4323.

kmauleon, 2022-11-21:
@rochfort1 fwupd is displaying the latest version but as you've said the last information is strange... but i think this is on the fwupd side... @hughsie can you please check?  thank you very much....

hughsie, 2022-12-02:
@rochfort1 can you share your /var/lib/fwupd/pending.db please.

rochfort1, 2022-12-03:
@hughsie I put the SQLite3 database file **pending.db** here:
https://www.dropbox.com/s/m6s6bzl1miconml/pending.db?dl=0

kmauleon, 2022-12-15:
hi @rochfort1 @hughsie since BIOS ESRT issue was already fixed, can we close this ticket and open another for the fwupd related issue for proper tracking?  please advise.  thank you very much.

closed 2022-12-15 by rochfort1