fwupd/firmware-lenovo #256: Lenovo X1 Carbon G9 firmware update failure

myllynen, 2022-08-10:


mrhpearson, 2022-08-17:
Hi - are you able to catch (maybe video) the error message when it happens? If you can do that I'll flag to the FW team. I know I've done this update succesfully on my X1C9 (from Fedora) so it should work...

myllynen, 2022-08-18:
Thanks for looking into this. I recorded a video and saw the complete message.

The update procedure starts as usual, first "Reset System" is shown on the left-corner of the screen and then few moments later below the progress bar "Please wait while we install a system update" text appears.

However, the progress bar is perhaps only at 10% when "FMP Capsule Update Fail" text is shown right under logo (so above the progress bar). The text is shown less than a second before the system reboots and boots up to OS.

mrhpearson, 2022-09-12:
Sorry for the slow reply - this got lost in the swamp.
We did have some issues where for Windows they were asking users to disable disk encryption (bitlocker) before doing FW updates; but guidance I was given is it didn't imply to Linux as bitlocker wasn't being used.
How do you have your disk encrypted?

myllynen, 2022-09-12:
I have my disk encrypted by the RHEL installer (using LUKS as is usual on Linux) but only the non-boot partitions. Given this and the fact that all the other updates applied without issues I don't think at least in my case that would be the root cause, especially when I already see the update process starting on my system before failing with the "FMP Capsule Update Fail".

FWIW, my system received a bunch of other firmware updates recently and all of them were applied ok.

mrhpearson, 2022-09-16:
@ChiWei-Chen - do you think this is related to the other couple of Capsule related errors we've got or is this something new? (LO-1964 in particular?

ChiWei-Chen, 2022-09-20:
@mrhpearson thanks. This seems TBT Retimer Firmware related, I'll need FW owner to check it. (LO-2035 for tracking)

adgud, 2022-09-21:
I hit the same error repeatedly and the solution in my case was to use a standard power supply instead of a docking station. The upgrade completed successfully.

myllynen, 2022-09-21:
I don't have a docking station so for me this issue occurs when using a standard power supply.

adgud, 2022-09-21:
> I don't have a docking station so for me this issue occurs when using a standard power supply.
I see, so the issue you are having is different than mine, despite the same symptoms.

jwflory, 2022-10-12:
I am able to replicate this issue on RHEL 8.6 (CSB) on a Red Hat standard issue X1 Carbon G9, except with two differences:

1. The firmware upgrade bricked the laptop, once it got stuck on the Lenovo boot splash with the updating firmware message. The upgrade hung here and was never completed, leaving the laptop fully unresponsive with both the power button and the reset pinhole on the bottom having no effect. I had to wait for the battery to die, and then I could boot back into my OS.
2. Before the hanging firmware upgrade message, I had the same `Could not apply capsules: volume full` message as #257.

I also had the `Thunderbolt Retimer Firmware 1.2.24.0` update as pending.

At the moment, restarting my laptop triggers the firmware upgrade cycle again and it gets stuck on every restart. One mitigation I am trying is to use Fedora Linux 37 boot media to trigger the firmware upgrade, but I keep having issues with storage space. I wasn't able to triage yet whether it was the USB storage or an issue with the filesystem on the laptop drive.

nodens, 2022-11-22:
I had the issue on the same firmware update.
Switching from the screen USB-C power supply to the original power brick worked for me, even though the screen provide more power supposedly…

Small trick for avoiding to wait for the battery to empty: unplug from power, and use a paperclip in the "reset hole", as described here: https://pcsupport.lenovo.com/us/en/solutions/ht513019-system-may-hang-at-lenovo-splash-screen-with-fmp-capsule-update-fail-message-thinkpad

mrhpearson, 2023-03-17:
Please do check how much space you have in your EFI partition - apparently at least 80MB is recommended