fwupd/firmware-lenovo #257: Could not apply capsules: volume full

rogueai, 2022-08-15:


olegk25, 2022-08-15:
I have the same issue and my /efi is only 18% full

hughsie, 2022-08-16:
@mrhpearson any ideas?

ChiWei-Chen, 2022-08-19:
@rogueai , I'm working with Mark and got these questions from FW team, could you help t comment them? Thanks!

> Could you help confirm where is this message show (OS,post..or..)?
> Please provide the detail information.
> BTW, could you let user try use “fwupdmgr update [BIOS Device ID]” to check if can update successfully? I want to confirm if can duplicate or not.
> Device ID can get from System Firmware by type “fwupdmgr get-devices” .
>  
> Ex:  

[image]
[image]

Jaeson813, 2022-08-22:
This also happens on my system—ThinkPad T14 Gen 2 (AMD) running Fedora 36.
**Kernel**: `Linux 5.18.17-200.fc36.x86_64`
**Output of `df -h /dev/nvme0n1p1`**:
```
Filesystem      Size  Used Avail Use% Mounted on
/dev/nvme0n1p1   96M   72M   25M  75% /boot/efi
```

The error message occurs after the machine reboots and says `fwupd-efi`’s version, which is 1.3 I believe.
Trying to update by targeting specifically the System Firmware has the same issue.

ChiWei-Chen, 2022-08-31:
Hello @Jaeson813 ,

Have you tried resizing the EFI partition?

asegarra, 2022-09-04:
Fedora 36 on X1 Carbon Gen 9

I have the same issue, tried “fwupdmgr update [BIOS Device ID]” and got the same error. My efi partition is 629MB of which 603 are free. Resizing in not an option for me at the moment, what is the recommended size for fwupd?

Jaeson813, 2022-09-05:
@ChiWei-Chen, I’m unable to resize the partition currently. I booted into Windows, and one of the optional BIOS updates actually did successfully update it to 1.19.

olegk25, 2022-09-05:
Hi, it's obviously a logical glitch in fwupdmgr while trying to assess available space. 
I used bootable image from the vendor and upgrade was successful.

ChiWei-Chen, 2022-09-06:
@hughsie is there any idea or anything we can do for fwupdmgr?

@Jaeson813 , @olegk25 , thanks for the comment.

hughsie, 2022-09-06:
@ChiWei-Chen I don't think it's fwupd at fault here, the userspace bit has already written the capsule to the ESP at the point the `Could not apply capsules: volume full` appears. The `volume full` is coming either from the UEFI runtime service, either `RT->QueryCapsuleCapabilities` or `RT->UpdateCapsule` -- perhaps it's telling us that NVRAM is full, rather than the ESP is full?

ChiWei-Chen, 2022-09-06:
Richard, Thanks, I'll give you back once I heard any clues from FW experts.

rogueai, 2022-09-28:
I've retested this with `v1.8.5` and it seems to be working now. Unfortunately, in the meanwhile I've setup secure boot + custom signing keys so my environment is not as it was when I first tested it with `v1.8.3`

jwflory, 2022-10-12:
This issue captures my problem, except with a few differences:

1. This bricked the laptop as my firmware upgrade got stuck while applying the firmware update, and it never completed. I had to wait for power loss before I could then reboot and access my OS
2. Using a X1 Carbon Gen9 on RHEL 8.6 (RH CSB)
3. I don't use rEFInd

I experienced this on the first day of using the laptop, so I am hesitant to blame storage as an issue. This happened after I applied the first round of updates in GNOME Software, after I completed initial user setup.

susman, 2022-11-16:
same issue here:

```
 ~ # fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.2
compile   com.hughsie.libjcat           0.1.12
runtime   org.kernel                    5.15.75-gentoo
runtime   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.gusb          0.4.2
```

```
 ~ # dmidecode -s system-version
ThinkPad X1 Carbon Gen 9
 ~ # dmidecode -s system-product-name
20XWCTO1WW
```

fwupd-efi 1.5 from latest `main` on https://github.com/fwupd/fwupd-efi doesn't solve it either. The result is the same:

```
Could not apply capsules: volume full
```

Can we please re-open this?

mrhpearson, 2022-12-02:
Looks like we didn't leave a comment...but this has been reported to the FW team (a couple of weeks back) and we're discussing with them

mrhpearson, 2022-12-20:
Sorry - no, we've been nagging the FW team for some insight into what would cause that message.
Just for my reference as it took me a while to find the notes - tracked by internal ticket LO-1964

noobsauce101, 2023-01-02:
Hi everyone- I am also having this exact problem. I have a lenovo thinkpad T490 with Fedora 36. My fwupd-efi version is also 1.3 , fwupd is 1.8.6

gstephan76, 2023-01-05:
Hi Everyone. Lenovo ThinkPad P1 Gen 3, running Fedora 37, with the same issue.
fwupdmgr --version: 
compile   org.freedesktop.fwupd         1.8.8
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.2
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.3
runtime   org.freedesktop.fwupd         1.8.8
runtime   org.kernel                    6.0.16-300.fc37.x86_64

ChiWei-Chen, 2023-02-15:
Hi @alexted , @noobsauce101 , @gstephan76 , our FW team wasn't able to reproduce this issue yet, but they suspect this is EFI space related, could you provide output of `df -h` for our investigation.

ChiWei-Chen, 2023-02-15:
@alexted , thanks.
Could you also provide the output of `fwupdmgr --version` and `fwupdmgr get-devices`, thanks!

noobsauce101, 2023-02-15:
Hi, this is my output. As you can see I have nvme and my EFI is relatively full

`Filesystem                          Size  Used Avail Use% Mounted on
devtmpfs                            4.0M     0  4.0M   0% /dev
tmpfs                               7.7G   15M  7.7G   1% /dev/shm
tmpfs                               3.1G  2.2M  3.1G   1% /run
/dev/dm-0                           327G  204G  122G  63% /
tmpfs                               7.7G  3.1M  7.7G   1% /tmp
/dev/dm-0                           327G  204G  122G  63% /home
/dev/loop10                          50M   50M     0 100% /var/lib/snapd/snap/snapd/17883
/dev/loop11                         348M  348M     0 100% /var/lib/snapd/snap/wine-platform-runtime/333
/dev/nvme0n1p5                      974M  258M  649M  29% /boot
/dev/nvme0n1p2                       95M   79M   17M  83% /boot/efi
tmpfs                               1.6G  280K  1.6G   1% /run/user/1000`

hughsie, 2023-02-15:
> /dev/nvme0n1p2                       95M   79M   17M  83% /boot/efi

@ChiWei-Chen is that too little? If you give me the "it must have xxxM free on the ESP" then I can add that as a check on fwupd.

ChiWei-Chen, 2023-02-15:
@hughsie , it looks a bit little, I'll need to get FW team to comment on this. Thanks!

ChiWei-Chen, 2023-02-15:
> Hi everyone- I am also having this exact problem. I have a lenovo thinkpad T490 with Fedora 36. My fwupd-efi version is also 1.3 , fwupd is 1.8.6

It looks Fedora 36.

@hughsie , here is the FW team's feedback, hope this helps:

> It is small.
> I am not sure what product is used. But basically recent BIOS capsule image size is appropriate 30 MB.
> I recommend to prepare double size of BIOS capsule file at least. And the space should be sequential.
> Roughly 80 MB available space is needed.

BTW, the issue of the following system needs to investigate separately. 

> ```shell
> /dev/nvme0n1p1  300M  312K  300M   1% /boot/efi
> ```

superm1, 2023-02-15:
> Roughly 80 MB available space is needed.

I don't recall if we are already, but we should make sure we're clearing space after updates finish the next boot too by the daemon.

noobsauce101, 2023-02-15:
Sorry I completely forgot to mention this but I have a dual-boot system, Windows 10 + Fedora 36. Looking through the /boot/efi/EFI directory I found the following:
```
du -hsx /boot/efi/EFI/fedora/fw*
37M	fedora/fw
61K	        fedora/fwupdx64.efi
```
```
du -hsx /boot/efi/EFI/fedora/fw/*
12M	fedora/fw/fwupd-2c0665e2-fdbd-495e-b8e4-69d92b9c119a.cap
754K	fedora/fw/fwupd-38ea6335-29ca-417b-8cd4-6b4e5e866f92.cap
24M	fedora/fw/fwupd-603baf73-b997-45b5-86b4-2f981a008e18.cap
`

hughsie, 2023-02-15:
> du -hsx /boot/efi/EFI/fedora/fw/*

We should be deleting old capsule files before scheduling the new ones, as long as BootNext isn't set. What's the creation time on those files please?

noobsauce101, 2023-02-15:
Feb 10 13:57 on all 3

noobsauce101, 2023-02-16:
Since the entire size of my EFI partition is just 95MB, will your latest commit mean that I am unable to use fwupd at all?

hughsie, 2023-02-16:
@noobsauce101 a 95MB partition is quite unsuitable for a dual boot system. For comparison, mine is 600MB. The commit message is slightly misleading, we now require 2*capsule size, plus 20MB.

noobsauce101, 2023-02-16:
Yeah I am looking into resizing it now. Thank you guys for all your help!

mrhpearson, 2023-02-24:
Are we OK to close this issue now?

VannTen, 2023-03-16:
Hi.

I'm getting the same problem (or maybe it's related to the new check introduced in fwupd ?)

#### system info

```
# dmidecode -s system-version
ThinkPad X1 Carbon Gen 9

# fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.2.6-arch1-1
runtime   org.freedesktop.fwupd         1.8.12

# Free space on ESP
df -h /efi
Filesystem      Size  Used Avail Use% Mounted on
/dev/nvme0n1p1  200M   73M  128M  37% /efi
```

<details>

<summary>fwupdmgr get-devices</summary>

```
LENOVO 20XXS3HC0G
│
├─ELAN0672:00 04F3:3187:
│     Device ID:          0867c07f040e810fa5ea153981e63e324c29eb1f
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              ab16bce2-e219-52f5-89a6-fd6b1c4244c0 ← HIDRAW\VEN_04F3&DEV_3187
│                         940805ff-ca7a-56b2-af86-32e995004af9 ← HIDRAW\VEN_04F3&DEV_3187&REV_00
│                         e4c6161e-2b1c-59cb-ad07-c8fb7feb155e ← HIDRAW\VEN_04F3&DEV_3187&MOD_0002
│                         4b808d27-c833-50cc-8177-9a401f5c08fe ← ELANTP\ICTYPE_14
│                         5b73c06e-3dfd-5e6c-b06a-24e3489206c2 ← ELANTP\ICTYPE_14&MOD_0002
│                         31a158de-a671-50bf-b787-3bba647a8c49 ← ELANTP\ICTYPE_14&MOD_0002&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─0000:00:1f.5:
│     Device ID:          b04e387fb80d2b91f37a4d0c7b21461c451775e1
│     Summary:            Memory Technology Device
│     Vendor:             DMI:LENOVO
│     GUIDs:              5f93d7e7-e282-59b9-b663-0146e382f8f6 ← MTD\NAME_0000:00:1f.5
│                         1d5d8ab8-c56e-59b0-9567-d4d306907af8 ← MTD\VENDOR_LENOVO&NAME_0000:00:1f.5
│                         e6be4099-0734-5369-9109-728488385b0a ← MTD\VENDOR_LENOVO&PRODUCT_20XXS3HC0G&NAME_0000:00:1f.5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│   
├─11th Gen Intel Core™ i7-1185G7 @ 3.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000a6
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
[... 573 lines cut ...]
Boot0023* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 0b 01 / 7f ff 04 00
Boot0024* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 0b 00 / 7f ff 04 00
Boot0025* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b 91 af 62 59 56 44 9f 41 a7 b9 1f 4f 89 2a b0 f6 / 7f ff 04 00
Boot0026* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b ae a2 09 0a df de 21 4e 8b 3a 5e 47 18 56 a3 54 / 7f ff 04 00

```

</details>

Message happening when running `fwupdmgr upgrade` (when rebooting):
```
fwupd-efi version1.4
WARNING: Could not apply capsule update: Volume Full
WARNING: Could not apply capsules: Volume Full
Reset System
```

From the previous comment, my EFI partition seems to have enough   space, So I think that's not the issue ?
Also, I'm using a Unified Kernel Image to boot, if that's relevant.

Any ideas ?

gstephan76, 2023-03-16:
Hi,

The issue that I had went away inexplicably. I thought it was due to some
new version in the software stack. Regardless, it follows the data that you
have requested.

BTW, in my case lack of space as pointed out by "df" has never been an
issue.

Thanks for your help,

 root > df -h
Filesystem      Size  Used Avail Use% Mounted on
devtmpfs        4.0M     0  4.0M   0% /dev
tmpfs            16G  114M   16G   1% /dev/shm
tmpfs           6.3G   25M  6.2G   1% /run
/dev/dm-0       473G  197G  273G  42% /
tmpfs            16G  9.3M   16G   1% /tmp
/dev/dm-0       473G  197G  273G  42% /home
/dev/nvme0n1p3  974M  435M  472M  48% /boot
/dev/nvme0n1p1  200M   57M  144M  29% /boot/efi
tmpfs           3.2G  7.1M  3.2G   1% /run/user/1000

root > fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.2.2-300.fc37.x86_64

 root > fwupdmgr get-devices
LENOVO 20TJS2F42V
│
├─CometLake-H GT2 [UHD Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    05
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              3777783a-3f83-56a5-95f4-533eb6a2bd19 ←
PCI\VEN_8086&DEV_9BC4
│                         6c3dbf6c-4e6f-5309-9954-c5ab7aca617e ←
PCI\VEN_8086&DEV_9BC4&REV_05
│                         5fde5d20-db24-5f21-afdd-247c1bf1efa1 ←
PCI\VEN_8086&DEV_9BC4&SUBSYS_17AA22C2
│                         07168636-0f3b-565c-8fe1-0f0a77d82cd8 ←
PCI\VEN_8086&DEV_9BC4&SUBSYS_17AA22C2&REV_05
│                         7ffe1cb7-395a-52a9-a172-70ec6caaf310 ←
PCI\VEN_8086&DEV_9BC4&REV_00
│                         b813dc18-ddf2-508d-a7eb-0e2fc8752b03 ←
PCI\VEN_8086&DEV_9BC4&SUBSYS_17AA22C2&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─Core™ i7-10850H CPU @ 2.70GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ←
CPUID\PRO_0&FAM_06
│                         a45b0522-5722-54bd-b802-86cd044262df ←
CPUID\PRO_0&FAM_06&MOD_A5
│                         7b9b6e8c-226c-5db6-86cb-ea3187578013 ←
CPUID\PRO_0&FAM_06&MOD_A5&STP_2
│     Device Flags:       • Internal device
│
[... 199 lines cut ...]
│     Summary:            UEFI ESRT device
│     Current version:    65794
│     Minimum Version:    65794
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               3fb9a55d-d7f1-4d1b-b216-74e328e28f51
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─UEFI Device Firmware:
      Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
      Summary:            UEFI ESRT device
      Current version:    24577
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               33967546-da89-4c51-9c95-5242bcb854e8
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

capybara-overdose, 2023-05-12:
Yeah this needs to be reopened, its still doing it.

x390 applying System firmware update from 0.1.93 to 0.1.95 - over 700MB spare on the EFI partition.

leleogere, 2023-06-27:
Maybe it can help someone, or maybe I had some luck, but here is what happed to me. I had the "Volume full" error as others, but I noticed that I had a bootable USB stick plugged. Without much hope, I unplugged it to see if it would have an impact, and after another reboot if worked.

I have no idea if unplugging this USB stick was the solution, but I don't think that I did anything else apart from that. Maybe something else happened at the sale time?

hughsie, 2023-06-27:
@leleogere it's certainly plausible; we added https://github.com/fwupd/fwupd/pull/5900 recently to use a better algorithm for choosing an ESP.

soig, 2023-10-04:
> Maybe it can help someone, or maybe I had some luck, but here is what happed to me. I had the "Volume full" error as others, but I noticed that I had a bootable USB stick plugged. Without much hope, [...]

Indeed that fixed the issue for me.
I'd still the install USB key plugged in the laptop dock.
Removing it made that error message to disapear.
Looks like a big issue with Lenovo firwmare.
Someone ought to report to Lenovo that their BIOS/UEFI has this issue

susman, 2024-01-26:
FYI I ended up resizing EFI partition from ~200M to ~300M and the issue went away.
I think the main problem for me was the "Volume full" message. It's a little cryptic: not clear which "volume" that is and "full" is something that really thrown me off the investigation path, I could clearly see that my ESP is not full. A message like "Not enough free space on ESP" would be much more informative.

mrhpearson, 2024-03-20:
Just a note as I hit this issue myself, and Google pointed at this bug. I scratched my head, and figured it out.
I had a bootable USB in my system (just did a fresh install of Fedora) and that was the problem. As soon as I unplugged the USB the update worked smoothly.

Not sure exactly why - will need to ask the FW team - but I assume that it tries to unpack to the USB and runs out of space?

hughsie, 2024-03-20:
> As soon as I unplugged the USB the update worked smoothly.

Older versions of fwupd didn't have a great default for choosing *which* ESP should be used by default. If you can reproduce this with 1.9.14 or newer please let us know!

jibsaramnim, 2024-05-07:
> Older versions of fwupd didn't have a great default for choosing _which_ ESP should be used by default. If you can reproduce this with 1.9.14 or newer please let us know!

I'm not the person you quote-replied, but I ran into this very issue today after doing a fresh re-install of Fedora Silverblue on my Thinkpad T480. The EFI partition was pretty much completely empty (587MB, or 97%, free), yet it too was complaining about not having enough space to do the system firmware update. 

I have a secondary SSD in my laptop that still had a test *BSD installation on it with a ~200mb or so EFI partition. After formatting this drive and trying again, the firmware upgrade went smoothly.

The drive Fedora is installed to is configured as the primary drive (ie. the only drive to boot from), but the secondary drive is named `nvme0n1` and the primary drive `nvme1n1`, so if whatever decides what drive's EFI partition to pick uses something like a named sort, it would show up first I suppose.

This is with `fwupdmgr` version `1.9.19`.

I hope this helps!

hughsie, 2024-05-07:
@jibsaramnim can you grab us the output of `sudo fwupdtool get-devices --plugins uefi_capsule -vv` and output of `sudo fwupdtool esp-list` for both volumes please.

jibsaramnim, 2024-05-07:
@hughsie I'd love to, but as I mentioned the secondary drive just had a test installation (with accompanying EFI partition) that I have nuked already to get the firmware updater to successfully do its thing. In case it helps, here are the results from my Fedora installation:

<details>
<summary>fwupdtool get-devices --plugins uefi_capsule -vv</summary>

```
09:32:11.380 FuDebug              verbose to debug (on console 1)
09:32:11.380 FuEngine             starting fwupd 1.9.19…
09:32:11.381 FuMain               locked /run/lock/fwupdtool
09:32:11.383 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [ -                                     ]09:32:11.383 FuConfig             loading config /etc/fwupd/fwupd.conf
09:32:11.383 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
09:32:11.383 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
09:32:11.383 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
09:32:11.384 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
09:32:11.384 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
09:32:11.384 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
09:32:11.384 FuConfig             ::configuration loaded
09:32:11.384 FuRemoteList         loading /usr/share/fwupd/metainfo
09:32:11.384 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
09:32:11.384 XbSilo               file: 8ed11c21-0797-f3c3-6520-6acaba8284b8, current:8ed11c21-0797-f3c3-6520-6acaba8284b8
09:32:11.384 XbSilo               loading silo with existing file contents
09:32:11.384 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
09:32:11.384 FuRemoteList         ::remote_list changed
09:32:11.384 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
09:32:11.385 FuRemoteList         ::remote_list changed
09:32:11.385 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
09:32:11.385 FuRemoteList         ::remote_list changed
09:32:11.385 FuEngine             priority local and vendor-directory is not download remote, so bumping
09:32:11.385 FuRemoteList         ordering lvfs-testing=lvfs+1
09:32:11.385 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]09:32:11.385 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
09:32:11.385 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:32:11.385 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
09:32:11.385 Jcat                 loaded 1 certificates
09:32:11.385 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
09:32:11.385 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:32:11.385 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
09:32:11.385 Jcat                 loaded 1 certificates
09:32:11.385 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
09:32:11.385 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
09:32:11.385 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]09:32:11.385 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
09:32:11.386 FuHistory            got schema version of 12
Loading…                 [*                                      ]09:32:11.392 FuPlugin             load(/usr/lib64/fwupd-1.9.19/libfu_plugin_modem_manager.so)
09:32:11.393 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
09:32:11.393 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
09:32:11.393 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
09:32:11.393 XbSilo               file: e2684fec-eccc-91c2-02ff-5880d8a570a3, current:e2684fec-eccc-91c2-02ff-5880d8a570a3
09:32:11.393 XbSilo               loading silo with existing file contents
Loading…                 [*                                      ]09:32:11.394 FuConfig             loading config /etc/fwupd/fwupd.conf
09:32:11.394 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
09:32:11.394 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
09:32:11.394 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
09:32:11.394 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
09:32:11.394 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
09:32:11.394 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
09:32:11.395 FuConfig             ::configuration loaded
09:32:11.395 FuStruct             SmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0x6f
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x0
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0xc07
  structure_table_addr: 0x6f026000
09:32:11.395 FuStruct             SmbiosStructure:
[... 813 lines cut ...]
```
</details>

<details>
[... cut ...]
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme1n1p1
/boot/efi/EFI/BOOT/fbia32.efi
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/BOOT/BOOTIA32.EFI
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/fedora/BOOTIA32.CSV
/boot/efi/EFI/fedora/BOOTX64.CSV
/boot/efi/EFI/fedora/mmia32.efi
/boot/efi/EFI/fedora/shimia32.efi
/boot/efi/EFI/fedora/mmx64.efi
/boot/efi/EFI/fedora/shim.efi
/boot/efi/EFI/fedora/shimx64.efi
/boot/efi/EFI/fedora/grubia32.efi
/boot/efi/EFI/fedora/grubx64.efi
/boot/efi/EFI/fedora/grub.cfg
/boot/efi/EFI/fedora/fwupdx64.efi
```
</details>

tianyu, 2024-07-30:
FYI, I was able to get past this error by taking out the USB Install disk from the USB port!