fwupd/firmware-lenovo #258: fwupdmgr shows incorrect ME Firmware upgrade for ThinkPad T14 Gen3

taoky, 2022-08-17:
When running `fwupdmgr get-updates`, it tells me that the Intel Management Engine on my laptop needs to be upgraded:

```
> fwupdmgr get-updates                                                                                                                                                                          
Devices with no available firmware updates: 
 • ELAN0676:00 04F3:3195
 • INTEL SSDPEKNU020TZ
 • Integrated RGB Camera
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Prometheus
LENOVO 21AHA014CD
│
├─Intel Management Engine:
│ │   Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.15.1735
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               23192307-d667-4bdf-af1a-6059db171246
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad X1 Yoga Gen 7/ThinkPad X1 Carbon Gen 10:
│       New version:      0.15.1778
│       Remote ID:        lvfs
│       Release ID:       13650
│       Summary:          Lenovo ThinkPad X1 Yoga Gen 7/X1 Carbon Gen 10 Consumer ME Firmware
│       License:          Proprietary
│       Size:             3.7 MB
│       Created:          2016-07-08
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N3ARG03W
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:      
│       AlderLake Production Release 4 (Best Known Configuration)
│       
│       Version 16.0.15.1778 (LVFS: 0.15.1778)
│       
│       This is an Initial Release for LVFS
│     
└─System Firmware:
  │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
  │   Summary:            UEFI ESRT device
  │   Current version:    0.1.1
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              30fbf24a-e1da-4840-84f6-6d7deb630e1d
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14 Gen 3 / P14s Gen 3 / T16 Gen 1 / P16s Gen 1:
        New version:      0.1.03
        Remote ID:        lvfs
        Release ID:       14830
        Summary:          Lenovo ThinkPad T14 Gen 3 / P14s Gen 3 / T16 Gen 1 / P16s Gen 1 System Firmware
        License:          Proprietary
        Size:             22.0 MB
        Created:          2022-06-10
        Urgency:          Critical
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Updated includes a security fix.

```

However, my laptop is ThinkPad T14 Gen3 -- neither ThinkPad X1 Yoga Gen 7 nor ThinkPad X1 Carbon Gen 10. [This page from Lenovo](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14-gen-3-type-21ah-21aj/downloads/ds556721-intel-management-engine-160-firmware-for-windows-11-version-21h2-or-later-10-version-20h2-or-later-thinkpad-t14-gen-3-t16-gen-1-p14s-gen-3-p16s-gen-1?category=Motherboard%20Devices%20%28core%20chipset,%20onboard%20video,%20PCIe%20switches%29) shows that the ME upgrade suitable for t14gen3 should be "n3brg02w" instead of "n3arg03w".

And this is what `fwupdmgr get-devices` tells about the Intel ME device:

```
├─Intel Management Engine:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.15.1735
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               23192307-d667-4bdf-af1a-6059db171246
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
```

kmauleon, 2022-08-19:
hi @taoky thank you for reporting we are looking into this... will feedback once we have update.  Thank you very much...

kmauleon, 2022-08-31:
hi @taoky can you please check if X1 Carbon Gen 10 MEFW Consumer update is still being offered to you?
Please advise.  Thank you very much...

taoky, 2022-08-31:
> hi @taoky can you please check if X1 Carbon Gen 10 MEFW Consumer update is still being offered to you? Please advise. Thank you very much...

It's no longer provided after refreshing metadata. Thank you.

closed 2022-08-31 by taoky

kmauleon, 2022-08-31:
thank you very much for the confirmation @taoky !