fwupd/firmware-lenovo #261: IME update and disk encryption

willismonroe, 2022-08-17:
I have an update waiting for the IME but it says "Note: Please suspend or disable any disk encryption tool prior to update." My entire SSD is encrypted (w/ luks) and as far as I can tell, there's really no way to remove or suspend the encryption. Am I just stuck on this version of the IME, or is there some way to install the update without suspending/removing the disk encryption.

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 192.72.1757 to 192.85.1869?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 0 Intel Platform Update 2021.2 Product Version Maintenance Release         ║
║                                                                              ║
║ Version 12.0.85.1869 (LVFS: 192.85.1869)                                     ║
║                                                                              ║
║ Problem Fixes                                                                ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities under issues. Please      ║
║ see fixed issues for details.                                                ║
║                                                                              ║
║ Note: Please suspend or disable any disk encryption tool prior to update.    ║
║                                                                              ║
║ 20QDCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
```

kmauleon, 2022-08-18:
hi @willismonroe you may proceed with the update... there's a similar scenario as yours and they seem to update their ME successfully... this is just a general reminder ... thank you

willismonroe, 2022-08-18:
Just reporting that the upgrade worked despite an active LUKS encrypted
disk.

kmauleon, 2022-08-19:
hi @willismonroe thank you for the update... will close this ticket but please reopen if any concern... thank you very much.

closed 2022-08-19 by kmauleon