fwupd/firmware-lenovo #263: Lenovo Thinkpad T480: Insecure configuration according to `fwupdmgr security`

VorpalBlade, 2022-08-28:
As suggested by @hughsie  in https://github.com/fwupd/fwupd/issues/4959#issuecomment-1229460665 I posted this as a separate bug report here.

On my Lenovo Thinkpad T480 `fwupdmgr security` reports some issues that I really would expect not to be the case, especially since some of those should then have been fixed via UEFI updates.

```
fwupdmgr security           
Host Security ID: HSI:0! (v1.8.3)

HSI-1
✔ CSME override:                 Locked
✔ CSME v0:11.8.92.4222:          Valid
✔ Platform Debugging:            Disabled
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✘ CSME manufacturing mode:       Unlocked
✘ SPI BIOS Descriptor:           Invalid

HSI-2
✔ IOMMU:                         Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard verified boot: Valid
✔ Platform Debugging:            Locked
✘ Intel BootGuard OTP fuse:      Invalid
✘ TPM PCR0 reconstruction:       Not found

HSI-3
✔ Intel BootGuard error policy:  Valid
✘ Intel CET Enabled:             Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✘ Linux kernel:                  Tainted
✘ Linux kernel lockdown:         Disabled
✘ Linux swap:                    Unencrypted
✘ UEFI secure boot:              Disabled

This system has a low HSI security level.
 » https://github.com/fwupd/fwupd/wiki/Low-host-security-level

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues

Host Security Events
  2022-07-18 09:17:35:  ✔ IOMMU device protection enabled
  2022-04-29 08:21:10:  ✘ MEI manufacturing mode changed: Locked → Unlocked
  2022-04-29 08:21:10:  ✘ SPI BIOS Descriptor changed: Locked → Invalid
  2022-04-29 08:21:10:  ✘ Intel BootGuard OTP fuse changed: Valid → Invalid
  2022-04-29 08:21:10:  ✔ Platform Debugging appeared: Disabled
  2022-04-29 08:21:10:  ✔ Supported CPU appeared: Valid
  2022-04-29 08:21:10:  ✔ Platform Debugging appeared: Locked
  2022-04-29 08:21:10:  ✘ Pre-boot DMA protection appeared: Disabled
  2022-02-01 16:37:06:  ✔ Intel DCI debugger disappeared: Disabled
  2022-02-01 16:37:06:  ✔ Intel DCI debugger disappeared: Locked
  2022-02-01 16:37:06:  ✘ Pre-boot DMA protection disappeared: Disabled
```

I have intentionally disabled "Secure Boot" as I run Arch Linux and don't see the benefit of going through the hassle of setting up secure boot (the system is physically secure). However there are plenty of strange things unrelated to secure boot:

* At HSI-1
  * "CSME manufacturing mode" should not be unlocked.
  * "SPI BIOS Descriptior" also seems bad.
* At HSI-2:
  * I had to manually enable the IOMMU using `intel_iommu=on` on the kernel command line. VT-x and VT-d are both on in UEFI settings, so that is odd.
  * Why would the BootGuard fuse be unset?
  * Why is TPM PCR0 not found?
 * At HSI-3:
   * I suspect Pre-boot DMA protection is disabled due to IOMMU being disabled before I forced it on with Linux kernel command line. Still, this is odd.
 
Ignore everything under "Runtime Suffix", I run nvidia drivers and some other DKMS modules (thus tainted kernel), I actively avoid the lockdown feature (makes kernel development a pain for a start), fwupdmgr fails at detecting swap on LVM2 and as I said earlier I have disabled secure boot intentionally.

```
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.19.4-zen1-1-zen
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10
$ dmidecode | grep -EA 8 "^System Information" 
System Information
	Manufacturer: LENOVO
	Product Name: 20L5CT01WW
	Version: ThinkPad T480
	Serial Number: <snipped for privacy>
	UUID: <snipped for privacy>
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_20L5_BU_Think_FM_ThinkPad T480
	Family: ThinkPad T480
```

In case it matters, this T480 was ordered via the Swedish Lenovo web site configuration tool. It is an i7 with nVidia graphics.

As of writing this fwupdmgr reports I'm on the latest firmware (0.1.45).

hughsie, 2022-08-28:
@mrhpearson is it plausible these machines have `CSME manufacturing mode` left enabled? We're reading HFSTS1 if that helps the internal ticket.

mrhpearson, 2022-09-12:
I don't know I'm afraid - seems odd. Raised internal ticket LO-2012 to the FW team.
As a note - my apologies - I completely missed this issue in my Inbox swamp :(

VorpalBlade, 2022-09-12:
I should mention that the motherboard was replaced on-site under warranty recently (weeks before this bug report was made) due to a broken thunderbolt controller. I'm not sure if that is relevant.

However this does *not* correspond to the dates from the fwupd event log (`2022-04-29 08:21:10:  ✘ MEI manufacturing mode changed: Locked → Unlocked`) etc. The warranty replacement happened in late july or early august from my memory.

VorpalBlade, 2022-09-13:
> Hi @VorpalBlade - would you mind sending me some details to my markpearson at lenovo dot com address please?
> 
> I need your serial number and any details you're able to share about the warranty work (dates/where it was done/etc). The support team are going to follow up on this as your system should not have been sent out with unlocked FW on it so we need to find out how that happened.

Sure. However as I said, I do not believe it to be related given the timings of the entries in the event log. Guess it doesn't hurt though.

VorpalBlade, 2022-09-14:
@mrhpearson I tried to send you an email, but apparently your email system rejects my selfhosted email server:

    lenovo-com.mail.protection.outlook.com[104.47.26.10] said: 550 5.4.1
    Recipient address rejected: Access denied. AS(201806281)
    [PSAAPC01FT011.eop-APC01.prod.protection.outlook.com] (in reply to RCPT TO
    command)

So until you fix that, there is nothing I can do.

mrhpearson, 2022-09-14:
Wow....that's strange. I guess we'll use my personal email if that's OK? banther at gmail. Dealing with IT on this is going to be an exercise in pain.
Outlook sucks....I loathe it (mind you - lotus notes which they used at IBM is even worse :D)

The serial number detail is so we can check where the update was done and who did it and find out if that is where the issue was introduced - because it really shouldn't have shipped like that.

VorpalBlade, 2022-09-14:
That is probably the sensible option (sidestepping IT). I have some vague memories of using Lotus Notes early in my career, but I can't say I remember much of it, other than it being generally awkward and difficult to use. Especially the groupware document handling features.

VorpalBlade, 2022-10-13:
Any updates on this? I haven't heard back from when I sent the emails.

mrhpearson, 2022-10-14:
Afraid not - I've sent another nag to the team in Europe (but got an OOO so not expecting a quick reply).

VorpalBlade, 2022-10-18:
I thought I should let you know that the charging on that replacement motherboard died yesterday evening and I have now created another service ticket for replacement.

Once that goes back to Lenovo you may want to pull it and prevent it from getting repaired and going out to another customer.

mrhpearson, 2022-10-18:
OK - thank for the note. They're ignoring me at the moment - not impressed :/

mcatanzaro, 2024-12-05:
I guess it's probably too late now, but would be interesting to see any update on this.