fwupd/firmware-lenovo #264: Thunderbolt controller firmware update - X1 Carbon gen6

LordChaos73, 2022-08-28:
Hi,

I'm one of the many owners with this laptop and I've lost HDMI output. I'm not sure when it exactly happened, but I think it was when I updated the Intel ME firmware a month or three ago. Basically, nothing happens when an external monitor is connected.
I've read a lot about this particular issue on (Lenovo) forums and it seems the following firmware update can fix this issue:

https://pcsupport.lenovo.com/nl/en/products/LAPTOPS-AND-NETBOOKS/THINKPAD-X-SERIES-LAPTOPS/THINKPAD-X1-CARBON-6TH-GEN-TYPE-20KH-20KG/20KH/downloads/DS502615

The laptop has the latest BIOS update (1.57) and the latest Intel ME firmware installed as well.
The thing is, my laptop is running Arch Linux solely. Could this update be made available for Linux users (fwupdmgr) as well?

hughsie, 2022-08-29:
@LordChaos73 what firmware version are  missing? Thanks.

LordChaos73, 2022-08-29:
@hughsie My question is: is the thunderbolt controller firmware update (as linked above) available on the LVFS platform?

mrhpearson, 2022-08-29:
Hmmm - looks like we last updated the TBT controller on the X1 Carbon 6 two years ago.. with version 43.
I'll check with the FW team when (hopefully not if) the next update is - it looks like they're now up to version 47.

All that being said - I don't see why a TBT controller update would have any impact on HDMI display - unless you're connected via a TBT dock maybe? Any reason you think this particular FW update will help? I'm happy to chase an update (it should be done) but I don't think it's going to fix the root cause of your issue.

I've created internal ticket  LO-1984 for tracking

LordChaos73, 2022-08-30:
@mrhpearson please see the following posts:

https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/X1-Carbon-6th-gen-HDMI-Thunderbolt-stopped-working/m-p/5120432

https://forums.lenovo.com/t5/thinkpad-x-series-laptops/x1-carbon-6th-gen-hdmi-not-working/td-p/4091511

dal00, 2022-09-07:
I have the same issue with thunderbolt not working and hdmi also having issues. I'm even unable to see thunderbolt in lspci.
Windows can't see it in the device manager too and I'm unable to flash it from windows. 

@LordChaos73 Are you even able to see the thunderbolt controller in lspci?

mrhpearson, 2022-09-07:
Do you know which version of FW you had on there previously? There was a critical update a while back (2019) that was published and I'm pretty sure we put those updates on LVFS.
https://support.lenovo.com/ca/en/solutions/ht508988-critical-intel-thunderbolt-software-and-firmware-updates-thinkpad

My understanding was that if the FW wasn't updated it was possible to brick your TBT controller.
If you already had that update then we can ignore that as a theory. If it's not working in Windows it is worth calling customer support for assistance - you may have a HW issue.

LordChaos73, 2022-09-07:
I remember about the critical update back in 2019 and I've always had the latest possible firmware installed since. Since so many X1C6 owners seem to have this issue, I'm wondering if the patch was successful in the first place.
Anyway, since this was my production machine until very recently, I know everything was working fine until we had the Intel ME issues that failed to install. I think I lost HDMI output around that time. And if I remember correctly, there was also a new BIOS release at the time.
I'm also not seeing my Thunderbolt controller using lspci (I've attached the output of 'sudo lspci -v').
[lscpi.txt](https://github.com/fwupd/firmware-lenovo/files/9515388/lscpi.txt)

mrhpearson, 2022-09-12:
Hi - can you confirm the BIOS, EC and ME FW versions please (easiest is to grab them from the BIOS setup screen - F1 during early boot). The note from the FW team says:

_I checked history, there was an issue that system will lost TBT controller and fixed by one EC firmware on X1 Carbon G8 project, never hear this issue on X1 Carbon G6 project. So that is why I suggest to have a try with upgrade BIOS/EC to newest, feel sorry that cannot give specifics of which version of EC._

Not sure why they can't tell me which version of FW...(I'm pretty annoyed about it, but it gets worse)

It looks like they released a new BIOS/EC combo update on the Lenovo support site - **but the EC has not been released on LVFS** (the BIOS has). I'm really grumpy about that (the BIOS is updated)

Can you try the cab file that is on the Lenovo site (https://download.lenovo.com/pccbbs/mobiles/n2wul23w.zip). I did the update on mine ('fwupdmgr install N2WHT20W.cab' - I had to set 'OnlyTrusted=false' in /etc/fwupd/daemon.conf) and it looks good - but I also wasn't having TBT issues.

I don't know if the new EC will fix the issue - there is nothing in the internal release notes that references it - but it sounds like it should be tried. I'm pushing them to get the EC on LVFS ASAP.

Mark

LordChaos73, 2022-09-12:
[image]

I´m not sure the cab files you provided are the correct ones?

[levid@x1carbon6 n2wul23w]$ fwupdmgr install N2WHT20W.cab
Decompressing…           [ -                                     ]
No supported devices found

[levid@x1carbon6 n2wul23w]$ fwupdmgr install N2WET37W.cab
Decompressing…           [    \                                  ]
No supported devices found

LordChaos73, 2022-09-12:
By the way, I downloaded the ISO of the latest BIOS and wanted to see if I could re-apply it. So I created a bootable USB stick with the latest BIOS, but I can't seem to re-apply it; the setup program states there is nothing to do. Is there any way to re-apply the BIOS update? I was hoping the ISO _is_ carrying the EC update as well.

mrhpearson, 2022-09-13:
Sorry - those are X1Carbon 8 items...my bad.
I'll look for the X1 Carbon 6 ones.

mrhpearson, 2022-09-16:
I checked and the X1 Carbon 6 versions I can find are the same you already have - so I went back to the FW team and asked for any other suggestions. Not heard back yet.

LordChaos73, 2022-09-26:
@mrhpearson what about the EC update that was never provided on lvfs?

mrhpearson, 2022-09-26:
EC 1.22 is here (and under stable) https://fwupd.org/lvfs/firmware/5728

Checking the released notes on the support site it's the latest version:
   https://download.lenovo.com/pccbbs/mobiles/n23ul35w.txt:
   1.57      (N23UL35W)  1.57 (N23ET82W)      1.22 (N23HT37W)   01    2022/08/25

LordChaos73, 2022-10-06:
Is there any hope on restoring Thunderbolt/HDMI functionality? A BIOS update maybe?

LordChaos73, 2022-10-06:
I see there's a new version of the Intel ME available - 11.8.93.4323. It's not on LVFS yet?

kmauleon, 2022-11-18:
> I see there's a new version of the Intel ME available - 11.8.93.4323. It's not on LVFS yet?

hi @LordChaos73 still under testing... hopefully will be available by end of this month... please open separate issue for proper tracking. thank you very much

RAY5D, 2022-11-19:
@cdrn Same issue here. I'm trying the full hardware route by buying W25Q80DVSNIG chips to replace the one on my motherboard that has gone bad, and programming them through a CH341-based programmer. Before the fix, the laptop was not shutting down properly by Windows, could only pull 15W from the charger instead of 65W, and had no HDMI output. The binary file provided by @dal00 fixed the charging issue, but nothing else. I'm not sure if that's a compatibility issue between BIOS, EC, and TB versions or there are other problems. I have BIOS 1.57 and EC 1.22, which I think is the latest. I just need to find a working binary for the TB controller.

tw0flower, 2022-11-20:
I also have this issue (HDMI and DP not working). When I first noticed it I updated all firwmares through fwupd so I have everything on the latest version now. I have no idea if the critical fix from 2019 was applied on this computer as it is second-hand so I'm not very hopeful in my case. Does anyone know if it was rolled out automatically through Windows Update?

I'm in the process of getting an EEPROM programmer myself, because the controller does not even appear in lspci output or in Windows Device Manager. 

Do you still have the original data that was on your chip before you flashed it by any chance @RAY5D ? I'd be interested in comparing it to what I have. It's interesting that after changing the chip for a new one you still seem to face some issues.

RAY5D, 2022-11-20:
@tw0flower Sorry, I don't have any data backed up. When I used the programmer on the old chip, it read all 0xff, which I think is the "default" value for this type of storage. It will still read 0xff even after writing whatever data, so the chip is most likely dead as dust since the programmer could still read its model number, indicating a correct connection. I'm pretty sure the only thing I need now is the binary file, but that is, unexpectedly, hard to find. Can you post your data here so I can try? Thanks!

tw0flower, 2022-11-21:
@RAY5D Ah, I misunderstood your comment. I thought you had already replaced the chip, and that some other component was writing 0xFF data all over the new chip after you flashed it.

I will post my data, but I don't think flashing it will be useful as my computer also has the HDMI/DP problem. Have you checked these binary files on the LVFS website? ﻿https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23TF.firmware

During my desperate search for answers on this issue, I also found [this blog post](https://nadim.computer/posts/2018-10-26-thinkpad.html) from a user that had a somewhat similar problem years ago on a Thinkpad P1. He apparently had to flash zeroes over the chip before being able to flash the actual firmware.

I goes without saying, I don't know much about hardware so anyone reading this please take it with a grain of salt. At this point I'm desperate and accepting the risk of bricking things.

RAY5D, 2022-11-21:
@tw0flower I did replace the chip and flash the firmware. I also saw that post earlier, but apparently I didn't notice the part about writing 0's first and then flashing the actual firmware (oops). I'll do more testing with the binary files I found and post updates here. However one thing that confused me was that the official binary files from both Lenovo and https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN23TF.firmware are exactly 260KB while the chip is 1MB, and the dumped binary in this issue has some content after that region, so I'm not sure about what should I write after the 260KB.

tw0flower, 2022-11-21:
@RAY5D That's what I hope I will find out if I can still extract some data from my chip. There might be some padding involved. I wonder if this information might not be available somewhere in the LVFS files? Because the flashing software must be aware of it right?

But in theory if you zero-out the chip first, then flash the firmware, this would ensure that only the firmware on the chip I think. If there is some padding involved you could also try writing it at the end of the memory by concatenating some zeroes followed by the 260kb file so that the total is 1MB. If the padding is some arbitrary value however we have to find another way.

RAY5D, 2022-11-21:
@tw0flower OK I did some more testing. I had 2 binary files, the first one was the 260KB official release, with the extra space filled with 0's, the second one is the dumped data from this issue posted by [dal00](https://github.com/dal00). For each test case, I filled the chip with a certain value (0x00 or 0xff), started the PC, shutdown, flashed the binary, and start again to see if it works. Here are the results: 

1. 0x00 fill, 260KB+0's: No HDMI, charging works, Windows couldn't shutdown properly. 
2. 0xff fill, 260KB+0's: No HDMI, charging works, Windows couldn't shutdown properly. 
3. 0x00 fill, dumped: No HDMI, charging works, Windows couldn't shutdown, sometimes freezes and goes into BSOD. 
4. 0xff fill, dumped: Windows did not complete boot, black screen at login, tried Manjaro, no HDMI. 

The results looked overall similar to my previous attempts. Only the charging speed was fixed. However, before the tests today, I noticed that the replacement chip was dead again (wtf!?!?). I think I tried enabling assist mode in BIOS earlier, maybe that caused the early death? Anyways, I just replaced the chip again before the tests since I bought 10, and was careful to only enable assist mode while using Linux, and the chip still works for now. I don't think there is a lot left to do before we can find more binaries to compare and test with. Should I try the official binary but extended with 0xff next time?

tw0flower, 2022-11-23:
I got my CH341 SPI today and I found a few things. 

First of all @RAY5D if you only see 0xFF bytes on your chip it might be that your clamp is not properly attached. I had this issue at first, I solved it by re-attaching it until I read some data and by using flashrom instead of ch341eeprom. If the RUN led on your SPI board is not on when reading/writing it means something is wrong in your setup.

The first thing I did was to extract the firmware I had installed on my machine. Compared to the firmwares I found online, it had an offset: the first 4096 bytes were all 0xFF. What remained was similar to the firmares I found online.

My initial situation is that everything worked except plugging a second display (HDMI or via Thunderbolt).
I then tried the following:
1.  I flashed the newest 47 firmware from the LVFS website. No changes.
2. I flashed the 47 firmware with a 4096 bytes offset. Charging was not using the full power from the adapter. 
3. I flashed the 43 firmware from the LVFS website. Charging was working normally again but no luck with the screen.
4. I went in the BIOS and reset the options to default with Windows 10 optimised settings. But unfortunately that didn't help.

So in my case:
- My problem is not cause by a hardware defect from the chip that stores the Thunderbolt firmware. I can read and write on it perfectly. I even made sure that reading after writing a file produced the exact same result.
- Changing the firmware does have an effect. I lost charging. But it had no effect on the external display.
- There was an offset on my firmware. I wasn't compatible with the 47 version. I have not tried it on the 43 version, but it made charging work again without it.

The files that I flashed on the chip are in the attached zip file.
[firmwares_x1cgen6.zip](https://github.com/fwupd/firmware-lenovo/files/10074138/firmwares_x1cgen6.zip)

RAY5D, 2022-11-24:
@tw0flower Pretty sure I'm indeed getting dead chips since I always check if the programmer can read the chip's model number as an indication of good connection. However, when I tried your binary, the result is basically the same as my previous attempts. Only charging works, still no HDMI, and Windows still couldn't shutdown the laptop properly. Also, the binary file from [dal00](https://github.com/dal00) has something in the first 4096B. I'm not sure why yours are just 0xff in that region, it looks like those binaries are more complex than the flash-and-done type of job I thought it would be... The only thing I can think of right now is to get a TB dock or something, so I can "activate" the TB controller by plugging it in. That way I can try to update the firmware through the OS. However, this will probably not work given that the controller is barely working. Did you try the binary in here?

> @mrhpearson I'm not sure as I'm unable to connect to the controller in any way that I know of.
> 
> But here is a dump of the fw from the W25Q80DVSNIG [x1c6_tb3.zip](https://github.com/fwupd/firmware-lenovo/files/9517570/x1c6_tb3.zip)

jnada2021, 2022-11-26:
Hello, I am the person who opened the case with Lenovo forums (the one in the first message on this thread) and I also have a post in reddit about this issue dated last year. 

I have the latest BIOS 1.58 firmware which I flashed today. Latest Intel ME Firmware and software installed.
I flash all the new TB firmwares posted in this forum using my  CH341 SPI with AS programmer. 
As with everyone, only the charging works and also with a USBC hub or USBC to USB A adapter I can connect basic USB devices (such as mouse etc.) but no Display from TB/USBC ports nor HDMI.
If it's any useful, here are the collection of all of the original Lenovo TBT firmwares from 12 to 24

I also have historical archive of X1 6C bioses starting with 19 to 31 which I had downloaded from Lenovo website. I can upload to this post due to 161MB filesize. If it is useful for someone (to roll back etc.) I can also upload to  google drive.

jnada2021, 2022-11-26:
And here is the latest N23TF24W firmware bin dated 25 May 2022 which I flashed as 260kb (only charging and basic USB work-No DP, No HDMI still) and read the chip again and saved 1MB this time.
[20221126_N23TF24W.zip](https://github.com/fwupd/firmware-lenovo/files/10096105/20221126_N23TF24W.zip)
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-6th-gen-type-20kh-20kg/downloads/driver-list/component?name=USB%20Device,%20FireWire,%20IEEE%201394,%20Thunderbolt

IIRC the original TB firmware had the same info written twice one in the first half and the second instance is the latter part. I am not sure if my bin was corrupted and had repeated twice or it is because we have 2 TB ports. But someting to think about (may be)

dal00, 2022-11-26:
There used to be separate .bins for assisted mode and native mode. You can find them in the extracted firmwares this guy collected if I remember correctly. https://www.badcaps.net/forum/printthread.php?t=82556

jnada2021, 2022-11-26:
>     * My problem is not caused by a hardware defect from the chip that stores the Thunderbolt firmware. I can read and write on it perfectly. I even made sure that reading after writing a file produced the exact same result.
> 
>     * Changing the firmware does have an effect. I lost charging. But it had no effect on the external display.
 

Same as you. I can read and write to the W25Q80DVSNIG chip with the CH341-based programmer. As mentioned. Erasing the chip (with AsProgrammer for example) or writing the wrong *.bin file disables the charging. But if I flash a TBT.bin from Lenovo etc. charging and basic USB functions are back (USB 2 mouse, USB harddisk, connect to Sony camera etc. but no TB, no DP and no HDMI at all.

I think the intel firmware messed up something in someother area and we are flashing for nothing. I wish we all could "factory reset" the X1 C6 to back to its 2018 out of factory state (with OS, BIOS etc.) so that we could start all over again.

Does anyone know in which chip Intel ME firmware is written at and we can also attack to that chip to erase/reset.

Lenovo has been very mum on this issue and I think the are deliberately downplaying since perhaps only some limited motherboard, chip combinations are affected or it might be a much bigger issue that they play dumb and hope that it will go away.

jnada2021, 2022-11-27:
Some updates.

1.
I think Lenovo's made some changes in their latest 1.58 BIOS firmware so that even if the TB chip is corrupted or "empty" charging function works.
I have just loaded a zero.bin file to erase the TB chip and rebooted to test, and charging still works. In the past it would not and would at least give the low power warning before and/or don't charge.
Here is the zero.bin file if you want to test:

[zero_erase.zip](https://github.com/fwupd/firmware-lenovo/files/10098020/zero_erase.zip)

2.
I opened my very first original .bin file dated 30/12/2021 with HxD hex editor and the 260kb info has been written on it twice. 
And the very first one starts at line 00004096:
[image]

And the second at line: 00532480:
[image]

The more I look at this the more I think that the "corruption/problem" etc. is inside the intel ME chip/firmware whose contents probably are classified as "state secret" :) LOL

RAY5D, 2022-11-27:
[image]
The BIOS and ME firmware is stored in another flash chip right above the CPU, near the fan connector. I think it's also readable with the same programmers for the TB chip. But I'm not sure how that has an affect on our issues...

jnada2021, 2022-11-28:
IIRC the Lenovo "2019 fix" was to disable Intel ME writing something somewhere but even that was also inevitable. Remember  them saying "after applying the fix" don't ever update intel ME ever again, which makes me believe above conclusion. Unbenownst to many Intel ME is a "master OS" it runs concurrently with any computer regardless of it being Linux, Windows etc.

Were we able to extract any known good copy of the TBT bin file from someone who doesn't have TB, DP, HDMI problems yet? What dall00 shared in Spet 7th is also a "bad bin" since he extracted them after having the issue.

RAY5D, 2022-11-28:
@jnada2021 Huh, that's interesting. Can you still find the article that mentioned it? Also, even if we manage to find someone with a fully working computer and extract both the ME and TB firmware, is it as easy as just flashing it into our chips? I would assume there are some bullshit encryption going on that bounds the binary with a single motherboard or something.

jnada2021, 2022-11-28:
1. These are the articles:
https://www.notebookcheck.net/ThinkPad-Thunderbolt-3-failure-What-s-happening-why-it-s-happening-and-how-to-fix-it.451207.0.html
https://www.reddit.com/r/thinkpad/comments/eszg2o/thinkpad_tb3_failure_what_we_know_repair/

Search: DO NOT FORCE UPDATE Thunderbolt Controller. This may damage the firmware.

2. By finding a "working bin" we at least eliminate one (important) possibility I am hoping. I am not sure of any encyription etc. since this guy was able to flash it with the programmer as we are trying but his machine is not a X1.
https://nadim.computer/posts/2018-10-26-thinkpad.html

jnada2021, 2022-11-28:
Oh BTW if any of you are in US:
https://chimicles.com/lenovo-thinkpad-usb-c-charging-defect-class-action-investigation/

LordChaos73, 2022-11-28:
I've always had my laptop on the latest (lvfs) updates; I haven't ran Windows in years. Still, I've lost HDMI/TB3 output a few months ago.
I'd like to thank @jnada2021 and @RAY5D for all their hard work regarding this issue.
Where is Lenovo in all of this? I have a (out of warranty) laptop I no longer can use at work since I no longer can connect it to an external display. I'd like to know if this issue is fixable through some firmware updates. Or should I buy a new machine?

tw0flower, 2022-11-28:
Thank you everyone for looking in this issue. It seems a lot of people are affected by this. What bugs me is that I am pretty sure that I did not update anything before the Thunderbolt connector stopped working. I then did some updates to try and fix it, but originally I did not do anything at all. This is a very wild guess, but could it be caused by a certificate expiring somewhere?

I need to do more research on Thunderbolt but from what I could gather from [this article](https://www.dell.com/community/Latitude/Demystifying-Thunderbolt-3-Security-Levels/td-p/7529712), there are 4 modes to Thunderbolt connectors:
- USB 2.0
- USB 3.x
- DisplayPort
- PCIe
It seems to me that we are locked in USB-only mode whereas we should be working in PCIe mode with our USB/display adapters (which are really miniaturized docks).

I have also confirmed that switching modes and enabling/disabling BIOS assist mode does not change anything to the content of the firmware chip.

I'd like to look into more details in how the firmwares work and how they are supposed to be verified by the BIOS or Intel ME. I will keep you updated if I make any progress.

tw0flower, 2022-12-01:
I did not have time to work directly on the firmwares that we have, but I searched Google again for information about Thunderbolt. I think these slides from a 2020 Black Hat conference could be useful: https://i.blackhat.com/USA-20/Thursday/us-20-Ruytenberg-When-Lightning-Strikes-Thrice-Breaking-Thunderbolt-3-Security.pdf

They describe the Thunderbolt architecture in detail with some hardware examples that are similar to the Carbon X1 that we have. And the different sections of the firmwares.

I'm attaching the PDF to this comment for safekeeping.
[us-20-Ruytenberg-When-Lightning-Strikes-Thrice-Breaking-Thunderbolt-3-Security.pdf](https://github.com/fwupd/firmware-lenovo/files/10128282/us-20-Ruytenberg-When-Lightning-Strikes-Thrice-Breaking-Thunderbolt-3-Security.pdf)

dal00, 2022-12-01:
I found a tool referenced in that pdf https://github.com/BjornRuytenberg/tcfp which looks like can parse the firmware.

When running it on TBT.bin extracted from n23th08w I get the following output:
I wonder if we're missing something important since it says the image is incremental.

```
$ ./tcfp.py  parse TBT.bin 
2022-12-01 11:25:25,998 - WARNING - File size in between 229376 and 1048576 bytes. Possible causes: 
 - Image may be an incremental firmware update. While tcfp may be able to parse the SL state, please note patching requires a full dump. 
 - Image dump may be incomplete, i.e. not include 'scratch pad' section. However, this should typically not cause any issues.
Vendor ID : 0x109
PCI ID : 0x15d3
PCI Device Name : JHL6540 Thunderbolt 3 Bridge (C step) [Alpine Ridge 4C 2016]
Model ID : 0x1706
NVM version : 47 (0x2f)
Vendor : Lenovo
Device : X1 Carbon
Image type : Incremental
Security Level : SL1
```

kubabuda, 2022-12-05:
@dal00 

> TBT.bin extracted from n23th08w 

How to do it? When I just unzip TB firmware, there are multiple .bss, .data, .edata etc sections, no memory map
Is there tool or commandline spell (linker?) to just get my .bin from that?
This is probably trivial, but I could not google anything useful

jnada2021, 2022-12-06:
> @dal00
> 
> > TBT.bin extracted from n23th08w
> 
> How to do it? When I just unzip TB firmware, there are multiple .bss, .data, .edata etc sections, no memory map Is there tool or commandline spell (linker?) to just get my .bin from that? This is probably trivial, but I could not google anything useful

Are you using Linux or Windows 10/11. If you are using Windows it creates a folder under C drive:
C:\DRIVERS\WIN\ThunderboltFW
BTW here is the latest FW TBT file:
[TBT.zip](https://github.com/fwupd/firmware-lenovo/files/10164767/TBT.zip)

tw0flower, 2022-12-08:
I tried downgrading the BIOS of my computer yesterday. I really don't get why our Thunderbolt are working in USB mode but completely transparent at the same time (no listing in Windows Device Manager and lspci). Could it be a killswitch that was introduced in some an update?

The paper published on the Thunderspy vulnerability states that there is no fix for computers released before 2019. The X1 Carbon Gen 6 was released in 2018. So this vulnerability simply cannot be fixed when it comes to memory protection.
Therefore it's also unlikely that the BIOS or Intel ME could detect our firmwares as being "untrusted" at least that's what the author of the paper seems to suggest in the Q&A:

> [Why has Intel not fixed Thunderspy?](https://thunderspy.io/#intel-not-fixed-thunderspy)
> In our vulnerability disclosure procedure, Intel has confirmed the following:
> 
>     All three versions of Thunderbolt are affected by the Thunderspy vulnerabilities.
>     Only systems shipping Kernel DMA Protection mitigate some, not all, of the Thunderspy vulnerabilities.
>     Only some systems that began shipping since 2019 come with Kernel DMA Protection.
>     Beyond [Kernel DMA Protection](https://thunderspy.io/#kernel-dma-protection), Intel will not provide any mitigations to address the Thunderspy vulnerabilities. Hence, Intel will not assign any CVEs to the Thunderspy vulnerabilities, or release any public security advisories to inform the general public.
> 
> Despite our repeated efforts, the rationale to Intel's decision not to mitigate the Thunderspy vulnerabilities on in-market systems remains unknown. Given the nature of Thunderspy, however, we believe it would be reasonable to assume these cannot be fixed and require a silicon redesign. Indeed, for future systems implementing Thunderbolt technology, Intel has stated they will incorporate additional hardware protections.
> 
> Kernel DMA Protection is only available on a limited number of recent systems. Hence, all systems released before 2019, and more recent systems that do not ship Kernel DMA Protection, will remain fully vulnerable to Thunderspy forever. Please refer to Kernel DMA Protection for more information.

Thunderspy involves several problems. One of which is that the signature of the firmware is not checked during boot. I believe this is what the author is referring to when he says that Intel has not done anything beyond Kernel DMA Protection.

But what if they found a way to change this recently and introduced a bug? And the chip is wrongly and silently being deactivated at some stage?

I've begun analyzing the files I have with tcfp yesterday, I'll write my results here tonight.

jnada2021, 2022-12-08:
> I tried downgrading the BIOS of my computer yesterday. I really don't get why our Thunderbolt are working in USB mode but completely transparent at the same time (no listing in Windows Device Manager and lspci). Could it be a killswitch that was introduced in some an update?
> 
> The paper published on the Thunderspy vulnerability states that there is no fix for computers released before 2019. The X1 Carbon Gen 6 was released in 2018. So this vulnerability simply cannot be fixed when it comes to memory protection. Therefore it's also unlikely that the BIOS or Intel ME could detect our firmwares as being "untrusted" at least that's what the author of the paper seems to suggest in the Q&A:
> 
> > [Why has Intel not fixed Thunderspy?](https://thunderspy.io/#intel-not-fixed-thunderspy)
> > In our vulnerability disclosure procedure, Intel has confirmed the following:
> > ```
> > All three versions of Thunderbolt are affected by the Thunderspy vulnerabilities.
> > Only systems shipping Kernel DMA Protection mitigate some, not all, of the Thunderspy vulnerabilities.
> > Only some systems that began shipping since 2019 come with Kernel DMA Protection.
> > Beyond [Kernel DMA Protection](https://thunderspy.io/#kernel-dma-protection), Intel will not provide any mitigations to address the Thunderspy vulnerabilities. Hence, Intel will not assign any CVEs to the Thunderspy vulnerabilities, or release any public security advisories to inform the general public.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Despite our repeated efforts, the rationale to Intel's decision not to mitigate the Thunderspy vulnerabilities on in-market systems remains unknown. Given the nature of Thunderspy, however, we believe it would be reasonable to assume these cannot be fixed and require a silicon redesign. Indeed, for future systems implementing Thunderbolt technology, Intel has stated they will incorporate additional hardware protections.
> > Kernel DMA Protection is only available on a limited number of recent systems. Hence, all systems released before 2019, and more recent systems that do not ship Kernel DMA Protection, will remain fully vulnerable to Thunderspy forever. Please refer to Kernel DMA Protection for more information.
> 
> Thunderspy involves several problems. One of which is that the signature of the firmware is not checked during boot. I believe this is what the author is referring to when he says that Intel has not done anything beyond Kernel DMA Protection.
> 
> But what if they found a way to change this recently and introduced a bug? And the chip is wrongly and silently being deactivated at some stage?
> 
> Downgrading the BIOS did nothing, and we cannot downgrade Intel ME through software solutions, I've also tried. Might be doable through a SPI programmer.
> 
> I've begun analyzing the files I have with tcfp yesterday, I'll write my results here tonight.

I hope your research bares fruit since so far you and dal000 have been the most resourceful on this issue.

I keep asking myself as to how to remove/roll back ME since the chip we have been writing the tbt.BIN to is not really corrupted, I think. I write zero.bin file and load any version of TBT.bin and boot and reconnect and read with the SPI clip again. Do a Beyond compare etc. and all versions of the bin files seem to be the same.

As of this week I have a brand new problem (after the BIOS update to 58 and consequently updating ME firmware to the latest version), no matter which TBT.bin I load (including very old ones that solved the low wattage charging problem), during boot and within the OS I have low wattage charging issue. It is neither the USBC cables, nor the port nor the power adapters since I have 2 original 65W usbc and also some other 3rd party (high quality) adapters.

To check if it were the USBC port or the cable problem, I disconnected the built in battery while in the OS and plugged AC in and it still kept working even reboot works but when I remove the battery and try to run on AC only from a full shutdown no dice. KB lits up, CPU fan spins for some seconds and then silence and no boot. Until of course I reconnect the battery and AC back things start working again.

I am very tempted to do remove as much ME and load the stripped BIOS back with SOIC clip but I fear I will brick the machine and render it completely useless. I have already extracted&downloaded the contents of the BIOS dump 4 times and all are the same according to beyond compare.
https://github.com/corna/me_cleaner/wiki/External-flashing

I loved thinkpads for over 20+ years but this issue I think is wearing me down and I will throw the towel and never touch a thinkpad ever again.

tw0flower, 2022-12-08:
Everyone's research on this has been very useful IMO. We have no idea what's causing the issue and it takes a lot of time to dig information. And we're all doing that on our free time.

I'm personally not convinced that the problem lies with the Thunderbolt chip itself. But this might be wrong. One interesting thing to try would be to patch one of the bin files with the Thunderspy vulnerability to force it to the SL0 level.
Unfortunately, and this is what my attempts from yesterday showed, none of the firmwares I've tried has a signature compatible with the signatures provided by the tcfp tool. So we'd have to find how to patch it ourselves.

All images provided by Lenovo are using the SL1 level (user authorization). One interesting thing to note is that the version 47 advertises a fix for a vulnerability without mentioning which one. I don't know any other Thunderbolt-specific vulnerability outside of Thunderspy.

As opposed to the newer firmwares provided by Lenovo, my original firmware's SL level is not even detected by the tool. When I looked manually it seemed close to one of the signatures listed in the tool's code (see [here](https://github.com/BjornRuytenberg/tcfp/blob/master/tcfp.py#L49)) in terms of offset but the values are different. I have not tried any of the other firmwares that people uploaded here but it might be worth a try.

I'd like to contact the researcher that worked on the Thunderspy paper to ask him if he heard of any mitigation that might have been developed recently. I'll write him a message tomorrow or this weekend.

I've only ever owned Thinkpads as well, my first one was a T42 that I might still have somewhere haha! I remember a common fix for the ATI GPU overheating was to put the motherboard in the oven. Maybe we should try that ;-)

jnada2021, 2022-12-08:
It being a holiday with heavy rains, I caved in and attempted to "disable" Intel ME b using below tutorial. As per the screen capture attached in BIOS ME doesnt report the version.
https://x86sec.com/posts/2022/09/08/removing-intel-ME-on-x1-carbon/

No change in the status of DP, HDMI or TB. Back to the drawing board.
[image]

jnada2021, 2022-12-10:
Is there a known way to get the whole dump of TBT.bin file 1MB inside the chip we have been flashing without the external clip? The fellow I used whose tutorial about disabling intel ME owns a 6th gen Carbon with i7 and is willing to share his file since his HDMI, DP and thunderbolt is working. Unfortunately he doesn't have access to the SOIC clip at the moment hence the question.
Cheers!

jnada2021, 2022-12-26:
Hey. I've managed to get a working bin from a generous donor.
Give it a try and let me know if it solves your issues. Happy holidays!
[x1c6-win-tb.zip](https://github.com/fwupd/firmware-lenovo/files/10304602/x1c6-win-tb.zip)

RAY5D, 2022-12-26:
> Hey. I've managed to get a working bin from a generous donor. Give it a try and let me know if it solves your issues. Happy holidays! [x1c6-win-tb.zip](https://github.com/fwupd/firmware-lenovo/files/10304602/x1c6-win-tb.zip)

Surprise Christmas present O.o?! Did you get it through a programmer or some sort of software means? My friend has an Thinkpad tablet on the problematic list and wants to back up the firmware before anything bad happens. However, the flash chip in that model is not readable without desoldering, and is a pretty annoying DFN package, so it would be very helpful if we can get the data without messing with the motherboard.

jnada2021, 2022-12-26:
Some good samaritan responded to my post from December last year. He extracted it with a SOIC clip.

RAY5D, 2022-12-26:
Still not working for me. If the bin works for other machines, the problem might be that it contains a public key as mentioned in the Black Hat Thunderbolt 3 Security slide page 29. I searched in the file, and the string "RSA" appeared at 0x38E00 and 0xB8000. The content seems to be the same at both locations, maybe the 2 copies are for assist mode and no assist mode? Anyways, page 31 says all TB3 controllers have unique IDs fused in silicon, but it also mentioned that the statement is inaccurate, not sure what that means. However if we need unique keys just to use the binary on our TB3 chips, we'll have to find a way to read the ID. Any ideas?
[image]

jnada2021, 2022-12-27:
Well, since many people in this post were using nothing but linux and never updated their Intel ME, I have a sneaky suspicion that the problem we have been experiencing may probably be a mix of both hardware and software fukups combined. Those 2 chips near the ROM chip that we have been flashing may be the culprit or some other BS issue that is injected via BIOS, Intel ME which burns/damages something somewhere.

My first USBC-Thunderbolt port was shot as of this summer (it was always finicky to begin with) and since a month or so my remaining USBC-TB port (the combination one) only charges at 15W no matter what I do (in terms of flashing the ROM chip). And I always used original Lenovo/Thinkpad USBC adapters and ports were never dinged etc, so go figure!

Lenovo folks finally made impossible possible and cracked the code of planned obsolescence. I feel bad for people who paid more than a grand for an X1 Carbon since let's be real, other than the keyboard and soft black coating the difference between mid/high range Lenovo consumer laptops and Carbon line is very negligable since 2017.

I think I will try to sell mine for a couple hundred euros i7-16GB, 1080p IPS and 6hr on battery and look for something else.

andreidragu, 2023-01-07:
> Hey. I've managed to get a working bin from a generous donor. Give it a try and let me know if it solves your issues. Happy holidays! [x1c6-win-tb.zip](https://github.com/fwupd/firmware-lenovo/files/10304602/x1c6-win-tb.zip)

can you help me get one for T580?

jclaveau, 2023-03-16:
> I've always had my laptop on the latest (lvfs) updates; I haven't ran Windows in years. Still, I've lost HDMI/TB3 output a few months ago. I'd like to thank @jnada2021 and @RAY5D for all their hard [...]

I have a similar issue but I'm waiting my SOIC clips for more than a month now (delivery issue) to be sure. 

Meanwhile, I bought an `usb-c/HDMI dongle` and I can now have a second screen: https://www.cdiscount.com/informatique/clavier-souris-webcam/hub-usb-c-adaptateur-multiport-10-en-1-vers-hdmi-e/f-1070229-elc0713325730599.html#desc (i can't find an english link but you should get the idea)

Hoping this `workaround` can help you.

jiahe224, 2023-08-05:
I installed the Thunderbolt Firmware Update Utility (https://pcsupport.lenovo.com/us/en/products/LAPTOPS-AND-NETBOOKS/THINKPAD-X-SERIES-LAPTOPS/THINKPAD-X1-CARBON-6TH-GEN-TYPE- 20KH-20KG/downloads/DS502615), 
but the NVM firmware version is 30 instead of 46, my problem is, I set the display to turn off after 10 minutes and the computer never goes to sleep, when the display turn off, the external display will sleep（I'm not sure that whether the status is sleep） because it loss of input signal。When pressed the keyboard  or moved the mouse   to light up the screen, the X1C will continue to emit a prompt sound that the hardware is not recognized, but the external display can display the image normally, and there is no yellow exclamation mark in the device manager, and the prompt sound will disappear after plugging and unplugging the display . My computer is connected to the display through type-C, and the external display can provide 65W power supply to the computer. I am not sure whether this problem is related to thunderbolt, because my other computer X13 is not installed with thunderbolt, But everything works fine.

atoch419, 2023-08-07:
Has anyone made any progress on this. I have exactly the same issue No thunderbolt no alt DP no hdmi no external display outputs showing up on Xrandr but fast charging still works. from reading this whole thread it seems like something is missing and the reported issue with a bad firmware isn't the whole story. Sounds like replacing the eporom and reflashing dosent work? could the issue be further down stream or are people thinking it's a signature issue? Interestingly I got my machine in mid 2020. I did all the updates on the Lenovo website in windows befor swapping to Linux so I should of installed the thunderbolt firmware fix. although I can't recall reading about this issue until it happened to me. could it be that damage was already done to the chip and even with new firmware it's life time shortened and this failure was going to happen sooner or later?

RAY5D, 2023-08-11:
> Has anyone made any progress on this. I have exactly the same issue No thunderbolt no alt DP no hdmi no external display outputs showing up on Xrandr but fast charging still works. from reading this [...]

I still think the issue is with the firmware flash chip. The reason mine wasn't fixed is that the original chip has lost all the data, which seemed to be somewhat unique to each computer. I also had a Thinkpad tablet that is vulnerable to this problem, but I backed up the original data, then replaced the chip and updated to the latest version. In this way, no further damage can be done during the update process, and it's still working fine.

fowoz, 2023-08-15:
> I still think the issue is with the firmware flash chip. The reason mine wasn't fixed is that the original chip has lost all the data, which seemed to be somewhat unique to each computer. I also had [...]

[image]

[image]

[image]

[image]

jiahe224, 2023-08-16:
> I installed the Thunderbolt Firmware Update Utility (https://pcsupport.lenovo.com/us/en/products/LAPTOPS-AND-NETBOOKS/THINKPAD-X-SERIES-LAPTOPS/THINKPAD-X1-CARBON-6TH-GEN-TYPE- 20KH-20KG/downloads/DS502615), but the NVM firmware version is 30 instead of 46, my problem is, I set the display to turn off after 10 minutes and the computer never goes to sleep, when the display turn off, the external display will sleep（I'm not sure that whether the status is sleep） because it loss of input signal。When pressed the keyboard or moved the mouse to light up the screen, the X1C will continue to emit a prompt sound that the USB device is not recognized, but the external display can display the image normally, and there is no yellow exclamation mark in the device manager(but the device list will be refreshed constantly), and the prompt sound will disappear after plugging and unplugging the display . My computer is connected to the display through Thunderbolt, and the external display can provide 65W power supply to the computer, I am not using an HDMI cable. I am not sure whether this problem is related to thunderbolt, because my other computer X13 is not installed with thunderbolt software, But everything works fine.

I no longer use the monitor to power the computer, but use the power supply that comes with the computer, and use a type-c cable to connect the thunderbolt interface and the monitor, and it seems to be normal now.

atoch419, 2023-09-14:
Hey REY5D So i think im in exactly the same boat as you now
I checked my eeprom and it is full of fs if I flash anything to it nothing changes. if use hex dump it there is absolutely nothing there. Can you remind me if you tried the zero filling and booting after replacing the chip befor flashing the firmware. I dont understand how the person at this link https://nadim.computer/posts/2018-10-26-thinkpad.html which has been posted befor could of flashed the chip with zeros then flashed the firmware and got a fully working machine if their has to be something on the chip that ids it. unless the chips ID or serial number or somthing is stored in the chip and recorded else where in the machine (maybe on the other bios chip?) or another chip?

[23 more replies]