fwupd/firmware-lenovo #265: Potentially wrong firmware on LVFS for ThinkCentre M70s

PhilippWendler, 2022-08-30:
I suspect there is some wrong firmware on LVFS for the ThinkCentre M70s (and likely also the M70t, M80s, M80t, M90s, M90t).

Summary of facts and current situation:
- I have an M70s (model 11DBSAPP00) with firmware version 0.1.71 / M2TKT47A from 2021-09-27 (was shipped with this version).  
- If I search for M70s on LVFS, I only get this [old version](https://fwupd.org/lvfs/devices/com.lenovo.M90t_M90s_M80t_M80s_M70t_M70s.BIOS.firmware) 0.1.45 / M2TKT2DA from 2020-09-02.  
- However, fwupd 1.8.4 actually prompts me to install [this M70**t**s firmware](https://fwupd.org/lvfs/devices/com.lenovo.M90ts_M80ts_M70ts.BIOS.firmware) in version 0.1.79 / M2TKT4FA from 2022-06-15.  
- On the Lenovo support page, the latest firmware for the M70s is version M2TKT51A from 2022-07-26. Under "Supported Systems" and in the readme there are the M70s, M70t, M80s, M80t, M90s, and M90t systems (matching [this LVFS entry](https://fwupd.org/lvfs/devices/com.lenovo.M90t_M90s_M80t_M80s_M70t_M70s.BIOS.firmware)), but *not* the M70ts.

This leads me to the following questions:
- Is it safe to install the M70ts firmware on the M70s as offered by fwupd even though the support page seems to indicate that these systems require different updates?
- If yes, should the respective LVFS entries [1](https://fwupd.org/lvfs/devices/com.lenovo.M90t_M90s_M80t_M80s_M70t_M70s.BIOS.firmware) and [2](https://fwupd.org/lvfs/devices/com.lenovo.M90ts_M80ts_M70ts.BIOS.firmware) be merged somehow or otherwise indicated that they apply to the same devices?
- Can we get the latest firmware M2TKT51A on LVFS as well?

However, if my suspicion is correct, then the M70ts firmware is wrongly offered for my M70s system and this should be avoided.

Here is the output of `fwupd get-devices --show-all-devices` with fwupd 1.8.4 (I only removed the serial of my SSD from this listing):
```
LENOVO 11DBSAPP00
│
├─Core™ i7-10700 CPU @ 2.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         a45b0522-5722-54bd-b802-86cd044262df ← CPUID\PRO_0&FAM_06&MOD_A5
│                         538440cb-7cac-5185-bc01-d93a1e474058 ← CPUID\PRO_0&FAM_06&MOD_A5&STP_5
│     Device Flags:       • Internal device
│   
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    05
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              a8dc77a9-e1b4-5ff1-9b0e-72dd2a64d3df ← PCI\VEN_8086&DEV_9BC5
│                         f677e5ac-41ac-5587-b2ef-1322dc207d18 ← PCI\VEN_8086&DEV_9BC5&REV_05
│                         93ddcacf-bf57-5d67-9293-ac7c93455565 ← PCI\VEN_8086&DEV_9BC5&SUBSYS_17AA3168
│                         768306a9-1c54-5ad2-866c-106768410c94 ← PCI\VEN_8086&DEV_9BC5&SUBSYS_17AA3168&REV_05
│                         da79ccae-8ac2-5694-bb66-7dec5c969d50 ← PCI\VEN_8086&DEV_9BC5&REV_00
│                         d912e4bf-2e20-5836-a2bc-fc1e74bda9fb ← PCI\VEN_8086&DEV_9BC5&SUBSYS_17AA3168&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    65547
│     Minimum Version:    65547
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               a7f3a1a9-fd4e-4df9-8c9b-0ce37d1a7a6e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    1612776549
│     Minimum Version:    1612776549
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               6159fb9c-e80b-4d13-9018-bb00bd2f353a
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
[... 122 lines cut ...]
      
      This stable release fixes the following issues:
      
      • Enhancement to address security vulnerability.

  └─ThinkCentre M90ts_ThinkCentre M80ts_ThinkCentre M70ts:
        New version:      0.1.79
        Remote ID:        lvfs
        Release ID:       13217
        Summary:          System Firmware for ThinkCentre M90ts_ThinkCentre M80ts_ThinkCentre M70ts
        License:          Proprietary
        Size:             16.9 MB
        Created:          2022-04-29
        Urgency:          Low
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Some of the platform secrets may be invalidated when updating this firmware. Please ensure you have the volume recovery key before continuing.
        
        This stable release fixes the following issues:
        
        • Enhancement to address security vulnerability.
```

If you need any further information, I will be glad to assist.

hughsie, 2022-08-30:
@mrhpearson can you ping the right person please? Thanks.

mrhpearson, 2022-08-30:
I think this is fine. The same BIOS is used for M70 t and s and checking the README on the Lenovo site it does list M2TKT4FA as a release for the M70s (two releases ago).
The newer version is on LVFS - it's just under embargo still and it looks like they're working on it (my guess is they're just waiting for the approval process to complete)
Afraid I don't understand the question about merging - they're different releases so that doesn't make sense to me.

I think you're good to go :)

PhilippWendler, 2022-08-31:
Thanks @mrhpearson! The update worked fine. And thanks a lot in general for working with LVFS! This is really nice and definitely an argument in buying decisions for us.

Sorry for the confusing part about "merging". If they are different things, maybe they can be renamed such that it is visible to users that it is indeed the correct update for M70(s|t|ts)?

From my point of view this issue can be closed if you don't want to keep it open for improving the naming.