fwupd/firmware-lenovo #266: Lenovo System Firmware updates wipe SLIC ACPI table

jevank, 2022-09-02:


hughsie, 2022-09-02:
@mrhpearson sounds like a regression to me, accidental I assume?

mrhpearson, 2022-09-19:
Hi @jevank 

Feedback from the FW team is they don't know what change might have caused this (bear in mind many of the changes in the BIOS come from Intel). They'll need to debug but need further information on how to reproduce.

They don't see a SLIC table under ACPI in either Windows or Linux. Can you share more details on how to reproduce and check this issue. We're not seeing any SLIC entry under /sys/firmware/acpi/tables

jevank, 2022-09-21:
Hi @mrhpearson 

the SLIC table is available on laptops with preinstalled Windows 10 and requires AFAIK to Windows activation process only.

I guess the firmware update somehow make it invisible or (I hope not) wipe it. I've never seen it before using firmware updates under Windows.

What can I help? Trying downgrade to any previous versions?

mrhpearson, 2022-09-26:
Hi - let me check what the FW team want to do with this. Kinda intrigued that they don't expect there to be a SLIC table - you're sure it was there before? (I haven't got a clue how Windows does this normally)

jevank, 2022-09-26:
> you're sure it was there before?

Absolutely. I use this table passed through to virtual machine to activate guest Windows system.

mrhpearson, 2022-09-26:
OK - I've passed them back to them

For my understanding - I'm not seeing the table under /sys/firmware/acpi/tables on any of my systems. Is that because they're all prototypes and won't be activated - or am I looking in the wrong place. I'd like to be able to show the FW team how to reproduce it but don't think I have a system that I can do that on (joys of getting prototypes...)

Mark

das-menschy, 2022-10-15:
I've got the same problem. On my Thinkpad X1 Carbon 5th Generation, the MSDM table with the Windows License Product Key is also gone. I guess the deletion also happened during the last Lenovo System Firmware update 1.55 => 1.56. 

I can say with 100% certainty that I had this Windows 10 Product License Key, because I **saved** it in my Password Manager (KeePassXC) some time ago, when it was still available! 
My former Windows 10 Product License Key in ``/sys/firmware/acpi/tables/MSDM`` was: 
```
CCRRB-<redacted>-<redacted>-<redacted>-V8RC2 
```

Now, the file MSDM and the corresponding ACPI table don't exist anymore: 
```
[root@linux-on-20hr002mmx ~]# ls -1 /sys/firmware/acpi/tables/
APIC
'ASF!'
BATB
BGRT
BOOT
data
DBG2
DBGP
DMAR
DSDT
dynamic
ECDT
FACP
FACS
FPDT
HPET
MCFG
POAT
SSDT1
SSDT10
SSDT2
SSDT3
SSDT4
SSDT5
SSDT6
SSDT7
SSDT8
SSDT9
TPM2
UEFI1
UEFI2
WSMT
```

thmo, 2022-11-10:
Happened already 2019 to my T460s, lost Windows key, as SLIC and MSDM were wiped. Unfortunately after the two years warranty period already ended..., so I didn't follow up further on this.

See [here (in German)](https://forums.lenovo.com/t5/T4-T5-und-neuere-T-Serie-ThinkPad-Notebooks/T460s-Win-Lizenz-verloren/m-p/4486012).

hughsie, 2022-11-10:
We've added the "upload hashes of your ACPI tables before and after the update" as part of the report. Which means when people using 1.8.7 and newer upload a report we can de-deploy the firmware automatically and then ping the right person. Longer longer term we want to require vendors to do a signed report before they're allowed to move a file to the stable remote.

closed 2023-09-21 by mrhpearson