fwupd/firmware-lenovo #267: Thinkpad X1 Extreme Gen2 failing update

mcamou, 2022-09-05:
There are 2 issues here:

1. Running `fwupdmgr get-updates` shows me an update to the Intel Management Engine. If I apply it and reboot, the update is still there:

```
$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • MZVLB1T0HBLR-000L7
 • MZVLB1T0HBLR-000L7
 • Prometheus
 • Prometheus IOTA Config
 • System Firmware
20QVCTO1WW
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    192.81.1753
  │   Minimum Version:    192.81.1753
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              993f4478-ab74-48e2-bf65-26e6462e01d6
  │                       b0c45b68-5e8b-5be2-93e1-70862b0add2c
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad P1 Gen2 Corporate:
        New version:      192.90.2072
        Remote ID:        lvfs
        Release ID:       14692
        Summary:          Lenovo ThinkPad P1 Gen2 X1 Extreme 2nd Corporate Corporate ME Firmware
        License:          Proprietary
        Size:             12.2 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N2ORM18W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad P1 Gen2 X1 Extreme 2nd Corporate (Consumer) ME Firmware
```

I don't know how to diagnose this, I have secure boot disabled and "Change boot order" enabled.

2. There is a new update to firmware 1.44 in https://pcsupport.lenovo.com/es/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-extreme-gen-2/20qv/20qvcto1ww/r90wad97/downloads/driver-list/component?name=BIOS%2FUEFI which does not yet appear in LVFS. If I download it and try to install, I get the following:

```
$ fwupdmgr install N2OET57W.cab
Decompressing…           [***************************************]
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer
```

Any ideas? I assume the second problem has to be dealt with by Lenovo, any ideas where I can report it?

mrhpearson, 2022-09-06:
Hi - I believe you've hit a known issue that is being fixed.
If you go into the BIOS setup screen (F1 during early boot) can you confirm which ME version you have there - it's likely the updated version. 
The BIOS isn't reported the updated version to the ESRT tables correctly, so fwupdmgr thinks an update is still needed.
For not just ignore it - the fix is rolling out on various platforms and I'll check what the status is on X1EG2

mcamou, 2022-09-08:
Thanks @mrhpearson. I've checked and the current ME version according to the BIOS is 192.90.2072 which is the same one that `fwupdmgr` says I need to update to, so I'm hitting that bug. I'll ignore it for now, it would be nice to know what the status is.

mcamou, 2022-09-16:
Hello again @mrhpearson, do you have any news on the status of the X1EG2? It's becoming annoying... each time that I run Plasma Discover to get updates, I need to remember to disable this one, or the next reboot is going to take a long time.

mrhpearson, 2022-09-16:
They've been rolling out the fix for a number of platforms - I believe you need to pick up the BIOS fix, and then install (or re-install) the ME FW.

I can see that for the X1EG2 there is a FW (1.44) that was uploaded to LVFS 10 days ago but is under embargo and I'm guessing that has the fix but not sure

@ChiWei-Chen - do you have confirmation this issue is fixed in 1.44 and any ETA from the FW team on when it will be moved to stable (and did I put anything incorrect in the above!)

ChiWei-Chen, 2022-09-20:
@mrhpearson @mcamou thanks for looping me in, this seems duplicated to #251 , the FW already merged the fix in the next BIOS release, I'll keep you update once there is clear ETA. I assume it would be in 1.45. (LO-1938 for my tracking)

rcbjr, 2022-09-26:
I have a potentially related issue:
My X1E G1 Intel ME  firmware update fails.

BIOS:  N2EET59W (1.41)
BIOS reports old firmware (12.0.85.1869)

MEInfoWin64.exe  reports old firmware (12.0.85.1869 H Corporate)
(not sure why corporate, it's my computer since day 1)

Lenovo Vantage reports that a critical update is available, appears to update the firmware, then reports update failed (and to try rebooting)
A reboot changes nothing.

This has been going on for months.
I've been trying periodically, hoping that the issue would resolve.

I've tried dropping my Antivirus and my VPN, charging to 100% before starting (instead of my normal 80%), and downloading and manually running the update from Lenovo.

Any other ideas?

ChiWei-Chen, 2022-09-26:
@rcbjr , thanks, but I think it would be better to report this issue to forums.lenovo.com , you can get more Vantage experts' assistance to resolve the issue of Windows environment. Thank you very much.

collinjc, 2022-11-04:
I have a Lenovo ThinkPad P1 Gen2 X1 Extreme 2nd with this same issue. fwupdmgr thinks that it is still on 192.72.1757, but the BIOS reports that it is on 192.90.2072.

mrhpearson, 2022-11-04:
Hi @collinjc - pretty sure you hit the ESRT table update issue. 
The fix unfortunately didn't make it into 1.44 BIOS release but will be in 1.45 which we're expecting to be released mid November.
Afraid for now just ignore the update suggestions :(

ChiWei-Chen, 2022-11-18:
Hi, 
The BIOS 1.45 is stable now, please give it a try, thanks.
It looks the problem was resolved. #251

collinjc, 2022-11-18:
I had to manually install it (perhaps fwupdmgr just isn't picking it up yet?), and it does appear to have resolved the issue.

closed 2022-11-21 by kmauleon