fwupd/firmware-lenovo #268: Qualcomm Snapdragon X55 5G "Update Error: Device was not found"

WillsyAu, 2022-09-10:
Hi,
My Thinkpad X1 Nano G2 is reporting the below device details for the X55 5G modem. 
├─Qualcomm Snapdragon X55 5G:
│     Device ID:          bba54f76de66958e5d70964b6ddd2d1e6c7dae9e
│     Summary:            Mobile broadband device
│     Current version:    T99W175.F0.6.0.0.6.TE.010.020
│     Vendor:             foxconn (PCI:0x105B)
│     GUIDs:              35e951d5-8ae0-59d7-99a5-61ef1f42c024 ← PCI\VID_105B&PID_E0BF&REV_0000&CARRIER_TELSTRA
│                         8315859b-9a21-589b-953d-414ff7d4e02c ← PCI\VID_105B&PID_E0BF&REV_0000
│                         b3281cfa-f7b5-5af9-9b8d-7b47662050b5 ← PCI\VID_105B&PID_E0BF
│                         c8591d06-63e4-55e7-9dec-3b0783b15e2c ← PCI\VID_105B
│     Device Flags:       • Updatable
│                         • System requires external power source

When I download the .cab file for the latest drivers to do a local install it reports "device not found".
├─Unknown Device:
│ │   Description:        
│ │   Updating the firmware on your T99W175 device improves stability and adds new features.
│ │   Update Error:       Device was not found
│ │   GUID:               54347058-eae5-545c-908c-8bde8d475944
│ │   Device Flags:       • Supported on remote server
│ │ 
│ └─T99W175:
│       New version:      T99W175.F0.1.0.0.9.TE.008.079
│       Remote ID:        lvfs-testing
│       Summary:          Firmware for Lenovo T99W175 5G modem (Telstra)
│       Licence:          Proprietary
│       Urgency:          High
│       Release Flags:    • Trusted payload
│                         • Trusted metadata
│       Description:      
│       Firmware release T99W175.F0.1.0.0.9.TE.008.079 packaged for LVFS
│       
│       This stable release fixes the following issues:
│       
│       • Disable feature "LwM2M" for carrier AT&T
│       • Fixed issue that AT command "AT^SLBAND" get WCDMA band information is incorrect sometimes

WillsyAu, 2022-09-11:
I found this on kernel.org. Seems my issue may be that I have the e-sim compatible version of the card... ?

diff --git a/drivers/bus/mhi/pci_generic.c b/drivers/bus/mhi/pci_generic.c
index 59a4896a8030..566483dc15db 100644
--- a/drivers/bus/mhi/pci_generic.c
+++ b/drivers/bus/mhi/pci_generic.c
@@ -423,6 +423,9 @@  static const struct pci_device_id mhi_pci_id_table[] = {
 	/* DW5930e (sdx55), Non-eSIM, It's also T99W175 */
 	{ PCI_DEVICE(PCI_VENDOR_ID_FOXCONN, 0xe0b1),
 		.driver_data = (kernel_ulong_t) &mhi_foxconn_sdx55_info },
+	/* T99W175 (sdx55) */
+	{ PCI_DEVICE(PCI_VENDOR_ID_FOXCONN, 0xe0bf),
+		.driver_data = (kernel_ulong_t) &mhi_foxconn_sdx55_info },

https://patchwork.kernel.org/project/linux-arm-msm/patch/20211028034431.3563-1-slark_xiao@163.com/#24561329

mrhpearson, 2022-09-12:
I don't think that should matter - but I'm not an expert here. I do know that it was qute the challenge getting the FW updates up and working and the version of kernel and fwupd are important.

Could you let me know what you're running - distro, kernel version, fwupdmgr version and (if possible) modem manager version and I'll get some input from the team who were working on this.

Thanks

WillsyAu, 2022-09-13:
Thanks Mark,
I am running Fedora 36.

$ uname -r
5.19.7-200.fc36.x86_64

$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.4

$ ModemManager --version
ModemManager 1.18.8-1.fc36

$ dmesg | grep 105b
[    0.552904] pci 0000:08:00.0: [105b:e0bf] type 00 class 0x0d4000

Thanks

mrhpearson, 2022-09-16:
As a note - tracked internally with LO-2027. Thanks for the details

SlarkXiao, 2022-09-26:
Hi,
  The FW you  used belongs to a newer code baseline. FW version named by F0.6.0.0.6.XX.
  And the target FW you want to use F0.1.0.0.9.XX doesn't share a same code baseline. So it's not allowed to update FW to another directly.
  Actually, both these 2 FW were released in this April. Don't worried about the FW is too old.
  More concerns about the new update version of F0.6.0.0.6.XX, please contact with Lenovo about this.

mrhpearson, 2022-09-26:
Hi @SlarkXiao - thanks for replying directly on here, appreciate it.
What should users be doing to get from the old to the new? Is there an interim version needed etc?
Thanks
Mark

WillsyAu, 2022-09-28:
Thanks Vishnuocv,
Yes I am having issues with FCC unlock.... Given your confirmation of the hardware difference, I will close this ticket and chase up the vendor regarding the FCC unlock scripts.

Thanks.

closed 2022-09-28 by WillsyAu