fwupd/firmware-lenovo #269: Trouble updating firmware on ThinkPad T16 Gen 1

davidstrauss, 2022-09-18:


hughsie, 2022-09-18:
Can you try with Secure Boot turned off?

davidstrauss, 2022-09-18:
> Can you try with Secure Boot turned off?
I already have it off.

hughsie, 2022-09-18:
> Fedora Silverblue 36

I'm wondering if this is somehow broken; we're seeing quite a few failure with SB in the last few months.

davidstrauss, 2022-09-19:
> I'm wondering if this is somehow broken; we're seeing quite a few failure with SB in the last few months.

For what it's worth, my ThinkPad T580 -- also on SB 36 -- has also had fwupd issues. You might be onto something.

For the T580, though, I've just resorted to using Lenovo's bootable update ISOs rather than using fwupd. It's just unfortunately the case that this particular T16 Gen 1 update to fix suspend is only released to LVFS right now.

I will attempt to update my T16 Gen 1 by using Fedora Workstation (not Silverblue), but only if I can manage to avoid breaking my main installation. I'm on the road for a couple weeks right now and can't just trash this machine. :wink:

mrhpearson, 2022-09-26:
It looks like you're making progress in the other thread. Are we good to close this one?
If stuck - can you try booting from a Fedora Live Workstation USB?

davidstrauss, 2022-09-26:
> It looks like you're making progress in the other thread. Are we good to close this one?

Yes.

> If stuck - can you try booting from a Fedora Live Workstation USB?

I attempted this, but it didn't help. Lenovo has since released the bootable ISO version, which I've since installed after converting it to a thumb drive image with the eltorito tool.

closed 2022-09-26 by davidstrauss