fwupd/firmware-lenovo #270: ThinkPad T480 - Unable to update Intel Management Engine on Kubuntu 18.04

matthkoch, 2022-09-19:
```
fwupdmgr update
Devices with no available firmware updates: 
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.90.3987 to 184.92.4222?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ • 8 Intel Platform Update 2022.1                                             ║
║                                                                              ║
║ Version 11.8.92.4222 (LVFS: 184.92.4222)                                     ║
║                                                                              ║
║ Problem Fixes                                                                ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities:                          ║
║                                                                              ║
║ 20L5S1L500 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
```

If I reboot the system. I see the 
'Could not query capsule capabilities: Unsupported...System Reset' Message

I had the exact same problem before updating my bios with other components. I updated my bios and now the only update that is missing is this one, but i'm still not able to install it.

```
fwupdmgr --version
runtime   com.dell.libsmbios            2.4                                                                                                               
runtime   org.freedesktop.gusb          0.3.10                                                                                                            
compile   com.hughsie.libjcat           0.1.11                                                                                                            
compile   org.freedesktop.gusb          0.3.10                                                                                                            
runtime   org.freedesktop.fwupd-efi     1.0                                                                                                               
runtime   com.hughsie.libjcat           0.1.11                                                                                                            
runtime   org.kernel                    5.4.0-125-generic                                                                                                 
compile   org.freedesktop.fwupd         1.8.4                                                                                                             
runtime   org.freedesktop.fwupd         1.8.4
```

```
fwupdmgr get-devices --show-all-devices                                                                                                
LENOVO 20L5S1L500                                                                                                                                         
│                                                                                                                                                         
├─Core™ i7-8550U CPU @ 1.80GHz:                                                                                                                           
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027                                                                                        
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               b87a926d-189c-49a1-b1ff-921099b1de89
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
[... 179 lines cut ...]
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                3.5"/720 kB floppy services are supported (int 13h)
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.45
        Firmware Revision: 1.22

Handle 0x0023, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Abbreviated
        Installable Languages: 1
                en-US
        Currently Installed Language: en-US
```

kmauleon, 2022-11-16:
hi... please update BIOS System Firmware first >> 1.46 is available
Then please try to update MEFW after successful BIOS update.  Thank you very much.

kmauleon, 2022-12-12:
closing with no update.  please reopen if needed.  thank you very much.

kmauleon, 2022-12-13:
hi @matthkoch can you please try to update one firmware at a time... BIOS System or EC firmware first... then after successful update, MEFW...

kmauleon, 2022-12-14:
BTW, can you please share MEFW version in BIOS Menu as well?  Thank you very much.

matthkoch, 2022-12-15:
Here is my currently installed version:

```
UEFI Device Firmware
  DeviceId:             ea0fd60df22477668e549841f6a22a8c31cae587
  Guid:                 9e21f98b-fe98-455c-b388-da5450ab6979
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.90.3987
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2022-12-15
  UpdateState:          success

```

The 2 upcoming updates
```
UEFI Device Firmware has firmware updates:
Device ID:               ea0fd60df22477668e549841f6a22a8c31cae587
GUID:                    9e21f98b-fe98-455c-b388-da5450ab6979
ID:                      com.lenovo.ThinkPadN24RN.firmware
Update Version:          184.93.4323
Update Name:             ThinkPad T480 Consumer ME Update
Update Summary:          Lenovo ThinkPad T480 Consumer ME Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(c9fb9faff6bc5dc8e5daae753628a69636084297)
Update Location:         https://fwupd.org/downloads/1a0b4c3098c9846a01c3533f9f792e9aebdb94bf8d6349095265cca1762f725d-Lenovo-ThinkPad-T480-ConsumerMEFirmware-11.8.93.4323.cab
Update Description:       • 8 Intel Platform Update 2022.3
                         
                         Version 11.8.93.4323 (LVFS: 184.93.4323)
                         
                         Problem Fixes Mitigated the following security vulnerabilities:
                         
ID:                      com.lenovo.ThinkPadN24RN.firmware
Update Version:          184.92.4222
Update Name:             ThinkPad T480 Consumer ME Update
Update Summary:          Lenovo ThinkPad T480 Consumer ME Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(3764d0099ea6e73185a69ea81e1c192b788e2adc)
Update Location:         https://fwupd.org/downloads/cf79536576f3ea4fa32a1d2ff76254a6ca4b2360f7d8598d2ccb7a8848fa5d77-Lenovo-ThinkPad-T480-ConsumerMEFirmware-11.8.92.4222.cab
Update Description:       • 8 Intel Platform Update 2022.1
                         
                         Version 11.8.92.4222 (LVFS: 184.92.4222)
                         
                         Problem Fixes
                         
                          • Mitigated the following security vulnerabilities:
                         
```

Currently using this:
```
fwupdmgr --version

client version: 1.2.14
compile-time dependency versions
        gusb:   0.2.11
        efivar: 34
daemon version: 1.2.14
```
It seems not possible to install a specific version with the version of fwupdmgr that i use.
Am I right?

kmauleon, 2022-12-16:
can you please update fwupd? version is too old

kmauleon, 2023-01-05:
hi @matthkoch happy new year... have you tried updating fwupd?

matthkoch, 2023-02-13:
Sorry for taking so long. I had to update my OS to get to the new fwupdmgr version.
Now i'm using:
```
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-60-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

I currently try to install these updates, but i always get the same error.

```
│
├─Intel Management Engine:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Summary:            UEFI ESRT device
│ │   Current version:    184.90.3987
│ │   Minimum Version:    0.0.1
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               9e21f98b-fe98-455c-b388-da5450ab6979
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─ThinkPad T480:
│ │     New version:      184.93.4323
│ │     Remote ID:        lvfs
│ │     Release ID:       15487
│ │     Summary:          Lenovo ThinkPad T480 Consumer ME Firmware
│ │     License:          Proprietary
│ │     Size:             2,2 MB
│ │     Created:          2022-09-14
│ │     Urgency:          High
│ │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N24RN21W
│ │     Vendor:           Lenovo
│ │     Release Flags:    • Is upgrade
│ │     Description:      
│ │     • 8 Intel Platform Update 2022.3
│ │     
│ │     Version 11.8.93.4323 (LVFS: 184.93.4323)
│ │     
│ │     Problem Fixes Mitigated the following security vulnerabilities:
│ │   
│ └─ThinkPad T480:
│       New version:      184.92.4222
│       Remote ID:        lvfs
│       Release ID:       12646
│       Summary:          Lenovo ThinkPad T480 Consumer ME Firmware
│       License:          Proprietary
│       Size:             2,2 MB
│       Created:          2016-07-08
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N24RN20W
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:      
│       • 8 Intel Platform Update 2022.1
│       
│       Version 11.8.92.4222 (LVFS: 184.92.4222)
│       
│       Problem Fixes
│       
│       • Mitigated the following security vulnerabilities:
│     
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    0.1.46
  │   Minimum Version:    0.1.0
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              dc7f0308-1ef8-4774-9ba5-89a58c4d731c
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T480:
        New version:      0.1.47
        Remote ID:        lvfs
        Release ID:       18728
        Summary:          Lenovo ThinkPad T480 System Firmware
        License:          Proprietary
        Size:             9,5 MB
        Created:          2023-01-03
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad T480 System Firmware
        
        • Updated the Diagnostics module to version 04.28.000.

```

I tried 'fwupdmgr get-updates' and 'fwupdmgr install <FILE>', but the error stays the same.
Thx I really appreciate the good support.

kmauleon, 2023-02-15:
can i please see screen shot of BIOS Setup Menu.... 
have you tried updating BIOS first?

kmauleon, 2023-02-16:
as per owner... please try this....
We use below command to update BIOS when connect network.
fwupdmgr refresh
  fwupdmgr update

Could you ask the user try to local update Command if still fail? Attached package is ME package.
fwupdmgr --allow-older --allow-reinstall install  [Name].cab

ME Link on web:
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480-type-20l5-20l6/downloads/ds552809-intel-management-engine-118-firmware-consumer-update-utility-for-linux-thinkpad-t480-types-20l520l6?category=Motherboard%20Devices%20%28core%20chipset,%20onboard%20video,%20PCIe%20switches%29

kmauleon, 2023-03-24:
@matthkoch any update from your side please

kmauleon, 2023-05-25:
hi @matthkoch already forwarded this to ODM... thank you

kmauleon, 2023-06-07:
hi @matthkoch as per owner
This issue appears to be a FWupd tool or kubuntu version issue

[http://github.com/fwupd/firmware-lenovo/issues/235](https://apc01.safelinks.protection.outlook.com/?url=http%3A%2F%2Fgithub.com%2Ffwupd%2Ffirmware-lenovo%2Fissues%2F235&data=05%7C01%7Ckmauleon%40lenovo.com%7Ca8783d03cfd9470555c408db6734e29a%7C5c7d0b28bdf8410caa934df372b16203%7C0%7C0%7C638217249127461472%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=Z%2B3wLZWXLXpjI%2Fh9UVcgPtHvvsg48mulf5IMolj5Sw8%3D&reserved=0)

feedback from users:
I just created a live usb disk using mkusb-dus based on Ubuntu 20.04 with an added persistent storage, booted my machine from it and was able to install the firmware updates including the Intel Media Engine update successfully using "fwupdmgr update" from within the live environment. Very strange though.

kmauleon, 2024-04-05:
hi @matthkoch any update from your side?  BTW, please install newer fwupdmgr. latest is 1.9.x

kmauleon, 2024-05-06:
closing as not firmware issue... thank you

closed 2024-05-06 by kmauleon