fwupd/firmware-lenovo #271: Missing firmware for Lenovo ThinkCentre M90n-1 Nano i3,i5,i7

johannbg, 2022-09-25:
The latest available version for the Thinkcentre nano's is M2AKT34A.
The platform is supported to 2026 according to psref so Lenovo should be including those.  

@mrhpearson could you have a look since it seems like the firmware updates stopped getting uploaded to lvfs after firmware update M2AKT25A.

mrhpearson, 2022-09-26:
Sigh - this is the 2nd ThinkCentre platform we've found this. They seem to have a flaw in their process so the FW never actually gets released. For all versions from 2CA (up to 34A) are all on LVFS - but under embargo so nobody can access them :/

I've sent a request to the FW team to please move them to stable once they've finished testing. From the download stats they have tested these so I have to assume it's just a mistake that they're not available. Will update when I hear back

Internal ticket LO-2047

mrhpearson, 2022-09-29:
FW released to stable - closing ticket, but let me know if any issues.
Thanks for flagging this!

closed 2022-09-29 by mrhpearson