fwupd/firmware-lenovo #272: Missing firmware for Thinkpad T495  -- 1.31   (R12UL61W)   (2022-09-01)

speachy, 2022-10-03:
https://download.lenovo.com/pccbbs/mobiles/r12ul61w.zip
https://download.lenovo.com/pccbbs/mobiles/r12ul61w.txt

Interestingly, they provide a direct Linux update cab in that zip file, but when trying 'fupdmgr install XXX.cab' it claims 'No supported devices found' This is with fwupd 1.8.5 on Fedora 36.

mrhpearson, 2022-10-03:
It is on LVFS - but under embargo and doesn't look like they have done the testing on it yet :( 
I've flagged this to the FW team (internal ticket  LO-2058).

speachy, 2022-10-03:
Well, it's publicly posted on the T495's download site.  :)

https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t495-type-20nj-20nk/20nj/20njcto1ww/pf21vbhd/downloads/ds539877-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-t495?category=BIOS%2FUEFI

(Come to think of it, the v1.30 firmware I got from LVFS last time never got posted to the download site, so.. methinks lenovo has some internal process hiccups going on..)

Thanks for escalating this!

ChiWei-Chen, 2022-10-06:
Hi @speachy , I work with @mrhpearson.
The FW steam seems released the new BIOS 1.37 to stable, please let me know if this doesn't work for you. Thanks!
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR12ETXXW.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR12ETXXP.firmware

speachy, 2022-10-06:
I just installed the update through fwupd, and it seems okay.  Thanks for getting this handled so quickly!

closed 2022-10-06 by speachy