fwupd/firmware-lenovo #273: Error "Use the MCU to update the DMC device" with `ThinkPad Universal Thunderbolt 4 Dock`

drahnr, 2022-10-04:
**Describe the bug**

There is no update path for the "ThinkPad Universal Thunderbolt 4 Dock"

Consequences are that I am hitting a known and fixed issue in a later firmware update regarding issues in multiscreen setups where EDID information is corrupted.

**Steps to Reproduce**

```
├─ThinkPad Thunderbolt 4 Dock:
│ │   Device ID:          c50d19eb8a842a9776858e143510d2b26de06066
│ │   Current version:    10.10
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Serial Number:      1S40B0ZDZJ08WA
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       ba3efaf3-4e18-5209-aa15-1401537203a3 ← USB\VID_17EF&PID_30B4&REV_0200
│ │                       275e4695-9b4e-5263-835e-8681bec8cd1a ← USB\VID_17EF&PID_30B4&CID_40B0
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │ 
│ ├─Audio Controller:
│ │     Device ID:        dc7c890bd6722b24e7ee5f2c35e3d729e2c6ad58
│ │     Current version:  49-0E-41
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             e1443fec-2b49-5bd0-b921-662ed11e88df ← USB\VID_17EF&PID_30B4&CID_AUDIO
│ │   
│ ├─Display Port 5:
│ │     Device ID:        ab264811c2a64d1ded0076c78bdaebd7b0d527fd
│ │     Current version:  5.06.002
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             755e347f-4ae7-53f8-ad69-45b3e9145ed8 ← USB\VID_17EF&PID_30B4&CID_DP5x
│ │   
│ ├─Display Port 6:
│ │     Device ID:        88ddeedd067f018c6e0263230ea2ce4fbdb5e676
│ │     Current version:  6.04.000
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             a7b49eae-1c1d-5b83-ac90-9bbe78ee39c1 ← USB\VID_17EF&PID_30B4&CID_DP6x
│ │   
│ ├─Dock Management Controller:
│ │     Device ID:        3df62531bf83371b93125751f5aec7834236f16e
│ │     Current version:  0.1.91
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             3edd934f-9800-59c9-b02a-11be480d72e5 ← USB\VID_17EF&PID_30B4&CID_DMC
│ │   
│ ├─Dock Management Controller:
│ │     Device ID:        4d25aff9108407039b1120921eb5b00d95cd9d70
│ │     Current version:  21.9
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             c287973a-cc6b-5d2a-ac71-3956a512f73d ← USB\VID_17EF&PID_30B4&CID_MCU
│ │   
│ ├─Dock Management Controller Information:
│ │     Device ID:        65f3ac8ac9e7bdfae865e24a716e297c0696f0c6
│ │     Current version:  10.10
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     Serial Number:    1S40B0ZDZJ08WA
│ │     Update Error:     Use the MCU to update the DMC device
│ │     GUIDs:            697a6a95-464e-5b90-a46e-73beab89316e ← USB\VID_17EF&PID_30B5
│ │                       699805f6-a065-5806-a5a2-cf4c4c5c8d21 ← USB\VID_17EF&PID_30B5&REV_1010
│ │   
│ ├─Ethernet Adapter:
│ │     Device ID:        5b77c4c22cce1f5a895d882ded8e5ce792d0f9cc
│ │     Current version:  1.73.1
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             abbc9b3e-43fe-538f-bfaf-b7e74b3fd47c ← USB\VID_17EF&PID_30B4&CID_I255
│ │   
│ ├─Power Delivery:
│ │     Device ID:        cfece361b188e708e823976613bc5638604ae514
│ │     Current version:  12.5.31
[... 368 lines cut ...]
│     Update State:       Success
│     GUID:               0b8282f0-4796-4cbc-9395-677ab9738050
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─USB4 host controller:
      Device ID:          b3577d6a456151fdc17c88c64a9fa126f8d423e4
      Summary:            Unmatched performance for high-speed I/O
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
      Device Flags:       • Internal device
                          • System requires external power source

```

**Additional questions**

- Operating system and version:
Fedora 36 x86_64 Workstation
- Have you tried rebooting?
Yes
- Is this a regression?
Probably not

drahnr, 2022-10-07:
CC @fbatschl could you post your spec, iirc you already got an update to your dock? I suspect that it's either a different dock or there has been no firmware for a while

lohedges, 2022-10-15:
There is currently no firmware for this dock in LVFS. As of yesterday (October 14th) there is no firmware available whatsoever, since Lenovo have pulled the Windows version. I only discovered this having finally got access to a Windows laptop in order to update. (What are the chances of the firmware being removed on the same day?) 

Having found the latest firmware on the Internet Archive's WayBack machine it turns out that the update wouldn't have worked anyway due to the Digicert issue reported [here](https://forums.lenovo.com/t5/Enterprise-Client-Management/Lenovo-System-Update-5-07-0137-won-t-open-when-installed-cleanly/m-p/5167181). (This wasn't obvious since running the installer directly would silently fail. Only when extracting and running manually do you get a log file containing the certificate error.)

I tried getting info from Lenovo chat, but they obviously have no clue as to when an update will be available. This obviously has nothing to do with the `fwupd` team, just wanted to highlight what an absolute car crash the firmware situation is with this dock. Given the above, I doubt we'll be seeing anything in LVFS soon. I'm close to cutting my losses and going back to the working TB3 dock.

drahnr, 2022-10-15:
For cross referencing ease, while somewhat related, https://forums.lenovo.com/t5/Fedora/Fedora-Linux-Thunderbolt-4-Dock-Update/m-p/5173614?page=1#5762965

mrhpearson, 2022-10-17:
We have the FW for the USB-c and TBT3 dock (though I think that needs updating now) but not the TBT4 dock yet. The docking team are working on releasing it but hit some issues in the last round of testing that have delayed it being released.

I've asked for an update and ETA. Will try and remember to update here (and in the linked forum topic) but please nag me if you don't hear anything.

lohedges, 2022-10-26:
Just to say that the v1.0.13 firmware installer has just been released for Windows and I have managed to update using a friend's Windows laptop. I'll let you know if it solves the flickering/drop-out issue, or any of the other problems that I have been experiencing. (From the README it looks like nothing of significance has chanded since v1.0.11, which was released back in August, but didn't work because of the Digicert issue.)

lohedges, 2022-10-26:
The firmware update appears to have fixed the flickering and drop-out issue. (I've not experienced any yet, at least.) However, the external displays are _still_ not detected on boot and suspend / resume doesn't work either. (I've experienced a crash, a hang, and a resume with no detection of the external displays.) I have no such issues with the TB3 workstation dock.

mrhpearson, 2022-10-26:
Thanks for the update. As a note (as this is the fwupd project) we are expecting that version to be released on LVFS at the beginning of November - there were some issues around 1.0.11 that needing resolving and delayed it being released.
Which platform are you testing against - I couldn't spot it in the above.

lohedges, 2022-10-26:
Thanks. I am running Arch Linux on a ThinkPad P1 Gen 5. I'm using the proprietary NVIDIA driver, with Optimus Manager managing the switching of graphics cards via the provided kernel parameters. Everything worked fine on the TB3 workstation dock that I was using before. I also have access to a P1 Gen 2 using the same setup and experience the exact same problem.

I recall reading reports of similar issues when I was first trying to debug, i.e. displays not detected on boot or resume. Will try to dig some links up tomorrow.

Happy to provide more system information if needed.

Cheers.

mrhpearson, 2022-10-26:
OK - thanks. We haven't done the (Ubuntu) dock cert with the P1G5 yet (still wrapping up the platform enablement - dock cert comes after because docks are....complicated/painful/difficult). I was using my dock with a P1 G3 and G4 previously succesfully though.
It's quite possible there are still fixes to come.

lohedges, 2022-10-28:
Just an update to say that the screens have been rock-solid for the past few days following the firmware update. No flickering or drop-outs whatsoever.

Following some debugging, I've discovered that the disconnected external displays on boot results from the `nvidia-drm modeset=1` option, which is enabled by default with [optimus-manager](https://github.com/Askannz/optimus-manager), as it should be to prevent tearing with PRIME. On disabling this option the external displays are correctly detected as connected when the X server starts, but I experience screen flickering and drop-outs. (Much like before applying the latest firmware, in fact.)

I'm not sure if this is in any way related to the dock firmware, so have posted a question on the Arch Forums [here](https://bbs.archlinux.org/viewtopic.php?pid=2064668#p2064668), as well as the optimus-manager [issue tracker](https://github.com/Askannz/optimus-manager/issues/526). I might also post the same question on the Linux subsection of the NVIDIA developers forum.

Let me know if you have any thoughts. As mentioned, the _exact_ same setup works with the Thunderbolt 3 Workstation Dock without issue.

lohedges, 2022-10-31:
I spoke too soon. The graphical glitches returned today. Although less frequent than before, I am still experiencing an annoying level of flickering and complete drop-out. Coupled with the startup/resume issue, my solution is simply to return to using the TB3 workstation dock. I simply can't afford to waste any more time getting the TB4 dock to work correctly.

DrBu7cher, 2022-11-14:
I need this update working too. Am I looking at the wrong place or is there still no support for the TB4 dock? (40B0) :/

mrhpearson, 2022-11-14:
I'm not sure on the flickering issue; but on the FW update issue the last update from the dock team was:

> After LVFS update to 1.8.6, one important notice in command window disappears.
> Looks like Linux update some setting in Oct which cannot fully compatible to our coding.

They're working on it but that's why it hasn't been released yet :(

DrBu7cher, 2022-11-15:
Thank you @mrhpearson for this information. Hope this does not take too long...

I would like to share a "solution" (more of a workaround) that worked for me and I am working with right now. 2 external Monitors connected, one over DP, one over HDMI, no flicker.
I have a P1 Gen 4 running (K)ubuntu 22.04 and the 40B0 dock (40B00300EU).

- I downgraded to kernel version `5.19.0-051900-generic` (using the [mainline kernel installer](https://www.linuxcapable.com/how-to-install-linux-mainline-kernel-on-ubuntu-22-04-lts/)) and uninstalled all kernels after that/younger kernel versions.
- I think I rebooted, not sure
- I disconnected the dock
- Then I got the DisplayLink Driver installed (from [master](https://github.com/AdnanHodzic/displaylink-debian) - I guess the latest release would work too?).
- I executed `sudo prime-select nvidia`
- Connect the dock
- exec `sudo nvidia-settings`
- One monitor worked for me (I think I had only one connected at this time)
- Close nvidia settings and rebooted - the other monitor worked too
- never rebooted since then. :sweat_smile: 

Don't take my words for a "How to get it working for you (for now)", but maybe it get's you on a path where it does work after some trial and error? Would make me happy to hear!

drahnr, 2022-11-24:
I successfully updated my dockingstation - thanks @mrhpearson for following through :heart:  , which was the initial concern of this issue. It never was intended as a place to rant about abysmal performance or any workarounds.

@mrhpearson you might want to direct people to a place where one can leave bugreports.

Closing

closed 2022-11-24 by drahnr

norru, 2023-05-20:
FYI, I have just observed this issue on

- ThinkPad Carbon X1 Gen11
- Debian Sid
- Linux thinkpad-x1-debian 6.1.0-9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux

```
│ ├─Dock Management Controller Information:
│ │     Device ID:        b513cf41bc1e11e220c56fcec9b93006bb7a1bc1
│ │     Current version:  10.5
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     Serial Number:    1S40B1ZEZA08GV
│ │     Update Error:     Use the MCU to update the DMC device
│ │     GUIDs:            697a6a95-464e-5b90-a46e-73beab89316e ← USB\VID_17EF&PID_30B5
│ │                       9b2a0c77-248f-59ad-afa6-cee22caae03b ← USB\VID_17EF&PID_30B5&REV_1005
```

danpawlik, 2023-06-14:
It seems that after update fwupd to 1.9.2:
```fwupdmgr --version

compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.2
runtime   org.kernel                    6.4.0-0.rc5.41.fc39.x86_64
```
And `fwupdmgr reinstall` on dockstation it helps. No more issue.
Earlier I was reinstalling the firmware on `fwupd-1.8.15-1` and the update error still appears.

Glad that new version work well.
Thank you!