fwupd/firmware-lenovo #276: Lenovo System Firmware update 1.55 -> 1.56 deletes all SecureBoot keys and certs, disables SecureBoot / turns on Setup Mode and deletes all UEFI Boot entries

das-menschy, 2022-10-15:


das-menschy, 2022-10-15:
I also made two photos during the Lenovo System Firmware update, but I guess they won't help; because they report success and no error: 
[image]
[image]

das-menschy, 2022-10-15:
Is this not a bug, but an undocumented feature to disable Secure Boot? :laughing: :smile: 
I guess this issue is a security vulnerability.... Maybe I should not have gone public. Anyway, here we are.

das-menschy, 2022-10-15:
Could be the same bug as #266 . Lenovo System Firmware updates seem to delete quite a lot of things...

hughsie, 2022-10-17:
@mrhpearson if this is disabling SecureBoot by just updating we should probably spin up secalart too. Do you want me to pull the update whilst we debug?

mrhpearson, 2022-10-18:
Sorry for the slow reply.
I'll definitely follow up with the FW team on this, but unsure on pulling it - looks like a lot of succesful downloads and updates from the stats.
My concern is that we're not going to be able to reproduce this and if it's a one off we end up stuck. Can you give me a day to follow up with the FW team and see what they say? 

It's hard enough getting them to do the FW updates for the older platforms where LVFS support wasn't in the plan of record without moving FW back and forth. If that's a bad reason and you think I'm under-estimating the severity I'm OK to be overridden :)
Mark

mrhpearson, 2022-10-18:
As a note - internal ticket # LO-2077

das-menschy, 2022-10-18:
During the firmware update, a **"System reset"** message was displayed. I guess the "System reset" is the problem. 
Solution: 
* either don't do a "System reset" during the firmware update (for this, the *.cab file would have to be changed, I guess) 
* or make a backup of all system settings (UEFI boot variables, UEFI secure boot certs, etc.) **before** the firmware update / system update, and **reload** the backup file into NVRAM **after** the firmware update / system reset. 

The absolute minimum should be a warning from fwupdmgr: fwupdmgr should advise the users to make a **backup** of all UEFI boot entries and all Secure Boot entries **before** running the firmware update.

mrhpearson, 2022-10-18:
Those entries really shouldn't be impacted by a FW update. The system is reset during the update so that's pretty normal and I don't think is necesarily the problem.

I saw your note in #266 and that is making me double guess if this should be pulled or not. We didn't get to the bottom of what was going on there.

Afraid I really need feedback from the FW team. What you're seeing is not common - I've not seen this reported on a FW update before.

das-menschy, 2022-10-18:
Let's see who is faster to implement: the firmware developers at Lenovo or the developers of fwupdmgr. 
I would guess the latter. ;-)

bubbleguuum, 2022-11-11:
I suppose I got hit by this, updating my ThinkPad P72 BIOS.
After BIOS update was completed, it rebooted into Windows.
My openSUSE EFI boot entry had been removed and I had to boot on a live Linux USB distro
to recreate it with `efibootmgr`.

mrhpearson, 2022-11-11:
Thanks for the note.
If possible - we really need to catch this and see if rolling back the BIOS to the previous version brings back the entries.
I personally don't think it will - but the BIOS team wanted to confirm

We're unable to reproduce which is making it really hard to figure out what the issue is. It's seemingly a rather rare occurence :(

mark

bubbleguuum, 2022-11-11:
Thank you for the comment.

Note that the same thing happened a few month ago on this machine when I updated the BIOS the same way.
I wanted to updated to a new BIOS today and check if that problem was still there, which it did.
Since I can probably reproduce it (downgrading the BIOS, recreating my GRUB EFI entry if necessary, then upgrading again), I can if you wish do that test with any additional info you'd need (more logging, etc) to understand what happens.
Let me know if I should enable anything for additional logging, or any other instructions.

bubbleguuum, 2022-11-12:
I also noticed that Secure Boot is disabled because it is in Setup Mode.
All keys managed by `mokutil` have been removed. 

```
~> mokutil --sb-state
SecureBoot disabled
Platform is in Setup Mode
``` 

So now I have to research how to fix this situation on openSUSE Tumbleweed...

mrhpearson, 2022-11-13:
Thanks for the note - useful you can reproduce it and I'll forward those details to the FW team.
I'm really intrigued by your system being in setup mode - can you go into the BIOS setup menu (F1 during early boot) and reset it so that it's no longer in setup mode? I wonder if somehow it's related? (grasping at straws a little bit)

das-menschy, 2023-02-04:
The same problem (all UEFI secure boot entries deleted, all UEFI boot entries deleted, Secure Boot disabled / Setup Mode activated) appeared again when updating the System Firmware from version 0.1.56 to 0.1.57: 
```
[das_menschy@linux-on-20hr002mmx ~]$ sudo fwupdmgr get-history
LENOVO 20HR002MMX
│
... unimportant stuff ...
│     
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Previous version:   0.1.56
  │   Update State:       Success
  │   Last modified:      2023-01-30 20:37
  │   GUID:               798ffd60-f10e-4ac4-8939-c8beabfe55b4
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 5th:
        New version:      0.1.57
        Remote ID:        lvfs
        Release ID:       17085
        Summary:          Lenovo ThinkPad X1 Carbon 5th System Firmware
        License:          Proprietary
        Size:             9,4 MB
        Created:          2022-11-28
        Urgency:          High
        Tested by Lenovo:
          Tested:         2022-12-07
          Distribution:   debian 11
          Old version:    0.1.56
          Version[fwupd]: 1.5.7
        Vendor:           Lenovo
        Description:      
        Lenovo System Firmware Version 1.57
        
        Important updates
        
        • Update includes a security fix.
        
        New functions or enhancements
        
        • Updated the Diagnostics module to version 04.27.000.

```
I could see the **"Reset system"** message during the firmware update. I guess the deletion of all these variables happened during the **"Reset system"**.

mrhpearson, 2023-02-06:
@kmauleon and @ChiWei-Chen - tagging you as this is still under investigation by the FW team.

mrhpearson, 2023-03-24:
As a quick update - we think we may have root cause on this issue. Still some work to do to confirm; and get a fix released but there is progress.
Mark