fwupd/firmware-lenovo #278: Thinkpad P15 gen2: ACPI BIOS Error (bug): Failure creating named object

acarlton, 2022-10-21:
Hello,

I'm dual-booting Ubuntu 22.10 (freshly upgraded distribution from 22.04 today) alongside Windows 11 on my Thinkpad P15 Gen 2i for a while. For quite some time (under 22.04), I've been seeing the following in `dmesg` logs during startup:

```
[    0.750660] ACPI BIOS Error (bug): Failure creating named object [\_SB.PC00.SAT0.PRT0.PWRG], AE_ALREADY_EXISTS (20220331/dswload2-326)
[    0.750671] fbcon: Taking over console
[    0.750682] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20220331/psobject-220)
```

I was hoping the next distribution upgrade and kernel would fix it, but nope. My research to this point is indicating that an error like this  might be a BIOS bug:
* https://bugzilla.kernel.org/show_bug.cgi?id=215101

I'm submitting this ticket in hopes of making the Thinkpad P15 firmware better, pointing toward a possible bug and learning more about the firmware systems in relationship to Linux. Admittedly, topics relating to BIOS and the kernel are pushing the boundaries of my current technical knowledge, but I'm able to provide more information and help out where I can with some guidance.

Thanks in advance. I've been really impressed with the level of support coming from the team in this repository on other issues. Keep up the good work!

mrhpearson, 2022-10-21:
Hi,

Thanks for the note. Would you mind posting this on the Lenovo Linux forums? https://forums.lenovo.com/linux
I try and keep the fwupd issues for things that are fwupd or firmware update related which this isn't.
I should pick up the thread on the Lenovo forum and can get a ticket open to the FW team from there.

Sorry to make you jump through hoops. As a note; I believe the issue is minor and has no impact on functionality - but I agree it's good to get these things fixed where we can

Mark

closed 2022-10-21 by mrhpearson

Adesoji1, 2023-07-27:
I still face this issue today on Lenovo Legion. How do I resolve it. Help