fwupd/firmware-lenovo #279: T470s: Thunderbolt NVM version 23 missing

dkadioglu, 2022-10-24:
Since 29th of September an updated version 23 is available for the T470s Thunderbolt. The update file for Windows has version 20.22.09.02-N1WTF16W. May this be added to fwupd.org? Thanks in advance.

kmauleon, 2022-11-18:
hi @dkadioglu already contacted owner... waiting for feedback

kmauleon, 2022-12-01:
hi @dkadioglu as per owner, they will target to release by end of January 2023.... thank you very much.

dkadioglu, 2022-12-02:
@kmauleon Thank you very much for asking them! I'll keep this open, until the update is there.

dkadioglu, 2022-12-26:
The update is already available on LVFS. However, when I try to upgrade, I get the following error message:
```
could not start thunderbolt device upgrade: could not write to /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0/0-0/nvm_authenticate: Das Argument ist ungültig
```
["Das Argument ist ungültig" is german and translates to "Invalid argument"]

Anything I can do here?

kmauleon, 2022-12-27:
@dkadioglu I will ping the owner for this one... Happy Holidays!

dkadioglu, 2022-12-27:
@kmauleon Thank you very much! And Happy Holidays!

hughsie, 2022-12-27:
> and translates to "Invalid argument

That might be a fwupd bug -- @superm1 do you remember the details here?

dkadioglu, 2022-12-28:
Is there anything I can provide to help investigate?

superm1, 2022-12-28:
> Is there anything I can provide to help investigate?

You can turn on dynamic debugging for the thunderbolt by adding `thunderbolt.dyndbg='+p'` to the kernel command line.

Then run the update specifically using `fwupdtool update --verbose` and share the `dmesg` from the boot you reproduce the problems as well as the fwupdtool verbose output. Hopefully dynamic debugging gives more insight.

dkadioglu, 2023-01-01:
Here is the dmesg with `thunderbolt.dyndbg='+p'` on the kernel command line:
```
[    0.000000] Linux version 6.1.1-zen1-1-zen (linux-zen@archlinux) (gcc (GCC) 12.2.0, GNU ld (GNU Binutils) 2.39.0) #1 ZEN SMP PREEMPT_DYNAMIC Wed, 21 Dec 2022 22:27:59 +0000
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-linux-zen root=UUID=26f043a3-5b9e-4dcb-bedb-a600719b86b9 rw loglevel=3 quiet rd.systemd.show_status=auto rd.udev.log_priority=3 resume=UUID=7b0455cd-0c5c-478f-aca8-18e3a577f3d8 nowatchdog printk.devkmsg=on sysrq_always_enabled fbcon=font:VGA8x16 msr.allow_writes=on intel_iommu=on iommu=pt i915.enable_dpcd_backlight=0 thunderbolt.dyndbg=+p
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 2032
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009cfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040400000-0x0000000065561fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000065562000-0x0000000065562fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000065563000-0x0000000065563fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000065564000-0x000000006eea9fff] usable
[    0.000000] BIOS-e820: [mem 0x000000006eeaa000-0x000000006ef71fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000006ef72000-0x000000006ff2cfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000006ff2d000-0x000000006ff99fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000006ff9a000-0x000000006fffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000006ffff000-0x000000006fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000070000000-0x0000000077ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000078600000-0x000000007c7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000006817fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.50 by Lenovo
[    0.000000] efi: TPMFinalLog=0x6ff92000 SMBIOS=0x6f0d9000 SMBIOS 3.0=0x6f0d6000 ACPI=0x6fffe000 ACPI 2.0=0x6fffe014 ESRT=0x6efb5000 MEMATTR=0x69e7d018 
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: LENOVO 20HGS00V00/20HGS00V00, BIOS N1WET69W (1.48 ) 09/26/2022
[    0.000000] tsc: Detected 2900.000 MHz processor
[    0.000000] tsc: Detected 2899.886 MHz TSC
[    0.001025] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001031] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001051] last_pfn = 0x681800 max_arch_pfn = 0x400000000
[    0.001328] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002533] last_pfn = 0x70000 max_arch_pfn = 0x400000000
[    0.023454] esrt: Reserving ESRT space from 0x000000006efb5000 to 0x000000006efb5088.
[    0.023471] Using GB pages for direct mapping
[    0.024583] Secure boot disabled
[    0.024584] RAMDISK: [mem 0x35339000-0x36993fff]
[    0.024590] ACPI: Early table checksum verification disabled
[    0.024595] ACPI: RSDP 0x000000006FFFE014 000024 (v02 LENOVO)
[    0.024601] ACPI: XSDT 0x000000006FFC1188 0000FC (v01 LENOVO TP-N1W   00001480 PTEC 00000002)
[    0.024611] ACPI: FACP 0x000000006FFF5000 0000F4 (v05 LENOVO TP-N1W   00001480 PTEC 00000002)
[    0.024620] ACPI: DSDT 0x000000006FFCF000 02134E (v02 LENOVO SKL      00000000 INTL 20160527)
[    0.024626] ACPI: FACS 0x000000006FF3E000 000040
[    0.024630] ACPI: SSDT 0x000000006FFFC000 0003CC (v02 LENOVO Tpm2Tabl 00001000 INTL 20160527)
[    0.024636] ACPI: TPM2 0x000000006FFFB000 000034 (v03 LENOVO TP-N1W   00001480 PTEC 00000002)
[    0.024641] ACPI: UEFI 0x000000006FF53000 000042 (v01 LENOVO TP-N1W   00001480 PTEC 00000002)
[    0.024646] ACPI: SSDT 0x000000006FFF7000 0030E4 (v02 LENOVO SaSsdt   00003000 INTL 20160527)
[    0.024651] ACPI: SSDT 0x000000006FFF6000 0005B6 (v02 LENOVO PerfTune 00001000 INTL 20160527)
[    0.024656] ACPI: HPET 0x000000006FFF4000 000038 (v01 LENOVO TP-N1W   00001480 PTEC 00000002)
[    0.024661] ACPI: APIC 0x000000006FFF3000 0000BC (v03 LENOVO TP-N1W   00001480 PTEC 00000002)
[    0.024666] ACPI: MCFG 0x000000006FFF2000 00003C (v01 LENOVO TP-N1W   00001480 PTEC 00000002)
[    0.024671] ACPI: ECDT 0x000000006FFF1000 000053 (v01 LENOVO TP-N1W   00001480 PTEC 00000002)
[    0.024676] ACPI: SSDT 0x000000006FFCD000 0016F3 (v02 LENOVO ProjSsdt 00000010 INTL 20160527)
[... cut ...]
could not start thunderbolt device upgrade: could not write to /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0/0-0/nvm_authenticate: Das Argument ist ungültig
12:02:37.234 FuPlugin             finalize(modem_manager)
12:02:37.235 FuPlugin             finalize(flashrom)
12:02:37.235 FuPluginTest         destroy
12:02:37.237 FuProgress           raw timing data was { 0,048, 0,001, 0,010, 0,001, 0,067, 0,002 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 37 1 8 1 52 2 ] at ../fwupd-1.8.8/plugins/uefi-capsule/fu-uefi-capsule-plugin.c:906
```

superm1, 2023-01-01:
Here's the problem:
```
[  133.126520] thunderbolt 0000:03:00.0: 0: failed to authenticate NVM, power cycling
```

The NVM that is flashed might be corrupted.  I think you'll need service done on your machine.

superm1, 2023-01-01:
(Assuming that was tied to bootup not the newly flashed image that is)

If it's tied to the newly flashed image than Lenovo made a mistake and didn't post a signed image, or an image signed with the wrong signature.

maru-sama, 2023-01-17:
Hi for what it is worth I can reproduce the same error on my T470s

dkadioglu, 2023-01-24:
@superm1 Thanks for pointing this out and your suspected diagnoses.
@maru-sama Thanks for your confirmation.

Is there anything, we can do to test/debug for the real cause? The only thing, which comes to my mind is updating via windows - but if that works, I won't be able to help with finding out, what the cause for this issue is on linux, right?

kmauleon, 2023-02-01:
hi all... dev owner asked to retry updating again as below.... thank you...

Could you please help try to online update to have a try ?  we re-upload nvm 23 to lvfs and seems can update success on our system. 
If online update still fail, we may share .cab file for the customer to local update to have a try.

kmauleon, 2023-02-02:
hi this is current version link >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1WTF.firmware can you please try this one?

dkadioglu, 2023-02-02:
With the reuploaded firmware, the update was successful! However, I had to enable the testing repository to see it. When I started the update, the download was not successful, according to the following output:
```
12:03:26.878 FuDebug              Verbose debugging enabled (on console 1)
12:03:26.879 FuMain               locked /run/lock/fwupdtool
12:03:26.880 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
12:03:26.887 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Laden …                  [-                                      ]12:03:26.887 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
12:03:26.887 FuCommon             reading /etc/fwupd/daemon.conf with 2494 bytes
12:03:26.887 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
12:03:26.887 FuConfig             using autodetected max archive size 4,3 GB
12:03:26.889 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Laden …                  [ -                                     ]12:03:26.890 FuRemoteList         loading /usr/share/fwupd/metainfo
12:03:26.891 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
12:03:26.891 XbSilo               file: aa8366b6-bfd4-9c00-54a2-5b8afbb6aaee, current:aa8366b6-bfd4-9c00-54a2-5b8afbb6aaee, cached: (null)
12:03:26.891 XbSilo               loading silo with file contents
12:03:26.891 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
12:03:26.892 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
12:03:26.892 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
12:03:26.893 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
12:03:26.893 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
12:03:26.893 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
12:03:26.894 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs-testing
12:03:26.894 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
12:03:26.895 FuRemoteList         ignoring unfound remote fwupd
12:03:26.895 FuRemoteList         ordering lvfs-testing=lvfs+1
Laden …                  [                                       ]12:03:26.896 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
12:03:26.896 Jcat                 loaded 1 certificates
12:03:26.896 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
12:03:26.897 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:03:26.897 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
12:03:26.898 Jcat                 loaded 1 certificates
12:03:26.898 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
12:03:26.898 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:03:26.899 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
12:03:26.901 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
12:03:26.902 FuEngine             client certificate now exists: nothing to do
Laden …                  [                                       ]12:03:26.902 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
12:03:26.903 FuHistory            got schema version of 8
Laden …                  [*                                      ]12:03:26.914 FuPlugin             load(/usr/lib/fwupd-1.8.10/libfu_plugin_modem_manager.so)
12:03:26.921 FuContext            added udev subsystem watch of platform
12:03:26.922 FuContext            added udev subsystem watch of block
12:03:26.922 FuContext            added udev subsystem watch of pci
12:03:26.923 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
12:03:26.923 FuPluginDell         Using libsmbios 2.4
12:03:26.923 FuContext            added udev subsystem watch of i2c
12:03:26.923 FuContext            added udev subsystem watch of i2c-dev
12:03:26.923 FuContext            added udev subsystem watch of hidraw
12:03:26.924 FuContext            added udev subsystem watch of gpio
12:03:26.924 FuContext            added udev subsystem watch of mei
12:03:26.924 FuContext            added udev subsystem watch of iommu
12:03:26.924 FuContext            added udev subsystem watch of video4linux
12:03:26.924 FuContext            added udev subsystem watch of msr
12:03:26.925 FuContext            added udev subsystem watch of mtd
12:03:26.934 FuContext            added udev subsystem watch of nvme
12:03:26.940 FuContext            added udev subsystem watch of drm
12:03:26.941 FuContext            added udev subsystem watch of drm_dp_aux_dev
12:03:26.941 FuContext            added udev subsystem watch of serio
12:03:26.941 FuPluginTest         init
12:03:26.941 FuContext            added udev subsystem watch of thunderbolt
12:03:26.941 FuContext            added udev subsystem watch of tpm
12:03:26.941 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
12:03:26.942 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
[... cut ...]
12:06:37.312 FuProgress           raw timing data was { 0,085, 0,002, 0,031, 0,001, 0,101, 0,005 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 38 1 14 1 45 2 ] at ../fwupd-1.8.10/plugins/uefi-capsule/fu-uefi-capsule-plugin.c:906
12:06:37.329 FuProgress           raw timing data was { 0,000, 9,064, 27,936, 0,001 } -- steps were set as [ 0 100 0 0 ] but should have been [ 0 24 75 0 ] at ../fwupd-1.8.10/plugins/thunderbolt/fu-thunderbolt-device.c:417
```

Thank you very much for fixing the firmware!

kmauleon, 2023-02-03:
hi @dkadioglu yes the firmware is still in testing mode... let's close this one once the firmware is put to stable.  thank you very much for the feedback

kmauleon, 2023-02-23:
firmware already moved to stable.... closing... thank you very much

closed 2023-02-23 by kmauleon