fwupd/firmware-lenovo #280: P17 Gen 2i: Failed ME firmware. No sound on Ubuntu

RayyanRiaz, 2022-10-24:
The firmware update from 1.22 to 1.23 was unsuccessful, and now at every reboot/start, I get into a screen which says:

`Reading ME Firmware...Please do not power off! 30% Complete`
After 48% progress, it restarts and quickly goes to 10% and then jumps to 60%, and then fails, saying:
` Failed to recover ME Firmware...`

Also, this thing happens twice before I can go to BIOS/grub.
I am giving these details because the issue seems similar to [226](https://github.com/fwupd/firmware-lenovo/issues/226).

Using `sudo dmidecode | grep SKU` I verified that I have a corporate SKU.

## Details of What Happened

### 1: Automatic Ubuntu update of firmware failed: 
Normally I do not have to install these updates, as they are automatically installed by ubuntu software & updates manager. I only noticed this issue when the automatic update from 1.22 to 1.23 failed. 

### 2: I tried to reinstall version 1.23 manually: 
Since I have ubuntu only, I used this link: https://download.lenovo.com/pccbbs/mobiles/n37ul10w.zip
The relevant file was N37ET42W.cab, because N37HT27W.cab is the same as in 1.22.
I used this command to force the reinstall: `fwupdmgr --allow-reinstall --force install N37ET42W.cab`
This did not solve the issue.

### 3: I tried to downgrade to version 1.22 manually: 
Now I tried to downgrade by downloading the old version from this link: https://download.lenovo.com/pccbbs/mobiles/n37ul09w.zip
I used this command: `fwupdmgr --allow-older --allow-reinstall --force install N37ET41W.cab`
This also did not solve the issue.
 
### 4: Since the error was with ME, I also tried manually installing that:
First, I tried installing the latest version(https://download.lenovo.com/pccbbs/mobiles/n37rk02w.zip) and then downgrading it to previous version (https://download.lenovo.com/pccbbs/mobiles/n37rk01w.zip). 
I used the command as mentioned in the readme i.e. `fwupdmgr --allow-older --allow-reinstall install <cab_filename.cab>`
But that also did not solve the problem. Currently the ME version in BIOS seems to be the latest one (15.0.41.2142), which seems to indicate that the update has been partially installed. Otherwise the version should have been 15.0.35.1951

### 5: Tried using bootable usb (https://download.lenovo.com/pccbbs/mobiles/n37ur10w.iso):
This did not work because I did not find a way to downgrade using a bootable usb. 

## Output of `fwupdmgr update --show-all`

```

Devices with no available firmware updates: 
 • Integrated Camera
 • SSD 970 EVO Plus 1TB
 • ThinkPad Thunderbolt 4 Dock
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Intel Management Engine
 • Micron MTFDKBA1T0TFH
 • Prometheus
 • Prometheus IOTA Config
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.22 to 0.1.23?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad P15 Gen 2i/ P17 Gen 2i/ P15g Gen 2i System Firmware Version  ║
║ 1.23                                                                         ║
║                                                                              ║
║ Important updates:- Update includes a security fix.                          ║
║                                                                              ║
║ New functions or enhancements:- Supported a new option to enable/disable     ║
║ Fingerprint Single-Touch Authentication in BIOS Setup.- Updated the          ║
║ iagnostics module to version 04.26.000.- Supported 5.15-5.35 GHz band in     ║
║ Indonesia.                                                                   ║
║                                                                              ║
║ Problem fixes:- Fixed an issue where no display on the external monitor via  ║
║ docking station while Intel Management Engine Firmware is updated.           ║
[... 105 lines cut ...]
│ │ 
│ └─  New version:      240.41.2142
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Previous version:   0.1.22
  │   Update State:       Success
  │   Last modified:      2022-10-23 20:55
  │   GUID:               a1392d82-62d5-4e24-863a-0f682993408f
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      0.1.22
        License:          Unknown
        Description:      
        The vendor did not supply any release notes.
      
```

hughsie, 2022-10-24:
> I used this command to force the reinstall: fwupdmgr --allow-reinstall --force

Why did you use `--force` out of interest?

RayyanRiaz, 2022-10-24:
@hughsie First I tried it without `--force` flag, and then with `--force` flag because I thought that if a partial update is applied, there might be some runtime check that is blocking the update.

mrhpearson, 2022-12-21:
Sorry - I keep finding issues that I've missed :(

@kmauleon or @ChiWei-Chen - is this one you can help with? I know we've had a few ME update issues on other platforms but wasn't sure if P17 G2 was one of them.

Mark

kmauleon, 2022-12-22:
I think we have yet again another set of BIOS issue?  this could be the same as #250 #303 #304 in different platforms

kmauleon, 2023-01-06:
can you please try as suggested here >> https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135

RayyanRiaz, 2023-01-06:
Just tried that, although I have tried that already as mentioned in the issue description. Also added the line `options snd-hda-intel model=auto`  as mentioned [here](https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135). Unfortunately, it didn't work.

hughsie, 2023-01-06:
IIRC, the ME controls the protected audio stream -- so it's not unexpected the audio is broken for a ME failed update.

kmauleon, 2023-01-13:
another suggestion for those still has issues >> https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135  please try... thank you very much

RayyanRiaz, 2023-01-27:
**TLDR**: The new updates work

**Description**
After Lenovo Premier Support suggested,  I tried the new version of the firmware(1.24) and ME(240.42.2235). I tried that with `fwupdmgr` first, but it didn't work. Then I made a Windows 10 Go-To USB and installed the `exe` files, which has now fixed these issues. The error is gone and now everything seems to work fine :)

I installed the updates in this order via Windows:
* Firmware (1.24)
* ME(240.42.2235)

Thanks to all the suggestions, and I think this issue can be closed now :)

kmauleon, 2023-01-29:
thank you very much for the updates @RayyanRiaz !

closed 2023-01-29 by kmauleon