fwupd/firmware-lenovo #281: ThinkBook 14s Yoga ITL: MAC pass through not working

a-andre, 2022-10-25:
With MAC pass through disabled the ACPI dump contains `MACA._AUXMAX_#XXXXXXXXXXXX#`

With MAC pass through enabled the ACPI dump contains `MACA._AUXMAC_#XCXAX1X1XCX2#`. As `XCXAX1X1XCX2` is an invalid MAC address, the kernel cannot configure MAC pass through.

The relevant kernel function is [`vendor_mac_passthru_addr_read`](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/net/usb/r8152.c?h=v6.1-rc2#n1601).

Tested with BIOS version FNCN40WW (v2.10). I couldn't test this with Windows.

I know, Linux on this ThinkBook isn't official supported (yet), but I couldn't find another way to report such kind of issue to Lenovo.

hughsie, 2022-10-25:
@mrhpearson do you want this kind of thing here, or should I direct users to report this somewhere else?

mrhpearson, 2022-10-25:
As it's not related to FW updates I would rather it went to the Lenovo linux forums please

@a-andre - please can you post this on the Lenovo Linux forums: https://forums.lenovo.com/linux

I'll look to pick it up there (and we have a fairly active community too who may be able to contribute).

Thanks!

closed 2022-10-25 by mrhpearson