fwupd/firmware-lenovo #282: Thinkpad x13 gen 2 AMD - Unable to charge battery after BIOS update.

Cyanose, 2022-10-26:
Arch Linux 6.0.2 
Recent bios update to version https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1NET.firmware 0.1.21 caused laptop battery to be unable to charge. Connecting the adapter was just preventing laptop from discharging (Laptop had been consuming the energy directly from the charger, but was unable to charge battery). The interesting fact I found was that the `/sys/class/power_supply/AC` directory was missing. also in the journalctl there were entry: `ACPI: \_SB_PCI0.LPC0.EC0_.AC__:  Error reading AC Adapter state: AE_NOT_FOUND`

Bios downgrade to version 0.1.20 fixed the issue

hughsie, 2022-10-26:
@mrhpearson help!

mrhpearson, 2022-10-26:
Thanks for the report. Raised ticket  LO-2092 to the firmware team.

I don't have a X13 AMD G2 to test with directly myself but I do have a T14s G2 AMD which uses the same FW. I'm seeing that ACPI message and the lack of the power_supply/AC directory - but the battery is charging so those might be a red-herring (do you not seen those with 1.20?)

I'll update when I hear back from the FW team
Mark

Cyanose, 2022-10-26:
Hi, thanks for the action sir.
On the 1.20 bios version the ACPI messege regarding adapter I get in log is: `ACPI: AC: AC Adapter [AC] (on-line)` 
And the `power_supply/AC` does exists with the proper content inside. The battery is charging just as it should.
Best regards

hughsie, 2022-10-27:
@mrhpearson can I flip that bit that makes the uploader have to specify R1NET54W?

mrhpearson, 2022-10-27:
How does that work? Means for that platform the next release has to be that?
I don't have a problem with that - unless a problem is found with 54W and they end up with 55W...

hughsie, 2022-10-28:
> How does that work

Ohh, sorry, I mean we make it mandatory to specify the tag e.g. `R1NET54W` for each upload. It's getting a bit confusing to compare version numbers to tags like `R1NET54W` when the latter keep being forgotten on the LVFS... :)

mrhpearson, 2022-10-28:
@Cyanose - yes, but I think 1.24 is going to be the next release. 1.22 got pulled because of issues found during internal testing and I have no idea what happened to 1.23 (I assume the same).
 
@hughsie - Thumbs up from me and if any bombs get thrown just bounce them my way. I never got a reply back to my previous probe about this so silence is taken as acceptance....

Cyanose, 2022-11-05:
@mrhpearson, @hughsie I've recently discovered that despite AC directory being-no existent in the file system, the battery can still be charged. What actually causes the problem with the battery charging is trying to limit the maximum battery treshold in various ways (I tried to limit it via the systemd daemon that wrote to the `/sys/class/power_supply/BAT0/charge_control_start_threshold` and `/sys/class/power_supply/BAT0/charge_control_end_threshold` files as well as using the `tlp`) both ways caused the same problem with charging. 
Additional info: I can force the charging by using `tlp fullcharge` or `tlp chargeonce` with raised privileges

mrhpearson, 2022-11-18:
1.24 has been released which should fix this issue - can we confirm and close if it's good

closed 2022-11-22 by mrhpearson