fwupd/firmware-lenovo #284: thinkpad x1 carbon 7th gen w/ secure boot: boot entry created by fwupd causes security violation

swilli, 2022-10-27:
Setup: Arch Linux with Secure Boot enabled (shim/MOK based)

Steps to reproduce:
1. Open gnome-firmware
2. Click any device's firmware on Reinstall. This will result in the following boot entries on my machine:
```
willist@THINKPAD ~ $ efibootmgr
BootNext: 0000
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0001,0006,0000,0019,001A,001B,001C,001D,001E,001F,0020,0022,0023,0024
Boot0000* Linux-Firmware-Updater	HD(1,GPT,ac6eef3e-1f3d-f74a-8505-30f2b2ebbfa9,0x800,0xfa000)/File(\EFI\arch\shimx64.efi) File(.\fwupdx64.efi)
Boot0001* Windows Boot Manager	HD(1,GPT,ac6eef3e-1f3d-f74a-8505-30f2b2ebbfa9,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d00000065000100000010000000040000007fff0400
Boot0002* GRUB	HD(1,GPT,ac6eef3e-1f3d-f74a-8505-30f2b2ebbfa9,0x800,0xfa000)/File(\EFI\GRUB\shimx64.efi)
Boot0006* MOK Utility	HD(1,GPT,ac6eef3e-1f3d-f74a-8505-30f2b2ebbfa9,0x800,0xfa000)/File(\EFI\mok\shimx64.efi)
...
```
4. Reboot
5. Computer reboots, Lenovo logo pops up
6. A blue screen appears with text "ERROR: Verification failed: 0x10 Security Violation"
7. Press enter, the computer reboots
8. GRUB launches

No firmware gets upgraded.

I also added the hash of `/boot/EFI/arch/fwupdx64.efi` using the MOK utility (the same way I successfully added `/boot/EFI/GRUB/grubx64.efi`) and tried again. Same outcome.

However, if I change the boot entries before step 4. as follows:
```
willist@THINKPAD ~ $ sudo efibootmgr -b 0000 -B
willist@THINKPAD ~ $ sudo efibootmgr -c -d /dev/nvme0n1 -p 1 -L "Linux-Firmware-Updater" -l '\EFI\arch\shimx64.efi'
willist@THINKPAD ~ $ sudo efibootmgr -n 0000
willist@THINKPAD ~ $ efibootmgr
BootNext: 0000
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0001,0006,0000,0019,001A,001B,001C,001D,001E,001F,0020,0022,0023,0024
Boot0000* Linux-Firmware-Updater	HD(1,GPT,ac6eef3e-1f3d-f74a-8505-30f2b2ebbfa9,0x800,0xfa000)/File(\EFI\arch\shimx64.efi)
Boot0001* Windows Boot Manager	HD(1,GPT,ac6eef3e-1f3d-f74a-8505-30f2b2ebbfa9,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d00000065000100000010000000040000007fff0400
Boot0002* GRUB	HD(1,GPT,ac6eef3e-1f3d-f74a-8505-30f2b2ebbfa9,0x800,0xfa000)/File(\EFI\GRUB\shimx64.efi)
Boot0006* MOK Utility	HD(1,GPT,ac6eef3e-1f3d-f74a-8505-30f2b2ebbfa9,0x800,0xfa000)/File(\EFI\mok\shimx64.efi)
...
```
and then reboot, instead of a security violation, fwupd starts and installs the firmware successfully.

Note the content of `/boot/EFI/arch/`:
```
willist@THINKPAD ~ $ ls /boot/EFI/arch/
fw  fwupdx64.efi  fwupdx64.efi.copied.to.grubx64.efi  grubx64.efi  shimx64.efi
```
where `grubx64.efi` is a renamed `fwupdx64.efi` and its hash has been previously added using the MOK utiliy.

mrhpearson, 2022-11-01:
@swilli - thanks for the clear and detailed description. 

@hughsie - does this one make any sense to you? I'm assuming the BIOS is barfing because fwupdx64.efi is not signed with the 3rd party cert but I'm happy to be corrected. I'm not sure how shim knows whether to call grub or fwupd....(I think I used to know...but I can't remember). I know we've ended up recommending disabling secure boot because of the various shim issues in the last year but not sure if this is related or something new.

swilli, 2022-11-05:
@mrhpearson AFAIK shim calls grubx64.efi by default (https://wiki.archlinux.org/title/Unified_Extensible_Firmware_Interface/Secure_Boot#shim) . And since I renamed my fwupdx64.efi in `/boot/EFI/arch/` to grubx64.efi, shim thinks it's calling GRUB when in fact it calls the firmware upgrader.