fwupd/firmware-lenovo #285: Lenovo Powered Travel Hub firmware not up-to-date and still in testing state

a-andre, 2022-10-28:
The [firmware](https://fwupd.org/lvfs/devices/com.lenovo.TravelHubGen2.extra) is in testing state for almost two years. What's missing to move it to stable?

There has also been [a new release in July 2021](https://download.lenovo.com/consumer/options/lenovo_powered_usb-c_travel_hub_fw_utility_101307051.txt) which should fix some monitor issues, maybe the firmware in LVFS can be updated to that version.

P.S. the device name on LVFS seems to be wrong, instead of "Lenovo Powered Travel Hub Gen 2" it should be ["Lenovo Powered Travel Hub"](https://pcsupport.lenovo.com/us/en/accessories/acc500082). There is a ["Lenovo Travel Hub Gen 2"](https://pcsupport.lenovo.com/us/en/accessories/acc500177) device (without "Powered"), but the firmware versions don't match with the versions on LVFS. I also couldn't find a second generation of the powered travel hub.

mrhpearson, 2022-11-01:
Honestly, I have no idea how that got uploaded in the first place as it's not a dock that is covered by the linux program at all. I wasn't aware of it being on LVFS.
I'll go and do some digging to find out if there is or was an exercise I don't know about.
My honest guess is to not expect any further updates (I'm sometimes wrong though)

a-andre, 2022-11-28:
Have you found something?

mrhpearson, 2022-12-01:
Sorry - I forgot to update this ticket, thanks for the nudge. 
There are no plans for adding LVFS support for this hub I'm afraid. It's slightly curious there is a version under testing to be honest.
Mark

closed 2022-12-01 by mrhpearson