fwupd/firmware-lenovo #286: Updating the dbx fails on T14

hughsie, 2022-11-01:
I updated the dbx from 77 to 217 on a Lenovo T14. It fails as post-reboot no dbx device is present in fwupd. Looking closer no EFI variable is set at all. As in /sys/firmware/efi/efivars/dbx-d719b2cb-3d3a-4596-a3bc-dad00e67656f used to exist and now does not. 

SN `FP-1X9RJH` -- which may be PVT or evaluation hardware. If it's because of that then no worries, otherwise, help! Thanks.

mrhpearson, 2022-11-01:
Hi @hughsie - Which gen T14 and Intel or AMD?

hughsie, 2022-11-01:
@mrhpearson AMD; details below:

	BiosVendor: LENOVO
	BiosVersion: R1BET24W(0.24 )
	BiosMajorRelease: 0
	BiosMinorRelease: 24
	FirmwareMajorRelease: 00
	FirmwareMinorRelease: 18
	Manufacturer: LENOVO
	Family: ThinkPad T14 Gen 1
	ProductName: GT4A1SIT19
	ProductSku: LENOVO_MT_GT4A_BU_Think_FM_ThinkPad T14 Gen 1
	EnclosureKind: a
	BaseboardManufacturer: LENOVO
	BaseboardProduct: GT4A1SIT19

mrhpearson, 2022-11-01:
Oh wow - that's an old version of the BIOS :) Ping me offline and we should figure out updating that because there are a ton of fixes for that platform. I'll need to go and review the internal notes as, if I remember correctly, the update sequence for that platform was a little bit tortured.
I've raised internal ticket  LO-2100 to get the FW team insight on what might have happened, and any recommendations on recovering the dbx

hughsie, 2022-11-01:
> Oh wow - that's an old version of the BIOS

Ya, it was PVT IIRC. Thanks for the help!