fwupd/firmware-lenovo #287: Intel Managment Engine update fails on P53

BigBadBassMan, 2022-11-03:
in relation to #140 and #255 
[ME Update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2NRM.firmware) fails to install on reboot. Only shows "Click a key to proceed" and reboots afterwards. Update is reported as not installed.

OS = Fedora 36
fwupd = Fedora package, 1.8.6
SecureBoot = off
BootGuard = on
Boot-Order-Lock = off

`fwupdmgr get-upgrades` shows only this remaining update
`fwupdmgr refresh --force` changes nothing
`fwupdmgr upgrade` downloads, reboots, fails repeatedly.

I'm at a loss, as to how to proceed, or fix this.

chofstede, 2022-11-07:
Exact same issue here

BigBadBassMan, 2022-11-15:
Since the latest ME firmware `192.92.2145` I see a slightly different behavior:

* download and reboot works
* Rebooting shows
  ```
  fwupd-efi version 1.3
  ```
  which gets amended after ~5 minutes with 
  ```
  Press a key to continue
             Reset system
  ```
* reboot to system
* update is not installed, fwupd uploads report, printing a link to (unhelpful) https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot

```
LENOVO 20QQS0LW00
│
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Summary:            UEFI ESRT device
  │   Current version:    192.81.1753
  │   Minimum Version:    192.81.1753
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               84b80435-53b7-4fc0-aa78-d225a59be53f
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  ├─>ThinkPad P53/P73:
  │     New version:      192.92.2145
  │     Remote ID:        lvfs
  │     Release ID:       16235
  │     Summary:          Lenovo ThinkPad P53/P73 Corporate ME Firmware
  │     License:          Proprietary
  │     Size:             12.2 MB
  │     Created:          2022-10-18
  │     Urgency:          High
  │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N2NRG18W
  │     Vendor:           Lenovo
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     • 0 Intel Platform Update
  │     
  │     Version 12.0.92.2145 (LVFS: 192.92.2145)
  │     
  │     Problem Fixes
  │     
  │     • Mitigated the following security vulnerabilities under issues. Please see fixed issues for details.
  │     Issues:           CVE-2022-29515
  │                       CVE-2022-33159
  │                       CVE-2022-29893
  │                       CVE-2022-27497
  │   
  └─>ThinkPad P53/P73:
        New version:      192.90.2072
        Remote ID:        lvfs
        Release ID:       13430
        Summary:          Lenovo ThinkPad P53/P73 Corporate ME Firmware
        License:          Proprietary
        Size:             12.2 MB
        Created:          2022-05-19
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N2NRG17W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        • 0 Intel Platform Update
        
        Version 12.0.90.2072 (LVFS: 192.90.2072)
        
        Problem Fixes
        
        • Intel TA-00613 and Intel TA-00610 IPU 2022.1 CSME.
        • Mitigated security vulnerabilities under issues.
        Issues:           CVE-2022-0004
                          CVE-2021-33159
```

In addition: 
* No Boot-Order-Lock
* No Secure-Boot
* LUKS encryption is active (mandatory, cannot deactivate, used to work for updates in the past.)

**fwupd version information**
```
runtime   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.hughsie.libjcat           0.1.12
compile   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    6.0.8-200.fc36.x86_64
compile   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.gusb          0.3.10
```

Behavior is now similar to #289, but without the error message.

kmauleon, 2022-11-16:
hi all... can you please confirm MEFW version as shown in BIOS Setup Menu?  we just want to rule out this is not BIOS ESRT related issue.  Thank you very much.

BigBadBassMan, 2022-11-21:
Strange, BIOS shows the offending ME update as installed:
[image]

kmauleon, 2022-11-23:
thank you very much for the update @BigBadBassMan !