fwupd/firmware-lenovo #289: After reboot for an update error is displayed: ERROR 105 - command line error

scrool, 2022-11-10:


hughsie, 2022-11-10:
@mrhpearson is this perhaps an older firmware that's not supported? Thanks.

mutax, 2022-12-10:
Today I got the same error on a T470s, also with fwupd-efi version 1.3.

I think this might be a regression - after installing an experimental firmware that fixed the issue last time, I was also able to update normally using fwupdmgr to the regular one.

For reference see: https://github.com/fwupd/firmware-lenovo/issues/201#issuecomment-1190299156

Florian

mrhpearson, 2022-12-20:
Oh wow - I'd missed this thread completely.
Not sure if it is old FW related or not - but my understanding was the shell flash utility shouldn't be used so there is something odd going on.

@mutax - could you confirm which BIOS, EC & ME you currently have installed please (easiest is from the BIOS setup screen - F1 during early boot)

@ChiWei-Chen - would you mind checking this one with the FW team please 

Thanks

ChiWei-Chen, 2022-12-21:
Thanks, Mark, I created LO-2205 for internal tracking.

morngrar, 2023-01-09:
I experienced the very same issue today while attempting to update the firmware on my Thinkpad P71, my setup is basically the same as OP, although I don't know the NVMe part number off the top of my head. Running arch linux and installed `fwupd` just now. Sent error reports using the build-in reporting tool. Watching this issue for any updates, and can supply additional system info if desired.

Tojaj, 2023-01-13:
I see the exact same issue as @scrool for few months now on my ThinkPad T470s with Fedora 37.

I can provide all the logs that @scrool provided if it would help you.

Pasting output of the `$ sudo fwupdmgr get-updates` after the failed update to make clear what updates are in question (Btw. Note I didn't get the nice `Devices that were not updated correctly:` message as @scrool):

    Devices with no available firmware updates: 
     • UEFI Device Firmware
     • ThinkPad Dock
     • MZVLW256HEHP-000L7
     • Unifying Receiver
    Devices with the latest available firmware version:
     • Embedded Controller
    Idle…                    [***************************************]
    Update failure is a known issue, visit this URL for more information: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
    Successfully uploaded 1 report
    LENOVO 20HGS22D0W
    │
    ├─Intel Management Engine:
    │ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
    │ │   Summary:            UEFI ESRT device
    │ │   Current version:    184.86.3909
    │ │   Minimum Version:    184.86.3909
    │ │   Vendor:             Lenovo (DMI:LENOVO)
    │ │   Update State:       Success
    │ │   GUID:               e9124c4a-fdff-42e5-b2ad-f745db345953
    │ │   Device Flags:       • Internal device
    │ │                       • Updatable
    │ │                       • System requires external power source
    │ │                       • Supported on remote server
    │ │                       • Needs a reboot after installation
    │ │                       • Device is usable for the duration of the update
    │ │ 
    │ ├─ThinkPad T470s:
    │ │     New version:      184.93.4323
    │ │     Remote ID:        lvfs
    │ │     Release ID:       15825
    │ │     Summary:          Lenovo ThinkPad T470s Corporate ME Firmware
    │ │     License:          Proprietary
    │ │     Size:             7,5 MB
    │ │     Created:          2022-09-29
    │ │     Urgency:          High
    │ │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N1WRM21W
    │ │     Vendor:           Lenovo
    │ │     Release Flags:    • Is upgrade
    │ │     Description:      
    │ │     This stable release fixes the following issues:
    │ │     
    │ │     • Updated includes a security fix
    │ │     Issues:           CVE-2022-26845
    │ │                       CVE-2022-27497
    │ │                       CVE-2022-29893
    │ │                       CVE-2021-33159
    │ │                       CVE-2022-29466
    │ │                       CVE-2022-29515
    │ │   
    │ ├─ThinkPad T470s:
    │ │     New version:      184.92.4222
    │ │     Remote ID:        lvfs
    │ │     Release ID:       12873
    │ │     Summary:          Lenovo ThinkPad T470s Corporate ME Firmware
    │ │     License:          Proprietary
    │ │     Size:             7,5 MB
    │ │     Created:          2016-07-08
    │ │     Urgency:          High
    │ │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N1WRM20W
    │ │     Vendor:           Lenovo
    │ │     Release Flags:    • Is upgrade
    │ │     Description:      
    │ │     This stable release fixes the following issues:
[... 115 lines cut ...]
      │     • Updated the Diagnostics module to version 04.24.000.
      │     • Updated the CPU microcode.
      │     • Update MEBx.
      │     Issues:           CVE-2022-0005
      │                       CVE-2022-21151
      │   
      └─ThinkPad T470s:
            New version:      0.1.45
            Remote ID:        lvfs
            Release ID:       12595
            Summary:          Lenovo ThinkPad T470s System Firmware
            License:          Proprietary
            Size:             9,4 MB
            Created:          2016-07-08
            Urgency:          Critical
            Vendor:           Lenovo
            Release Flags:    • Is upgrade
            Description:      
            This stable release fixes the following issues:
            
            • Update includes a security fix.
            • Updated the CPU microcode.

One thing that may be relevant for the debugging is that the system I have was initially installed on *28th April 2021* (thus likely it was Fedora 34) and since then I do `dnf system-upgrade`.
Mentioning this in case that it may be relevant for how updates are done. I know that for example Grub although RPMs are updated during system-upgrade, the installed or embedded bootloader is never updated automatically.. So I wonder if there may be a similar case for fwupd (?)

ChiWei-Chen, 2023-01-16:
Hi,
The FW team replied the issue was resolved in 1.47 or later version of BIOS. Could you try to update the BIOS from the link below? Please let me know if this doesn't work for you. Thanks!
https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t470s/downloads/ds120418

morngrar, 2023-01-16:
> Hi, The FW team replied the issue was resolved in 1.47 or later version of BIOS. Could you try to update the BIOS from the link below? Please let me know if this doesn't work for you. Thanks! [...]

I have this same issue on a P71. Is it resolved for this machine as well?

ChiWei-Chen, 2023-01-17:
@morngrar , could you post the output of the `$ sudo fwupdmgr get-updates`, we need P71 FW team to check their FW. Thank you.

morngrar, 2023-01-17:
Here you go, @ChiWei-Chen:

```
Devices with no available firmware updates:
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • MZVLB512HAJQ-000L7
 • UEFI dbx
LENOVO 20HKCTO1WW
│
├─Intel Management Engine:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Summary:            UEFI ESRT device
│ │   Current version:    184.86.3877
│ │   Minimum Version:    184.86.3877
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               604799ec-12a5-4b22-817a-358135296a78
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │
│ ├─>ThinkPad P71:
│ │     New version:      184.93.4323
│ │     Remote ID:        lvfs
│ │     Release ID:       16285
│ │     Summary:          Lenovo ThinkPad P71 Corporate ME Firmware
│ │     License:          Proprietary
│ │     Size:             7,5 MB
│ │     Created:          2022-10-19
│ │     Urgency:          High
│ │     Details:          https://pcsupport.lenovo.com/de/en/search?query=N1URG35W
│ │     Vendor:           Lenovo
│ │     Release Flags:    • Is upgrade
│ │     Description:
│ │     • 8 Intel Platform Update
│ │
│ │     Version 11.8.93.4323 (LVFS: 184.93.4323)
│ │
│ │     Problem Fixes
│ │
│ │     • Mitigated the following security vulnerabilities under issues. Please see fixed issues for details.
│ │     Issues:           CVE-2022-29515
│ │                       CVE-2022-33159
│ │                       CVE-2022-29893
│ │                       CVE-2022-27497
│ │
│ └─>ThinkPad P71:
│       New version:      184.90.3987
│       Remote ID:        lvfs
│       Release ID:       11714
│       Summary:          Lenovo ThinkPad P71 Corporate ME Firmware
│       License:          Proprietary
│       Size:             7,5 MB
│       Created:          2022-01-21
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N1URG33W
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:
│       • 8 Intel Platform Update
│
│       Version 11.8.90.3987 (LVFS: 184.90.3987)
│
│       Problem Fixes
│
│       • Intel CSME TA-00473 IPU 2021.2:
[... 110 lines cut ...]
  │
  │     New functions or enhancements:
  │
  │     • Updated the CPU microcode.
  │     Issues:           INTEL-TA-00527
  │                       INTEL-TA-00532
  │
  └─ThinkPad P71:
        New version:      0.1.35
        Remote ID:        lvfs
        Release ID:       10205
        Summary:          Lenovo ThinkPad P71 System Firmware
        License:          Proprietary
        Size:             9,4 MB
        Created:          2021-08-11
        Urgency:          Critical
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:
        Lenovo ThinkPad P71 System Firmware Version 1.35

        Important updates:

        • Update includes a security fix.
```

ChiWei-Chen, 2023-01-18:
@morngrar , thanks, I've shared this information to the FW team, I'll keep you post if any progress available.

ChiWei-Chen, 2023-01-18:
@morngrar FW team confirmed the next official P71 BIOS will fix this issue. They also give a trial BIOS. If you are willing to give it a try, please feel free to drop me a mail ( cchen50 [at] lenovo [dot] com ) , thanks!

Tojaj, 2023-01-19:
> Hi, The FW team replied the issue was resolved in 1.47 or later version of BIOS. Could you try to update the BIOS from the link below? Please let me know if this doesn't work for you. Thanks! [...]

Hi @ChiWei-Chen, I have downloaded the `BIOS Update Utility (Linux)` (the `n1wul41w.zip` file), extracted it and wanted to follow the README instructions for manual install [1]:

	Manual Update Using fwupdmgr

	  1. Make sure the AC adapter is firmly connected to the target computer.
	  2. Launch Terminal.
	  3. Move to the directory where the cabinet file was placed.
	  4. Run fwupdmgr to schedule firmware update.

		# fwupdmgr install xxx.cab

	  5. Restart the system.
	     You may be able to use "reboot" command.
	     If it is not available, select "Reboot" from Power icon on the desktop.
	  6. The computer will be restarted and the EFI BIOS will be updated.

However, the ZIP archive (`n1wul41w.zip`) contains *two* `.cab` files and README doesn't comment on such situation:

    $ ls -lh
    total 5,3M
    -rw-r--r--. 1 tojaj tojaj 5,0M 26. zář 09.50 N1WET69W.cab
    -rw-r--r--. 1 tojaj tojaj 348K  8. srp 10.16 N1WHT40W.cab

So I wonder which one to install? If I need to install both (which I assume), then I wonder if the order of the installation is important or not?

[1] https://download.lenovo.com/pccbbs/mobiles/n1wul41w.txt

scrool, 2023-01-19:
> However, the ZIP archive (`n1wul41w.zip`) contains _two_ `.cab` files and README doesn't comment on such situation:
>
> ```
> $ ls -lh
> total 5,3M
> -rw-r--r--. 1 tojaj tojaj 5,0M 26. zář 09.50 N1WET69W.cab [...]

Those codes are listed on https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t470s/downloads/ds120418-bios-update-utility-bootable-cd-for-windows-10-64-bit-81-64-bit-7-32-bit-64-bit-linux-thinkpad-t470s

Package: 1.48
UEFI BIOS: 1.48 (N1WET69W)
ECP: 1.22 (N1WHT40W)

so if the goal is to update BIOS here I think you might be interested in first one?

But for authoritative answer let's wait for @ChiWei-Chen

Tojaj, 2023-01-19:
@scrool good point

@ChiWei-Chen I tried to install the `N1WET69W.cab` as @scrool suggested but got an error:

    $ sudo fwupdmgr install N1WET69W.cab
    Decompressing…           [  -                                    ]
    firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer

Is it a bug on the Lenovo side (missing signature)? Or in the `fwupdmgr` that is missing Lenovo key (signature not trusted)?

Although checksum of the zip archive matched what is on the Lenovo page, I'm reluctant to install an unsigned (wrongly signed) firmware.

ChiWei-Chen, 2023-01-20:
@scrool thanks! It's correct.

@Tojaj could you try the instruction as the comment in https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1203769040 ? Thanks!

Tojaj, 2023-01-20:
@ChiWei-Chen the instructions from the linked comment suggest setting `OnlyTrusted=false` which is what I mentioned I'm reluctant to do.

The `fwupdmgr` error message hints about this possibility as well, but it also warns to do that `ONLY if you are a firmware developer`.

Is there a chance that Lenovo will sign the firmware in the `cab` file?

ChiWei-Chen, 2023-01-23:
@Tojaj , thanks for the update, please let us know if there is any further question.

BTW, could you let me know the output of `fwupdmgr --version` ?
I'll check the sign problem with firmware team once they return from holidays.

scrool, 2023-01-23:
Now, when Tomas managed to update his system manually, would it be possible to get that bios uploaded to LVFS so I could use automatic approach with `sudo fwupdmgr update`?

ChiWei-Chen, 2023-01-23:
@scrool , it seems the latest version 1.48 was already uploaded after this issue was posted. 
 https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1WET.firmware

Could you give it a try and let us know how it works? Thanks!

Tojaj, 2023-01-23:
@ChiWei-Chen here you go:

    [tojaj@thinkpad ~]$ fwupdmgr --version
    compile   org.freedesktop.fwupd         1.8.9
    compile   com.hughsie.libjcat           0.1.12
    runtime   org.freedesktop.fwupd-efi     1.3
    compile   org.freedesktop.gusb          0.4.3
    runtime   com.hughsie.libjcat           0.1.12
    runtime   com.dell.libsmbios            2.4
    runtime   org.freedesktop.gusb          0.4.3
    runtime   org.freedesktop.fwupd         1.8.9
    runtime   org.kernel                    6.1.6-200.fc37.x86_64

scrool, 2023-01-23:
I managed to update my system as well, although it was with a gotcha, slight hiccup.

First, just a reminder that I got two updates pending - "Intel Management Engine" and "System Firmware". I run `sudo fwupdmgr update` and I was presented with both updated:

```
Devices with no available firmware updates: 
 • MZVLW256HEHP-000L7
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.83.3874 to 184.93.4323?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Updated includes a security fix                                            ║
║                                                                              ║
║ 20HMS1FC02 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Waiting…                 [***************************************]
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.41 to 0.1.47?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Update includes a security fix.                                            ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Updated the Diagnostics module to version 04.27.000.                       ║
║                                                                              ║
║ 20HMS1FC02 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
 • UEFI dbx

An update requires a reboot to complete. Restart now? [y|N]: 
```

But after a reboot I got the very same error message as in the original report.

----

I was not able to answer `N` for the first update because that canceled the operation all together. 

----

Snippet of the help:

[... 100 lines cut ...]

Devices that have been updated successfully:

 • System Firmware (0.1.41 → 0.1.47)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]:
y
```

but nevertheless `Intel Management Engine` seem to be running latest version:

```
fwupdmgr get-devices | grep -A 3 'Intel Management Engine'     
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.93.4323
```

---

Since this resolved my issue I'm going to close this.

Thanks :pray: @ChiWei-Chen for help, @Tojaj for testing this the hard (manual) way.

closed 2023-01-23 by scrool

maakuth, 2025-11-20:
This still seems to happen with T470s update from 0.1.45->0.1.55. For me too, limiting the update to system firmware only also seems to have updated the ME. I don't know if the ME update could be masked somehow in LVFS metadata so that this solution would happen automatically. Finding this issue through search was useful, thank you.