fwupd/firmware-lenovo #291: T470s: Intel Management Engine 11.8 Firmware version 11.8.93.4323 is missing

dkadioglu, 2022-11-15:
Since 14th of November an updated version 11.8.93.4323 is available for the T470s Intel Management Engine. The update file for Windows has version 11.8.93.4323-N1QRG31W. May this be added to fwupd.org? Thanks in advance.

kmauleon, 2022-11-15:
hi @dkadioglu will ping owner, this should be under testing... will update of target release... thank you very much

kmauleon, 2022-11-16:
As per owner, it will be available by 11/30... thank you very much.

dkadioglu, 2022-11-16:
@kmauleon thank you very much for clarifying and the update on this! Could I ask you to do the same for another missing firmware update, as there was no reply so far: https://github.com/fwupd/firmware-lenovo/issues/279 That would be great! Thank you very much!

dkadioglu, 2022-11-19:
It has already been published, great! Thanks to @kmauleon and whomever at Lenovo is involved!
The update has been applied successfully!

closed 2022-11-19 by dkadioglu