fwupd/firmware-lenovo #292: Intel Management Engine (CSME) version is invalid

gdonval, 2022-11-25:
I opened a bug report for `fwupd` proper and was told to report here instead.

https://github.com/fwupd/fwupd/issues/5277#event-7875194609

This is on a Lenovo X1 Carbon Gen 7.

- **Current** Intel Management Engine version: 192.49.1534
- **Latest** Intel Management Engine version: 192.47.1524

I am in a weird situation:

- `fwupd` says the current 192.49.1534 version is invalid;
- **downgrading** to 192.47.1524 fails (it reboots to the bios, the firmware management engine is loading, the little spinning thingy spins but after a while I just get "update failed" written on the top-left).

It feels either a new version is in order or, if 192.49.1534 is valid, that it should be marked valid.

**All** `fwupd` info are provided in the original bug report: https://github.com/fwupd/fwupd/issues/5277

hughsie, 2022-11-25:
@mrhpearson I think the Lenovo X1 Carbon Gen 7 is missing an CSME update.

kmauleon, 2022-11-28:
hi @gdonval may i know result of "fwupdmgr get-devices"   thank you very much

gdonval, 2022-11-30:
@kmauleon: it looks like the version on LVFS is more recent than what I have yet it is not available to me through `fwupd` update.

Ultimately, what's the difference between their Consumer and Corporate version (I suppose it is not just marketing)?

gdonval, 2022-11-30:
Went ahead and downloaded both firmware to analyse them with `fwupdmgr get-details`. They both threw an `Update Error`

* **Corporate** variant: ```GUID 4bbc40fa-f81e-4206-bc70-a1f7b744d964 was not found```
* **Consumer** variant: ```Device was not found```

kmauleon, 2022-11-30:
hi @gdonval  you have indeed very old MEFW version.... even older than the first LVFS MEFW release T_T 
Corporate is for systems with AMT.  The update error is looking for W-BIOS GUID. Thus, it will not install on your system.
Let me confirm first with BIOS Dev for P-BIOS case.  Will feedback later.  Thank you very much.

kmauleon, 2022-12-01:
hi @gdonval can you please try to update again by executing as below
     fwupdmgr refresh --force
     fwupdmgr update

gdonval, 2022-12-01:
This did not work. 

I will retry tomorrow in case things were not totally in sync server-wise.

kmauleon, 2022-12-01:
yes please... sometimes it takes time.... thank you very much

gdonval, 2022-12-01:
@kmauleon, it worked. A new release was made available and the installation went well!

Thanks a lot for looking into this!

closed 2022-12-01 by gdonval

kmauleon, 2022-12-02:
great!  thank you very much @gdonval !