fwupd/firmware-lenovo #294: X13 Gen 3 AMD: TPM PCR0 reconstruction Invalid

dgsiegel, 2022-11-29:
**Describe the bug**
fwupd reports that the TPM PCR0 reconstruction is invalid.

**Steps to Reproduce**
`fwupdmgr security`

If I run `fwupdtool security --force -v` I can find the following output:

`FuPluginTpm          comparing TPM 6a9848d0a7fb307084845a69ea3b0825d6e2430bf2879d23cb97b4eba7c72f13 and EVT 6a9848d0a7fb307084845a69ea3b0825d6e2430bf2879d23cb97b4eba7c72f13
`

**Expected behavior**
No such warning, because this is a brand-new system with a fresh installation of Fedora 37.

**fwupd version information**
```
compile   org.freedesktop.fwupd         1.8.7
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.1
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.2
runtime   org.kernel                    6.0.9-300.fc37.x86_64
runtime   org.freedesktop.fwupd         1.8.7
```

Installed via `dnf`.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 21CMCTO1WW
│
├─Unknown Device:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│   
├─Fingerprint Sensor:
│     Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│     Summary:            Match-On-Chip fingerprint sensor
│     Current version:    01000300
│     Vendor:             Goodix (USB:0x27C6)
│     Install Duration:   10 seconds
│     Serial Number:      UID3D39D704_XXXX_MOC_B0
│     GUIDs:              14450e82-8667-54c3-98e1-098d8c6dc3b8 ← USB\VID_27C6&PID_6594
│                         6322b4f7-0571-5f07-8538-e041bbc70677 ← USB\VID_27C6&PID_6594&REV_0100
│     Device Flags:       • Updatable
│                         • Device can recover flash failures
│                         • Signed Payload
│   
├─Integrated Camera:
│     Device ID:          91c9a317b30acf7c0cd967f2ce6925d518ef19a7
│     Current version:    10.22
│     Vendor:              (USB:0x174F)
│     Serial Number:      0001
│     GUIDs:              0cf3aef8-c29e-5cb8-997c-d498ed52b666 ← USB\VID_174F&PID_1812
│                         da6609bb-ca33-5ab4-aba3-a28b193c1d9d ← USB\VID_174F&PID_1812&REV_1022
│     Device Flags:       • Updatable
│   
├─MZVL21T0HCLR-00BL7:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    CL2QGXA7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S64PNX0T909746
[... 157 lines cut ...]
│     Minimum Version:    66048
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               9b7ef120-9e99-404e-8cad-0aeca4d09ee6
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─USB4 host controller:
      Device ID:          fba88f7dbbd16af1e2d4f10000f897b67774f5d8
      Summary:            Unmatched performance for high-speed I/O
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
      Device Flags:       • Internal device
                          • System requires external power source
```

</details>

**Additional questions**

- Operating system and version: Fedora 37, Kernel 6.0.9-300.fc37.x86_64
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2022-11-29:
Maybe a big ask, but could you compile fwupd and then run `sudo ./plugins/tpm/fwupdtpmevlog --pcr 0`? @superm1 maybe we should install that binary into /usr/libexec/fwupd for this kind of debugging?

dgsiegel, 2022-11-29:
> Maybe a big ask, but could you compile fwupd and then run `sudo ./plugins/tpm/fwupdtpmevlog --pcr 0`?

No worries! Here's the output:

```
$ sudo build/plugins/tpm/fwupdtpmevlog --pcr 0
PCR:                    BIOS (0)
Type:                   0x8
Description:            EV_S_CRTM_VERSION
ChecksumSha256:         265ffa8ec06a229d4e03172d6630916179c3625744f56890847331b4b7cde7d4
BlobStr:                UgAyADIARQBUADUANQBXACAAAAA=

PCR:                    BIOS (0)
Type:                   0x80000008
Description:            EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha256:         0d0d9517da007bc0450fdef51e98a8448c6d1ba5f7ae694d7139508c787f2879
BlobStr:                AADACQAAAAAAABoAAAAAAA==

PCR:                    BIOS (0)
Type:                   0x1
Description:            EV_POST_CODE
ChecksumSha256:         7faa18ccd6074fff7900f48242d2ff322343d07354633c519850674d8f003d80
BlobStr:                QUNQSSBEQVRB

PCR:                    BIOS (0)
Type:                   0x1
Description:            EV_POST_CODE
ChecksumSha256:         608072d8953921f15718897cbb0a47623e0c29fa7286d20beb0d733756acb643
BlobStr:                QUNQSSBEQVRB

PCR:                    BIOS (0)
Type:                   0x4
Description:            EV_SEPARATOR
ChecksumSha256:         df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119
BlobStr:                AAAAAA==

Reconstructed PCRs:
  PCR 0:                SHA256(6a9848d0a7fb307084845a69ea3b0825d6e2430bf2879d23cb97b4eba7c72f13)

```

superm1, 2022-11-29:
> @superm1 maybe we should install that binary into /usr/libexec/fwupd for this kind of debugging?

I don't think generally that makes sense. The bugs in our reconstruction comparison are so sparse now, it's usually firmware bugs now isn't it?

I guess let's see what the outcome of this particular issue is to decide.

mrhpearson, 2022-12-01:
I'll flag this to the FW team - but as a note it is likely to be a slow burner. I've got a similar issue flagged on the X1Carbon9 and trying to review with the TPM team but it's slow going (but not dead!)

superm1, 2022-12-18:
This may be a duplicate of https://github.com/fwupd/fwupd/issues/5344#event-8063641475

Can you please try a snapshot of main?

If you need an rpm I'll volunteer @hughsie to build you one 😜.

dgsiegel, 2022-12-19:
@superm1 no worries, I can build the project on my own :) 

Although the `main` branch (https://github.com/fwupd/fwupd/tree/main) still produces the same result:

```
$ sudo build/plugins/tpm/fwupdtpmevlog --pcr 0
PCR:                    BIOS (0)
Type:                   0x8
Description:            EV_S_CRTM_VERSION
ChecksumSha256:         265ffa8ec06a229d4e03172d6630916179c3625744f56890847331b4b7cde7d4
BlobStr:                UgAyADIARQBUADUANQBXACAAAAA=

PCR:                    BIOS (0)
Type:                   0x80000008
Description:            EV_EFI_PLATFORM_FIRMWARE_BLOB
ChecksumSha256:         0d0d9517da007bc0450fdef51e98a8448c6d1ba5f7ae694d7139508c787f2879
BlobStr:                AADACQAAAAAAABoAAAAAAA==

PCR:                    BIOS (0)
Type:                   0x1
Description:            EV_POST_CODE
ChecksumSha256:         7faa18ccd6074fff7900f48242d2ff322343d07354633c519850674d8f003d80
BlobStr:                QUNQSSBEQVRB

PCR:                    BIOS (0)
Type:                   0x1
Description:            EV_POST_CODE
ChecksumSha256:         608072d8953921f15718897cbb0a47623e0c29fa7286d20beb0d733756acb643
BlobStr:                QUNQSSBEQVRB

PCR:                    BIOS (0)
Type:                   0x4
Description:            EV_SEPARATOR
ChecksumSha256:         df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119
BlobStr:                AAAAAA==

Reconstructed PCRs:
  PCR 0:                SHA256(6a9848d0a7fb307084845a69ea3b0825d6e2430bf2879d23cb97b4eba7c72f13)

```

superm1, 2022-12-19:
Does the security command still fail though?

dgsiegel, 2022-12-19:
> Does the security command still fail though?

That I couldn't test as I would have to install the daemon as well, otherwise I'll get a version mismatch.

superm1, 2022-12-19:
That's the part that should work now if you can please try.

dgsiegel, 2022-12-19:
This might be due to the way I installed it locally, but it doesn't seem to be able to find TPM at all:

```
Host Security ID: HSI:0 (v1.8.9)

HSI-1
✔ Supported CPU:                 Valid
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ TPM v2.0:                      Not found

HSI-2
✔ BIOS rollback protection:      Enabled
✘ IOMMU:                         Not found

HSI-3
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted
```

This is the output of the installed fwupd (1.8.8):

```
Host Security ID: HSI:1 (v1.8.8)

HSI-1
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ SPI write protection:          Enabled
✘ TPM PCR0 reconstruction:       Invalid

HSI-3
✔ Pre-boot DMA protection:       Enabled
✔ SPI replay protection:         Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Encrypted RAM:                 Encrypted
✔ Processor rollback protection: Enabled

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted
```

superm1, 2022-12-19:
Could you please share a verbose log from `sudo fwupdtool security --verbose 2>&1"

dgsiegel, 2022-12-19:
Here you go: [fwupdtool.log](https://github.com/fwupd/firmware-lenovo/files/10260266/fwupdtool.log)

I'm quite sure an RPM would make things way easier at this point :)

superm1, 2022-12-19:
> 14:52:04.215 FuConfig             trying to load config values from /tmp/fwupd/build/var/etc/fwupd/daemon.conf
> 14:52:04.215 FuConfig             failed to read UpdateMotd key: Key file does not have group “fwupd”

This is probably the root cause of some weird abnormalities.  Did this conffile not get populated?  Or is it in the wrong place?
@hughsie I think in general we might want to consider making sure that everything in the daemon/fwupdtool works the same with default conffile and no conffile.  That's a bit unexpected to me that the test plugin for example is enabled by default..

>   BestDevice:           /dev/urandom

The test plugin is enabled, it shouldn't be.

> I'm quite sure an RPM would make things way easier at this point :)

@hughsie can you help with that?

hughsie, 2022-12-19:
Here is the git main branch from today: https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/

dgsiegel, 2022-12-19:
Thanks @hughsie!

@superm1 you were right, the git main branch now lists everything as valid:

```
$ sudo fwupdmgr security
Host Security ID: HSI:4 (v1.8.9)

HSI-1
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ BIOS rollback protection:      Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ SPI write protection:          Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Pre-boot DMA protection:       Enabled
✔ SPI replay protection:         Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Encrypted RAM:                 Encrypted
✔ Processor rollback protection: Enabled

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted
```

Thanks!

superm1, 2022-12-19:
Great thanks!

@hughsie I don't know about you but this is the first time I've seen a full "4" in the wild 😜

closed 2022-12-19 by superm1

hughsie, 2022-12-20:
> this is the first time I've seen a full "4" in the wild

I saw one the other day which exposed a bug in the GNOME panel that was only ever tested up to HSI 3 :)