fwupd/firmware-lenovo #295: Lenovo T14 Gen2i - No sound nor input after LVFS ME update (Ubuntu 20.04)

ruimbribeiro, 2022-12-04:
After updating to the latest ME version, I got several problems on my laptop.

- First a screen started to appear on restart: `Recovering ME update ... Failed to recover (this was failing around 60%).`
- I got a DummyOutput in the Sound settings
- No input is detected as well.

After doing `fwupdmgr reinstall` I have no more the "Recovering ME" screen.

Still I have no sound nor input.

I find it odd that the version that LVFS installed is not supported for my Lenovo laptop 

As per Lenovo Website.
https://support.lenovo.com/in/en/downloads/ds558339-intel-management-engine-150-firmware-consumer-update-utility-thinkpad

```
Support models      
ThinkPad P15  Gen 2i  (Machine types: 20YQ, 20YR)
ThinkPad P15v Gen 2i  (Machine types: 21A9, 21AA)
ThinkPad T15  Gen 2i  (Machine types: 20W4, 20W5)
ThinkPad T15p Gen 2i  (Machine types: 21A7, 21A8)
```

**System info**

```
fwupdmgr get-history
20W0S0RW00
│
├─SKHynix HFS512GDE9X081N:
│ │   Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│ │   Previous version:   41720C20
│ │   GUID:               34f9d6fa-e1d6-5d7d-909e-0406644298c5
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─PC711 512GB:
│       New version:      41730C20
│       Remote ID:        lvfs
│       Release ID:       12987
│       Summary:          SK hynix PC711 512GB NVMe SSD Firmware for Lenovo PC
│       License:          Proprietary
│       Size:             1,4 MB
│       Created:          2022-04-20
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       Firmware update may fail when HDD password is set in BIOS.In that case, remove HDD password then try update again.
│       
│       Firmware and device description: SKHynix_HFS512GDE9X081N, Firmware Version: 41730C20
│       
│       Supported device description: SKHynix_HFS512GDE9X081N, Firmware Version: 41710C20, 41720C20
│       
│       Supported Product Scope - Lenovo ThinkPad, ThinkCentre, ThinkStation, IdeaCentre
│     
├─Embedded Controller:
│ │   Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│ │   Previous version:   0.1.41
│ │   Update State:       Success
│ │   GUID:               2a5d161c-0ba2-4136-94b3-c027f0a7e5fc
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2:
│       New version:      0.1.42
│       Remote ID:        lvfs
│       Release ID:       14474
│       Summary:          Lenovo ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2 Embedded Controller Firmware
│       License:          Proprietary
│       Size:             767,3 kB
│       Created:          2022-06-15
│       Urgency:          Medium
│       Vendor:           Lenovo
│       Description:      
│       Embedded Controller Firmware N34HT42W version 1.42
│     
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Previous version:   0.1.52
│ │   Update State:       Success
│ │   Last modified:      2022-12-04 16:50
│ │   GUID:               973a10fa-79c0-453e-b78c-27b55cc0e6eb
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2:
│       New version:      0.1.52
│       Remote ID:        lvfs
│       Release ID:       14062
│       Summary:          Lenovo ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2 System Firmware
│       License:          Proprietary
│       Size:             18,3 MB
│       Created:          2021-06-15
│       Urgency:          Critical
│       Vendor:           Lenovo
│       Description:      
│       This stable release fixes the following issues:
│       
│       • Updated includes a security fix.
│     
└─Intel Management Engine:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   240.42.2235
  │   Update State:       Success
  │   Last modified:      2022-12-04 20:57
  │   GUID:               5d19c74d-6a93-489a-a5ed-65cb7f1e13bf
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2:
        New version:      240.42.2235
        Remote ID:        lvfs
        Release ID:       15678
        Summary:          Lenovo ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2 Consumer ME Firmware
        License:          Proprietary
        Size:             3,8 MB
        Created:          2021-11-17
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N34RN17W
        Vendor:           Lenovo
        Description:      
        This stable release fixes the following issues:
        
        • Updated includes a security fix.

```

```
fwupdmgr update --show-all
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • SKHynix HFS512GDE9X081N
 • System Firmware
 • UEFI dbx
```

mrhpearson, 2022-12-20:
Sorry - I'd missed this thread. Is this still an issue or did it get resolved? Were you able to reinstall the ME firmware?

Tagging @kmauleon in case she has any suggestions

kmauleon, 2022-12-21:
will try to ping the owner and feedback thank you very much

ruimbribeiro, 2022-12-22:
Hey @mrhpearson 

```
fwupdmgr reinstall
```

After running reinstall and after a reboot the system took a while but seemed to recover and now I don't have the recover message on boot anymore.

Although I did have to deal with both missing input and output devices. 

Changing the file `/etc/modprobe.d/alsa-base.conf`

And adding: `options snd-hda-intel model=auto` fixed it.

adityamukho, 2023-01-12:
@mrhpearson still fails on my ThinkPad P15V Gen 2i (Corporate) after `fwupdmgr reinstall` - meaning the firmware update fails.

adityamukho, 2023-01-12:
I have written in more detail about this here: https://github.com/fwupd/firmware-lenovo/issues/250

mrhpearson, 2023-01-12:
Ack - I see Chi-Wei and Kathreene have commented on that thread and they're best placed to help so let's see what they come back with

kmauleon, 2023-01-13:
@mrhpearson should we close this particular ticket? thanks to @ruimbribeiro fixed it by fwupdmgr reinstall with additional steps.... and track the other platforms separately

kmauleon, 2023-01-17:
will close this one as fixed... let me know if need to reopen... thank you very much.

closed 2023-01-17 by kmauleon

kmauleon, 2025-11-24:
@PauloMorgs let's discuss in issue #556 for proper tracking... thank you