fwupd/firmware-lenovo #296: Lenovo T590 Upgrade Intel Management Engine from 192.90.2072 to 192.92.2145 failing

ehsavoie, 2022-12-05:
Upgrade Intel Management Engine from 192.90.2072 to 192.92.2145 I'm on Fedora 37 Plasma.
On reboot the upgrade fails, something about a wrong capsule, with fwupdmgr or Discover (plasma GUI for distribution updates).

kmauleon, 2022-12-07:
hi @ehsavoie already pinged MEFW owner... will update once I got feedback from them... thank you very much

kmauleon, 2022-12-12:
hi @hughsie could this be a fwupd issue? owner's findings as below.  Please advise.  Thank you very much.
Fedora 35	1.6.4	     PASS
Fedora 37	1.8.7	     Failed

hughsie, 2022-12-12:
> Could not query capsule capabilities: Unsupported

This means that fwupd-efi did `RT->QueryCapsuleCapabilities` and that returned `EFI_UNSUPPORTED`. In that case we log a warning, and then assume `EfiResetWarm` _and then_ call `RT->UpdateCapsule`.

@ehsavoie can you confirm the version you have installed of fwupd-efi please. Do you also see a debug line of `Capsules: 1` on the boot screen, on the next line after ...`assuming EfiResetWarm: Unsupported`?

kmauleon, 2022-12-13:
also can you please try to update one firmware at a time... BIOS System Firmware or EC Firmware first then MEFW

kmauleon, 2023-01-17:
as per owner this could be fwupd related @hughsie 
I think it related to fwupd tool, because 1.6.4 version is PASS but 1.8.8 failed.
Could you let linux owner to check this issue firstly?

btw @ehsavoie can you please try >> fwupdmgr reinstall

kmauleon, 2023-01-29:
hi @ehsavoie have you tried updating BIOS to 1.78?

kmauleon, 2023-03-24:
@ehsavoie any update from your side please

kmauleon, 2024-02-19:
no update for a year... closing this for now...

closed 2024-02-19 by kmauleon