fwupd/firmware-lenovo #297: Thinkpad X1 carbon 9th gen won't install new embedded controller fw

PixelHegel, 2022-12-07:
**Describe the bug:**
1. I got the X1c 9th from a secondary market and found it has a low version of BIOS. 0.x.x BIOS (I don't remember the exact version), and N32HT10x (0.10) embedded controller fw
2. I'm running an Archlinux, at the beginning I burn a bootable CD into a USB disk to update the BIOS, and the BIOS is updated to N32ET81W(1.57) successfully, but the Embedded Controller is still N32HT10x (0.10). And it started show `2nd ECDSA region was changed. Please update ECFW which corresponds it` when power on.
3. Then I tried to use `fwupdmgr` to update the embedded controller, I tried ` fwupdmgr update, fwupdmgr install XXXXXX.cab`, but after install/reboot( looks like the update process went well, no error message, and I can see a progress bar under the Lenovo logo), but after reboot, the version of embedded controller is still  N32HT10x (0.10)

**Other ways I have tried:**
1. I have tried downgrading the BIOS to the lowest version(1.47) I can get with fwupdmgr, and updating it again,  the EC version is still N32HT10x (0.10)
2. I also tried using another disk with windows installed, using the windows Lenovo fw update tool to do this, but the issue still remains, the version of embedded controller fw never changed, and it's still N32HT10x (0.10)

**Steps to Reproduce**
sudo fwupmgr update
reboot

**fwupdmgr get-updates**

> Devices with no available firmware updates: 
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • WDC PC SN730 SDBPNTY-1T00-1101
>  • ELAN0673:00 04F3:3188
>  • 0000:00:1f.5
>  • Integrated Camera
>  • Prometheus IOTA Config
> Devices with the latest available firmware version:
>  • Intel Management Engine
>  • System Firmware
>  • UEFI Device Firmware
>  • Prometheus
> LENOVO 20XWSITH04
> │
> └─Embedded Controller:
>   │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
>   │   Summary:            UEFI ESRT device
>   │   Current version:    0.0.10
>   │   Minimum Version:    0.0.10
>   │   Vendor:             Lenovo (DMI:LENOVO)
>   │   Update State:       Success
>   │   Problems:           • Device requires AC power to be connected
>   │   GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
>   │   Device Flags:       • Internal device
>   │                       • System requires external power source
>   │                       • Supported on remote server
>   │                       • Needs a reboot after installation
>   │                       • Device is usable for the duration of the update
>   │                       • Updatable
>   │ 
>   ├─ThinkPad X1 Carbon 9th / X1 Yoga 6th:
>   │     New version:      0.1.33
>   │     Remote ID:        lvfs
>   │     Release ID:       15079
>   │     Summary:          Lenovo ThinkPad X1 Carbon 9th / X1 Yoga 6th Embedded Controller Firmware
>   │     License:          Proprietary
>   │     Size:             767.4 kB
>   │     Created:          2022-08-18
>   │     Urgency:          High
>   │     Vendor:           Lenovo
>   │     Release Flags:    • Is upgrade
>   │     Description:      
>   │     Lenovo Embedded Controller Firmware Version 1.33
>   │     
[... 482 lines cut ...]
> │     Device ID:          8ea4a41a1436a078768d0c5e5298b866c412309c
> │     Summary:            Unmatched performance for high-speed I/O
> │     Problems:           • Device requires AC power to be connected
> │     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
> │     Device Flags:       • Internal device
> │                         • System requires external power source
> │                         • Signed Payload
> │   
> └─WDC PC SN730 SDBPNTY-1T00-1101:
>       Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
>       Summary:            NVM Express solid state drive
>       Current version:    11130001
>       Vendor:             Sandisk Corp (NVME:0x15B7)
>       Serial Number:      20095T451410
>       Problems:           • Device requires AC power to be connected
>       GUIDs:              fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2 ← NVME\VEN_15B7&DEV_5006
>                           a39943dd-3afb-54f8-b110-c5a21f071200 ← NVME\VEN_15B7&DEV_5006&REV_00
>                           12c86995-0b90-5ec5-98f3-7a6ed4ca50e0 ← NVME\VEN_15B7&DEV_5006&SUBSYS_15B75006
>                           7b974856-317d-5538-a915-de0a6353750f ← NVME\VEN_15B7&DEV_5006&SUBSYS_15B75006&REV_00
>                           6820f074-c740-5edd-b5fb-7db6ec7ff103 ← WDC PC SN730 SDBPNTY-1T00-1101
>       Device Flags:       • Internal device
>                           • System requires external power source
>                           • Needs a reboot after installation
>                           • Device is usable for the duration of the update
>                           • Updatable

hughsie, 2022-12-07:
@mrhpearson PVT perhaps?

mrhpearson, 2022-12-20:
So if you're on 0.0.x firmware - that's, I believe, pre-release. Tread very carefully as it's easy to brick a unit (voice of experience...)

I think I'm going to have to go through the FW release notes and find the interim steps to get you updated. Can you send me a note (markpearson at lenovo dot com) and I'll have to do some digging and then will figure out some instructions for you. We should be able to get this fixed....but whoever sold you that system did not do you any favours :/

PixelHegel, 2022-12-22:
> So if you're on 0.0.x firmware - that's, I believe, pre-release. Tread very carefully as it's easy to brick a unit (voice of experience...)
>
> I think I'm going to have to go through the FW release [...]

Thanks a lot! I have sent you an email

mrhpearson, 2023-01-05:
Hi - cleaning up my inboxes after the inevitable post holiday swamp...
Did I reply to your email - I couldn't find it and was worried I might have missed it if it arrived just before the holidays. Please send me a nudge with wherever we were at

PixelHegel, 2023-01-06:
> Hi - cleaning up my inboxes after the inevitable post holiday swamp... Did I reply to your email - I couldn't find it and was worried I might have missed it if it arrived just before the holidays. Please send me a nudge with wherever we were at

Hi @mrhpearson  looks like I haven't gotten your reply,  I sent you an email again just now, thanks for the help!

mrhpearson, 2023-01-12:
I'm sorry - I'm just not seeing it :( 
Can you try my personal account - the lenovo email server may be filtering you for some reason. mpearson-lenovo at squebb dot ca.
Mark

mrhpearson, 2023-01-16:
Discussed by email. 
I need to close this case as the system is an unsupported SIT proto system that is not able to be updated by software (nd should not have been released outside Lenovo).
Mark

closed 2023-01-16 by mrhpearson

Thinkpadx12, 2023-08-22:
Help, my notebook also has this problem, also BIOS update, EC version or 0.15 has not changed, I also bought a second-hand market notebook, maybe also a pre-release version of BIOS, landlord your problem solved? My laptop model is thinkpad x12D-GEN1