fwupd/firmware-lenovo #299: [ThinkPad Thunderbolt 4 Workstation Dock] 2 external displays not working

vrumjantsev, 2022-12-12:
Two external displays not working while connected to dock under Fedora 36, tried to connect them via 2xDP and DP+HDMI

kernel `6.0.9-200.fc36.x86_64`

`fwupdtool get-devices` output
LENOVO 20YQS0S800
│
├─Unknown Device:
│     Device ID:          cf294bf55b333004beb7c41f952c1838c23e1f4a
│     Current version:    38.0
│     Vendor:             USB:0x8087
│     GUIDs:              b3f8b518-ebdd-5969-9b44-6b096eff5b9b ← USB\VID_8087&PID_0B40
│                         27f4ed60-ed53-52a7-ae28-ea87d9a23302 ← USB\VID_8087&PID_0B40&REV_1234
│                         350ec5bb-25c0-543e-8d4c-72c30d2b8df0 ← TBT-01082031
│     Device Flags:       • Updatable
│                         • Signed Payload
│   
├─CT1000P3PSSD8:
│     Device ID:          c430a03ca2a65dfe2412ff950c79c51f6aec1317
│     Summary:            NVM Express solid state drive
│     Current version:    P9CR409
│     Vendor:             Micron/Crucial Technology (NVME:0xC0A9)
│     Serial Number:      2227E6457653
│     GUIDs:              024834f2-5768-5565-b929-e3f57c0fcd69 ← NVME\VEN_C0A9&DEV_540A
│                         0d324641-55e0-5fb4-a26c-267c7813d196 ← NVME\VEN_C0A9&DEV_540A&REV_01
│                         6d4f3936-021a-57df-bbd9-493e8f74da7e ← NVME\VEN_C0A9&DEV_540A&SUBSYS_C0A95021
│                         56de3854-3426-5ff3-a9c3-6d75fd360a90 ← NVME\VEN_C0A9&DEV_540A&SUBSYS_C0A95021&REV_01
│                         ef8a1f6f-bc19-5799-8bd3-142cf5db443c ← CT1000P3PSSD8
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│   
├─CT1000P3PSSD8:
│     Device ID:          f428758c51104d245255ec79a167bab59a5596b5
│     Summary:            NVM Express solid state drive
│     Current version:    P9CR409
│     Vendor:             Micron/Crucial Technology (NVME:0xC0A9)
│     Serial Number:      2227E645766B
│     GUIDs:              024834f2-5768-5565-b929-e3f57c0fcd69 ← NVME\VEN_C0A9&DEV_540A
│                         0d324641-55e0-5fb4-a26c-267c7813d196 ← NVME\VEN_C0A9&DEV_540A&REV_01
│                         6d4f3936-021a-57df-bbd9-493e8f74da7e ← NVME\VEN_C0A9&DEV_540A&SUBSYS_C0A95021
│                         56de3854-3426-5ff3-a9c3-6d75fd360a90 ← NVME\VEN_C0A9&DEV_540A&SUBSYS_C0A95021&REV_01
│                         ef8a1f6f-bc19-5799-8bd3-142cf5db443c ← CT1000P3PSSD8
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│   
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    01
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              a8de746a-5e87-5e0d-b546-d58484229fe9 ← PCI\VEN_8086&DEV_9A70
│                         6dd7321c-23ee-5079-b69f-18ef8bd54591 ← PCI\VEN_8086&DEV_9A70&REV_01
│                         020fa351-4b0a-5657-8877-b6c66b9596c6 ← PCI\VEN_8086&DEV_9A70&SUBSYS_17AA22D8
│                         71e17c94-e13d-55a8-88db-19216615c1dd ← PCI\VEN_8086&DEV_9A70&SUBSYS_17AA22D8&REV_01
│                         eab97dfd-8f6d-5224-96f4-d2aa7dade187 ← PCI\VEN_8086&DEV_9A70&REV_00
│                         a7dfeb7d-9745-5976-8fd2-3488f4760356 ← PCI\VEN_8086&DEV_9A70&SUBSYS_17AA22D8&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─GA104GLM [RTX A5000 Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE, PCI:0x8086)
│     GUIDs:              5fcc1f70-8421-5ca1-ba0f-86fde921a136 ← PCI\VEN_10DE&DEV_24B6
│                         b232b771-5cc7-5811-be1b-d473f16515b1 ← PCI\VEN_10DE&DEV_24B6&REV_A1
│                         8eb9ecea-bf2a-5b4e-85e8-81dc59bcaccf ← PCI\VEN_10DE&DEV_24B6&SUBSYS_17AA22D8
│                         7587cf00-c467-5464-a7a3-6878d7311a0b ← PCI\VEN_10DE&DEV_24B6&SUBSYS_17AA22D8&REV_A1
│                         c7dac7da-12ca-57be-ba1d-8617ec295689 ← PCI\VEN_8086&DEV_9A01
│                         7305c01c-2139-594e-ace6-16ab469a5d85 ← PCI\VEN_8086&DEV_9A01&REV_05
[... 327 lines cut ...]
│   
├─Unifying Receiver:
│     Device ID:          10d6fb30dfd575cb7f6071b23174e5648d1ac232
│     Summary:            Miniaturised USB wireless receiver
│     Current version:    RQR12.11_B0032
│     Bootloader Version: BOT01.04_B0016
│     Vendor:             HIDRAW:0x046D|USB:0x046D
│     Install Duration:   30 seconds
│     GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
│                         279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│                         fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
│     Device Flags:       • Updatable
│                         • Signed Payload
│   
└─Xeon W-11955M CPU @ 2.60GHz:
      Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
      Current version:    0x00000040
      Vendor:             Intel
      GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
                          30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
                          a6bd4ca5-75a6-5796-b564-66b5cab1b11b ← CPUID\PRO_0&FAM_06&MOD_8D
                          d9dd5e77-df17-5bab-b5ec-22827598bfed ← CPUID\PRO_0&FAM_06&MOD_8D&STP_1
      Device Flags:       • Internal device
`
`

mrhpearson, 2022-12-20:
Hi,
This is somewhat of a known issue - we've got a bunch of exercise ongoing for improving docking support but no answers yet.
This isn't the best place to track this as it's not really a fw update issue. I'll find one of our forum topics that are covering that and post as it will be better tracked there.
Mark

vrumjantsev, 2022-12-27:
Hi @mrhpearson 
I believe that you are talking about this topic