fwupd/firmware-lenovo #301: Update fails on Thinkpad X1 Carbon Gen 10

echovl, 2022-12-14:


kmauleon, 2022-12-14:
hi @echovl can you please try to update MEFW separately.  You may do this via Gnome Software or download the cab file from LVFS site https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3ARN.firmware

Manual Update Using fwupdmgr
Make sure the AC adapter is firmly connected to the target computer
Launch Terminal
Login as root user
Navigate to the directory where the cabinet file was placed
Execute as below:

        fwupdmgr --allow-older --allow-reinstall install <cab_filename.cab>

Restart the system manually if system did not restart automatically
Wait for the system update to complete then system will automatically restart

echovl, 2022-12-15:
Hey @kmauleon , just tried that but got the same error

kmauleon, 2022-12-15:
is secure boot disabled?  could this be distro specific issue?
I have below setup with Pass results
     Ubuntu 22.04   with fwupd 1.7.5 / 1.7.9
     Fedora WS Live 37 with fwupd 1.8.5

jrgiacone, 2023-01-02:
I on a thinkpad t14 gen 3 amd and have  been having the same issue for the last few months. Just says capsule not found and resets itself. Is there any update on this?

kmauleon, 2023-01-04:
hi @jrgiacone please open another ticket for proper tracking as they have different firmware owners... thank you and Happy New Year!

kmauleon, 2023-01-06:
can you please try **fwupdmgr reinstall**

jvmarti, 2023-01-10:
Can you send the output of the following command?
   **$ bootctl status**

I think the reason it says UEFI NOT FOUND is caused by the system trying to update the "UEFI Secure Boot Forbidden Signature Database"

> ├─UEFI Device Firmware:
> │     Device ID:          2656b5b7e7c4f91fef1537f93095449cda5fb264
> │     Summary:            UEFI ESRT device
> │     Current version:    0.0.0.1
> │     Vendor:             Lenovo (DMI:LENOVO)
> │     Update State:       Failed
> │     Update Error:       �[31m�[1mfailed to update to 0: unsuccessful�[0m
> │     GUID:               7e5534a3-2069-414b-90b6-3e365d2ccd09

and no capsule file is found for the corresponding GUID (7e5534a3-2069-414b-90b6-3e365d2ccd09) added under fw.
_(also kinda weird that it's current version is the minimum)_

> │   ├── arch
> │   │   ├── fw
> │   │   │   └── fwupd-23192307-d667-4bdf-af1a-6059db171246.cap
> │   │   └── fwupdx64.efi

I also noticed that there is an entry for a Platform Key.

> │ ├─UEFI Platform Key:
> │ │     Device ID:        6924110cde4fa051bfdc600a60620dc7aa9d3c6a
> │ │     Summary:          Lenovo Ltd. PK CA 2012
> │ │     Vendor:           Lenovo Ltd.
> │ │     GUID:             71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E

Was there a time, you tried setting up the "Secure Boot"?

kmauleon, 2023-01-29:
hi @echovl any update from your side?  have you tried updating with secure boot disabled?  Please advise.  thank you very much.

kmauleon, 2023-03-07:
this might be related as well >> https://github.com/fwupd/fwupd/issues/1042

kmauleon, 2023-03-10:
might help >> https://github.com/fwupd/firmware-lenovo/issues/319#issuecomment-1459620317

kmauleon, 2023-03-24:
@echovl any update from your side please

kmauleon, 2024-02-19:
already a year with no update closing this for now...

closed 2024-02-19 by kmauleon

roobre, 2024-02-20:
I had a similar (not sure if the same) issue on my X1 Gen 10 where EFI bios was stuck in 0.0.0.1, and upgrades failed several times. I was able to make an upgrade process to succeed by enabling Intel IME (and other similar IME-ish toggles whose name I don't remember) on the BIOS, running the upgrade, and then disabling them again. Not sure if that's _the_ problem but leaving that workaround written here just in case.