fwupd/firmware-lenovo #302: ThinkPad T590 unable to update Intel Management Engine from 192.72.1757

pschichtel, 2022-12-17:


kmauleon, 2022-12-19:
as per owner, test results PASS with Ubuntu and RedHat.  Could be distro specific issue or fwupd version?
BTW, can you please share ME version as shown in BIOS Setup Menu?

@hughsie what's the recommended fwupd version?  Please advise.... thank you very much

hughsie, 2022-12-19:
> @hughsie what's the recommended fwupd version

For UEFI updates, 1.6.x, 1.7.x or 1.8.x would be what I'd advise, and fwupd-efi 1.2 or 1.3. It would be really helpful if we knew what the errors like `Device Error` actually meant.

pschichtel, 2022-12-19:
@kmauleon 

| Name | Version|
|----------|------------|
| UEFI BIOS Version | `N2IET99W (1.77 )` |
| UEFI BIOS Date (Year-Month-Day) | `2022-07-15` |
| Embedded Controller Version | `N2IHT41W (1.25 )` |
| ME Firmware Version | `12.0.90.2072` |
| Machine Type Model | `20N4CTO1WW` |

kmauleon, 2022-12-21:
@hughsie how can we determine the meaning of that error?  should BIOS dev know that?

kmauleon, 2023-01-17:
as per owner this could be fwupd related @hughsie 
I think it related to fwupd tool, because 1.6.4 version is PASS but 1.8.8 failed.
Could you let linux owner to check this issue firstly?

btw @pschichtel  can you please try >> fwupdmgr reinstall

pschichtel, 2023-01-17:
I received an embedded controller upgrade to 0.1.26 and it now apparently it now updated to 192.90.2072, which is older than the initially offered version. It doesn't offer a newer version now. Does that make sense to you @kmauleon ?

kmauleon, 2023-01-17:
hi @pschichtel due to high failure rate for the latest version, i think it was automatically disabled by LVFS.  can you please manually download the latest file if possible and please execute the following... thank you very much
for Corporate ME >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2IRM.firmware
for Consumer ME >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2IRN.firmware

fwupdmgr reinstall

  1. Make sure the AC adapter is firmly connected to the target computer
  2. Launch Terminal
  3. Login as root user
  4. Navigate to the directory where the cabinet file was placed
  5. Execute as below:

        fwupdmgr --allow-older --allow-reinstall install <cab_filename.cab>

  6. Restart the system manually if system did not restart automatically
  7. Wait for the system update to complete

pschichtel, 2023-01-18:
@kmauleon I get the same device error as initially reported with the "Corporate ME" cab file. the "Consumer ME" doesn't install because it doesn't find appliable devices.

nez-nix, 2023-01-18:
I tried the latest Consumer ME cab file on my T590 which was stuck on the same version of firmware as yours (12.0.90.2072). I ran it as root and rebooted and it stuck on that annoying boot screen saying "installing firmware please wait" or whatever the message is. I ended up hitting a key to continue booting after it was obvious that nothing was happening after 5 minutes.

Then I ran software centre in Fedora 37 which is my daily driver. I chose to install the firmware beginning with "HC" in software centre from the 4 firmware packages available. One of them was ME firmware but I can't remember the other two. After the "HC" update it rebooted, did the red bar across the screen and booted into Fedora afterwards.

Then I open software centre again and refresh to scan for updates, expecting to see the other three updates that were present alongside the "HC" one. There is nothing to do and my system is up to date!

nez-nix, 2023-01-18:
`LENOVO 20N5S00G00
│
├─UEFI dbx:
│ │   Device ID:          362301da643102b9f38477387e2193e57abaa590
│ │   Previous version:   77
│ │   Update State:       Success
│ │   Last modified:      2023-01-16 23:05
│ │   GUID:               14503b3d-73ce-5d06-8137-77c68972a341
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      217
│       Licence:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
├─Intel Management Engine:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Previous version:   192.72.1757
│ │   Update State:       Failed
│ │   Update Error:       failed to run update on reboot
│ │   Last modified:      2023-01-18 18:52
│ │   GUID:               bf2d0f8b-f9a9-400c-8914-36c225d16eb4
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T490/ThinkPad P43s/ThinkPad T590/ThinkPad P53s:
│       New version:      192.92.2145
│       Remote ID:        lvfs
│       Release ID:       16551
│       Summary:          Lenovo ThinkPad T490/P43s/T590/P53s Consumer ME Firmware
│       Licence:          Proprietary
│       Size:             2.7 MB
│       Created:          2022-10-27
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N2IRN33W
│       Vendor:           Lenovo
│       Description:      
│       Version 12.0.92.2145 (LVFS: 192.92.2145)
│       
│       Problem Fixes
│       
│       • Intel CSME IPU 2022.3
│       • Mitigated the following security vulnerabilities under issues:
│       Issues:           CVE-2021-33159
│                         CVE-2022-29893
│                         CVE-2022-27497
│                         CVE-2022-26845
│     
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Previous version:   0.1.77
│ │   Update State:       Success
│ │   Last modified:      2023-01-18 19:01
│ │   GUID:               603baf73-b997-45b5-86b4-2f981a008e18
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T490/ThinkPad P43s/ThinkPad T590/ThinkPad P53s/ThinkPad T490 HC:
│       New version:      0.1.78
│       Remote ID:        lvfs
│       Release ID:       17265
│       Summary:          Lenovo ThinkPad T490/ThinkPad P43s/ThinkPad T590/ThinkPad P53s/ThinkPad T490 HC System Firmware
│       Variant:          W-BIOS Machine types: 20N2, 20N3, 20N4, 20N5, 20N6, 20N7, 20Q9, 20QH, 20RH, 20RJ
│       Licence:          Proprietary
│       Size:             25.1 MB
│       Created:          2020-01-10
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       Lenovo ThinkPad T490 System Firmware version 1.78
│       
│       This stable release fixes the following issues:
│       
│       • Fixed an issue where an incorrect Intel Management Engine firmware version might be
│       
│       returned at capsule update under Linux.
│     
└─Embedded Controller:
  │   Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
  │   Previous version:   0.1.25
  │   Update State:       Success
  │   Last modified:      2023-01-18 22:18
  │   GUID:               38ea6335-29ca-417b-8cd4-6b4e5e866f92
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T490:
        New version:      0.1.26
        Remote ID:        lvfs
        Release ID:       15829
        Summary:          Lenovo ThinkPad T490 Embedded Controller Firmware
        Licence:          Proprietary
        Size:             767.1 kB
        Created:          2022-07-15
        Urgency:          High
        Vendor:           Lenovo
        Description:      
        Embedded Controller Firmware N2IHT42W version 1.26. please update BIOS to N2JET28W or later before update ECFW.
      
`

pschichtel, 2023-01-18:
I'm pretty sure this is a duplicate of #296

pschichtel, 2023-01-18:
nvm, the title reads identical, but the description is totally different

kmauleon, 2023-01-18:
hi all... have you tried updating firmware one at a time?   BIOS > EC > MEFW
owner keeps on pointing to fwupd as root cause... but I will try to send him ss of failure rate of latest MEFW

kmauleon, 2023-01-19:
as per owner, please update BIOS to 1.78 which implemented a fix for MEFW capsule update... thank you very much

pschichtel, 2023-01-19:
@kmauleon 

I already am on that version:

```
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.78
│ │   Minimum Version:    0.1.71
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              603baf73-b997-45b5-86b4-2f981a008e18
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
```

I already was on that version, when I tried the manual upgrade you suggested: https://github.com/fwupd/firmware-lenovo/issues/302#issuecomment-1386194082

pschichtel, 2023-02-08:
I just updated successfully to `192.92.2145`.

kmauleon, 2023-02-08:
hi @pschichtel this is great news! may I ask what did you do from last time? thank you very much

pschichtel, 2023-02-08:
@kmauleon nothing at all. I run `fwupdmgr update` as part of my system updates on a daily basis, today it offered the update again and it just worked.

kmauleon, 2023-02-08:
hmmm... that is weird but great news.... thank you very much for the updates... let's close this one?

pschichtel, 2023-02-09:
I guess the one difference that could explain this: due to the version being pulled, i updated a while back to 192.90.2072 (see my comment above), maybe that changed something to make it work? Anyway, closing this.

closed 2023-02-09 by pschichtel