fwupd/firmware-lenovo #303: Failed to recover ME Firmware on Thinkpad T15 Gen2

ScrupPi, 2022-12-19:
After each reboot i get messages prior to BIOS start messages:

> (1) "Reading ME Firmware... Please do not power off! 48% Completed"
> (2) "Recovering ME Firmware...Please do not power off!   10% Completed"
> (3) "Failed to recover ME Firmware..."

After that:
- BIOS starts;
- Ubuntu 22.04 start, but **audio devices are missing**, both output and input;
- Windows 11 starts. Audio works.

What should i do to fix this problem?

I've tried to install recent ME FW manually or downgrade to previous version, but those failed as well.

To install recent FW i've done:
1) Downloaded "Intel Management Engine 15.0 Firmware (Corporate) Update Utility for Linux - ThinkPad" from [https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t15-gen-2-type-20w4-20w5/downloads/driver-list/component?name=Motherboard%20Devices%20%28core%20chipset,%20onboard%20video,%20PCIe%20switches%29](url)
2) $ sudo fwupdmgr --allow-older --allow-reinstall install n38rk03w.cab

> $ sudo fwupdmgr --allow-older --allow-reinstall install n38rk03w.cab
```
Decompressing…           [***************************************]
No supported devices found
```

> $ sudo dmidecode | grep SKU
```
 		Corporate SKU
 	SKU Number: LENOVO_MT_20W5_BU_Think_FM_ThinkPad T15 Gen 2i
 	SKU Number: Not Specified
```

To downgrade:
> (1) $ sudo fwupdmgr downgrade 349bb341230b1a86e5effe7dfe4337e1590227bd
```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Downgrade Intel Management Engine from 240.42.2235 to 240.41.2142?           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities:                          ║
║                                                                              ║
║ 20W5S2SE19 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]:
```
> (2) Confirm, reboot.
> (3) After reboot and "Failed to recover ME Firmware..." message i get this:

```
Error 354: Update to Image with lower VCN is not allowed
Reset System
```
> (4) After that msg "Reading ME Firmware... Please do not power off! 48% Completed" and so on...

Downgrade prevention in BIOS is deactivated.

BIOS info screenshot:
[image]

Devices list:
[fwupdmgr_get-devices.txt](https://github.com/fwupd/firmware-lenovo/files/10259238/fwupdmgr_get-devices.

kmauleon, 2022-12-21:
will ping the owner... thank you very much for reporting

nemo-mona, 2022-12-25:
I have the exact same issue. ME failure message and no audrio. I tried the exact same solution myself but also hit a wall that downgrade isn't possible:
After reboot and ME read failure I get:
Specified firmware is older than installed '240.41.2142 < 240.42.2235'

And the it restarts the same with the latest version again. BIOS settings also allow downgrade but are ignored. 
@kmauleon the link you provided doesn't seem to have the solution. 

Anyone knows how I can downgrade the firmware or get fixed FW?

Thanks

kmauleon, 2022-12-27:
hi @nemo-mona still asking BIOS owners for this one... downgrading MEFW is prohibited... sorry.  Happy holidays!

ScrupPi, 2023-01-06:
> can you please try as suggested here >> [#295 (comment)](https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135)

It worked, problem is gone. Thank you @kmauleon !

1. Did `fwupdmgr reinstall` & `reboot`.
2. Got `Reading ME ...` -> `Recovering ME ...` -> `Failed ...` -> BIOS started boot process.
4. Black screen for 3-5 minutes (suppose it was ME update process).
5. Then normal reboot occured - without `Reading-Recovering-Failed`.
6. Sound issue in Ubuntu is fixed without editing `/etc/modprobe.d/alsa-base.conf`

kmauleon, 2023-01-06:
great! thank you very much @ScrupPi !  closing this ticket ^_^

closed 2023-01-06 by kmauleon

poti1, 2024-03-04:
Worked great for me as well.
Tiny note for others:

ME stand for management engine.
So select that firmware option.

aphta, 2025-01-01:
Also worked, for me for Ubuntu 24.10. on a T14s . Thank you @kmauleon  !

1. Did fwupdmgr reinstall 
2. Choose Option 2
3. I still got "Recovering ME failed" 
4. Waited for 2-3 Minutes (actually I once pressed Enter)
5. Then normal reboot occured - without Reading-Recovering-Failed.
6. Sound issue in Ubuntu is fixed without editing /etc/modprobe.d/alsa-base.conf

Thanks to everyone for the Help